MIT-SPARK/Kimera-VIO #1: Build errors with OpenCV 4

tonioteran, 2019-09-23:
**Description:**
Prerequisites state that an OpenCV version >= 3.3.1 is needed, but the cmake `find_package` command cannot find a newer major version. When removing the 3.3.1, some components fail to build, e.g., `StereoVisionFrontEnd.cpp`, `Tracker.cpp`, `StereoFrame.cpp`.

**Command:**
Running make with OpenCV 4.0.0.

**Console output:**
OpenCV C API variables appear to not be within the scope; an additional include might be warranted for OpenCV >= 4. Some examples include:
- `CV_TM_SQDIFF_NORMED`
- `CV_FONT_HERSHEY_COMPLEX`
- `CV_FILLED`
- `CV_THRESH_TOZERO`

Some example output:

```
/home/tonio/repos/Kimera-VIO/src/UtilsOpenCV.cpp: In static member function ‘static void VIO::UtilsOpenCV::ExtractCorners(const cv::Mat&, std::vector<cv::Point_<float> >*, const double&, const double&, int, const double&, bool)’:
/home/tonio/repos/Kimera-VIO/src/UtilsOpenCV.cpp:241:29: error: ‘CV_TERMCRIT_EPS’ was not declared in this scope
   cv::TermCriteria criteria(CV_TERMCRIT_EPS + CV_TERMCRIT_ITER, 40, 0.001);
                             ^~~~~~~~~~~~~~~
/home/tonio/repos/Kimera-VIO/src/UtilsOpenCV.cpp:241:47: error: ‘CV_TERMCRIT_ITER’ was not declared in this scope
   cv::TermCriteria criteria(CV_TERMCRIT_EPS + CV_TERMCRIT_ITER, 40, 0.001);
                                               ^~~~~~~~~~~~~~~~
/home/tonio/repos/Kimera-VIO/src/UtilsOpenCV.cpp: In static member function ‘static void VIO::UtilsOpenCV::MyGoodFeaturesToTrackSubPix(const cv::Mat&, const int&, const double&, double, const cv::Mat&, const int&, const bool&, const double&, std::pair<std::vector<cv::Point_<float> >, std::vector<double, std::allocator<double> > >*)’:
/home/tonio/repos/Kimera-VIO/src/UtilsOpenCV.cpp:288:55: error: ‘CV_THRESH_TOZERO’ was not declared in this scope
     cv::threshold(eig, eig, maxVal * qualityLevel, 0, CV_THRESH_TOZERO);
                                                       ^~~~~~~~~~~~~~~~
/home/tonio/repos/Kimera-VIO/src/UtilsOpenCV.cpp:400:31: error: ‘CV_TERMCRIT_EPS’ was not declared in this scope
     cv::TermCriteria criteria(CV_TERMCRIT_EPS + CV_TERMCRIT_ITER, 40, 0.001);
                               ^~~~~~~~~~~~~~~
/home/tonio/repos/Kimera-VIO/src/UtilsOpenCV.cpp:400:49: error: ‘CV_TERMCRIT_ITER’ was not declared in this scope
     cv::TermCriteria criteria(CV_TERMCRIT_EPS + CV_TERMCRIT_ITER, 40, 0.001);
                                                 ^~~~~~~~~~~~~~~~
/home/tonio/repos/Kimera-VIO/src/UtilsOpenCV.cpp: In static member function ‘static void VIO::UtilsOpenCV::PlainMatchTemplate(cv::Mat, cv::Mat, cv::Mat&)’:
/home/tonio/repos/Kimera-VIO/src/UtilsOpenCV.cpp:611:24: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
   for (size_t i = 0; i < result_rows; i++) {
                      ~~^~~~~~~~~~~~~
/home/tonio/repos/Kimera-VIO/src/UtilsOpenCV.cpp:612:26: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
     for (size_t j = 0; j < result_cols; j++) {
                        ~~^~~~~~~~~~~~~
/home/tonio/repos/Kimera-VIO/src/UtilsOpenCV.cpp: In static member function ‘static void VIO::UtilsOpenCV::DrawCirclesInPlace(cv::Mat&, const std::vector<cv::Point_<float> >&, const Scalar&, double, const std::vector<int>&, int)’:
/home/tonio/repos/Kimera-VIO/src/UtilsOpenCV.cpp:640:48: error: ‘CV_FONT_HERSHEY_COMPLEX’ was not declared in this scope
                   imagePoints[i] + textOffset, CV_FONT_HERSHEY_COMPLEX, 0.5,
                                                ^~~~~~~~~~~~~~~~~~~~~~~
```

**Additional files:**
* OpenCV version used: 4.0.0
* Operating system and version (e.g. Ubuntu 16.04 or Windows 10): Ubuntu 18.04
* Did you change the source code? (yes / no): no

mhkabir, 2019-09-23:
Can you send a PR with the changes? It should just be a matter of substituting the old constants to the C++ versions.

tonioteran, 2019-09-23:
The only changes needed entail adding the include

```cpp
#include <opencv2/highgui/highgui_c.h>
```

to the following files:

- `src/Histogram.h`
- `src/StereoFrame.h`
- `src/StereoVisionFrontEnd.h`
- `src/Tracker.h`
- `src/UtilsOpenCV.h`

I can go ahead with a pull request, but I'm not sure if that'll break if for versions < 4?

marcusabate, 2019-09-23:
From my quick tests it appears that simply adding this include in to the header files you've listed does not break Kimera when using OpenCV 3.3.1. 

You can definitely go ahead with the pr and we will test more extensively before merging.

closed 2019-09-25 by ToniRV