MIT-SPARK/Kimera-VIO #4: make error

cumtchenchang, 2019-09-25:
Thank you for your good work and I am very interested in it. I download your code and install it, but an error encountered.

CMake Error at /usr/local/share/cmake-3.14/Modules/GoogleTestAddTests.cmake:40 (message):
  Error running test executable.

    Path: '/home/SENSETIME/chenchang1/test/Kimera-VIO/build/testSparkVio'
    Result: Segmentation fault
    Output:
     

CMakeFiles/testSparkVio.dir/build.make:524: recipe for target 'testSparkVio' failed
make[2]: *** [testSparkVio] Error 1
make[2]: *** Deleting file 'testSparkVio'
CMakeFiles/Makefile2:267: recipe for target 'CMakeFiles/testSparkVio.dir/all' failed
make[1]: *** [CMakeFiles/testSparkVio.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2

I tried cmake3.10 and cmake3.14, but this error is still on. Could you help me solve it? Sincerely.

ToniRV, 2019-09-25:
@yunzc did you see this error before?

yunzc, 2019-09-25:
Hi @cumtchenchang could you tell us what versions of gtsam and opengv you are using and also the cmake configurations for gtsam?

cumtchenchang, 2019-09-26:
@yunzc  GTSAM = 4.0, opengv=1.0. cmake3.14 for gtsam.  I will try your suggestion. Thank you.

ToniRV, 2019-09-30:
@cumtchenchang any update? We haven't been able to reproduce your issue.

ToniRV, 2019-09-30:
Closing for now.

marcusabate, 2019-10-06:
@cumtchenchang if this is still an issue, try the latest master.

Also, try installing the `libparmetis-dev` package (`sudo apt-get install libparmetis-dev`). We were able to reproduce a similar error, which was solved by installing this dependency.

Updated installation instructions in 69519e5351b90a4e839662fc84e08617758d62e2.

cumtchenchang, 2019-10-08:
@ToniRV  Thank you for your suggestion, the original issue is still on. Because of China's National Day, I am sorry that I did not update in time. I will follow @marcusabate 's suggestion and update it soon. Thank you.

cumtchenchang, 2019-10-08:
@marcusabate  Thank you for your help. I followed your suggestion and solved the original issue. But a new error occurred.
~/test/Kimera-VIO/build$ make
[  9%] Built target gflags
[ 19%] Built target glog
[ 58%] Built target KimeraVIO
[ 59%] Linking CXX executable stereoVIOEuroc
libKimeraVIO.so: undefined reference to `DUtils::Random::RandomInt(int, int)'
libKimeraVIO.so: undefined reference to `DUtils::Random::SeedRandOnce()'
collect2: error: ld returned 1 exit status
CMakeFiles/stereoVIOEuroc.dir/build.make:143: recipe for target 'stereoVIOEuroc' failed
make[2]: *** [stereoVIOEuroc] Error 1
CMakeFiles/Makefile2:458: recipe for target 'CMakeFiles/stereoVIOEuroc.dir/all' failed
make[1]: *** [CMakeFiles/stereoVIOEuroc.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2

In addition, I checked dependencies by 
sudo apt-get install -y --no-install-recommends apt-utils
sudo apt-get install -y cmake.
However, the error is still on. Thank you.

marcusabate, 2019-10-08:
@cumtchenchang good to hear the suggestion worked, at least partly.

`DUtils::Random::RandomInt` comes from the [DLib](https://github.com/dorian3d/DLib) library which was once a dependency of [DBoW2](https://github.com/dorian3d/DBoW2). Since Kimera uses `DBoW2` for loop closure detection, you're seeing the missing dependency there. 

However, The `DLib` dependency was removed from `DBoW2` a while back in [this commit](https://github.com/dorian3d/DBoW2/commit/532fbb3aecade05297452d18a6a78d512c76f840). 

I'm guessing you may have an older version of `DBoW2` installed that is still inking to the `DLib` library. Can you tell me what commit you are using on `DBoW2` and that `DLib` is not referenced in the `CMakeLists.txt` of your local clone of `DBoW2`?

cumtchenchang, 2019-10-09:
Thank you for your answer. DBoW2-v1.1-free was used in my system. I used the latest code and solved the error. In addition, I find a little error with gcc-5.4.
Kimera-VIO/tests/testStereoVisionFrontEnd.cpp:401:38: error: ‘isnan’ was not declared in this scope
       EXPECT_TRUE(isnan(s.second.uR()));
Replace isnan with std::isnan will be more robust.

closed 2019-10-09 by cumtchenchang