MIT-SPARK/Kimera-VIO #6: compile error about DBoW2, KimeraRPGO.

highlightz, 2019-10-09:
**Description:**
I encontered two errors. The first one is at cmake stage, the second one is at make stage.

**Command 1:**
```
# in directory Kimera-VIO/build, I ran command:
cmake ..
```

**Console output 1:**
```

CMake Warning at CMakeLists.txt:29 (find_package):
  By not providing "FindDBoW2.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "DBoW2", but
  CMake did not find one.

  Could not find a package configuration file provided by "DBoW2" with any of
  the following names:

    DBoW2Config.cmake
    dbow2-config.cmake

  Add the installation prefix of "DBoW2" to CMAKE_PREFIX_PATH or set
  "DBoW2_DIR" to a directory containing one of the above files.  If "DBoW2"
  provides a separate development package or SDK, be sure it has been
  installed.

CMake Warning at CMakeLists.txt:35 (find_package):
  By not providing "FindKimeraRPGO.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "KimeraRPGO", but CMake did not find one.

  Could not find a package configuration file provided by "KimeraRPGO" with
  any of the following names:

    KimeraRPGOConfig.cmake
    kimerarpgo-config.cmake

  Add the installation prefix of "KimeraRPGO" to CMAKE_PREFIX_PATH or set
  "KimeraRPGO_DIR" to a directory containing one of the above files.  If
  "KimeraRPGO" provides a separate development package or SDK, be sure it has
  been installed.

```

**Command 2:**
```
# in directory Kimera-VIO/build, I ran command:
make -j2
```

**Console output 2:**
```
[ 49%] Building CXX object CMakeFiles/KimeraVIO.dir/src/utils/ThreadsafeImuBuffer.cpp.o
[ 50%] Building CXX object CMakeFiles/KimeraVIO.dir/src/utils/Statistics.cpp.o
[ 51%] Building CXX object CMakeFiles/KimeraVIO.dir/src/pipeline/Pipeline.cpp.o
In file included from /home/xs/Desktop/opensource/Kimera-VIO/include/kimera-vio/pipeline/Pipeline.h:30:0,
                 from /home/xs/Desktop/opensource/Kimera-VIO/src/pipeline/Pipeline.cpp:15:
/home/xs/Desktop/opensource/Kimera-VIO/include/kimera-vio/loopclosure/LoopClosureDetector.h:27:25: fatal error: DBoW2/DBoW2.h: No such file or directory
compilation terminated.
CMakeFiles/KimeraVIO.dir/build.make:662: recipe for target 'CMakeFiles/KimeraVIO.dir/src/pipeline/Pipeline.cpp.o' failed
make[2]: *** [CMakeFiles/KimeraVIO.dir/src/pipeline/Pipeline.cpp.o] Error 1
CMakeFiles/Makefile2:597: recipe for target 'CMakeFiles/KimeraVIO.dir/all' failed
make[1]: *** [CMakeFiles/KimeraVIO.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
```

**Additional files:**
Please attach all the files needed to reproduce the error.

Please give also the following information:
* SparkVio branch, tag or commit used: branch master SHA: 80e26054ebdcb515086766d0ccc80959d32f55bd
* GTSAM version used: master
* OpenGV version used: master SHA: 306a54e6c6b94e2048f820cdf77ef5281d4b48ad
* OpenCV version used: type `3.3.1`
* Operating system and version (e.g. Ubuntu 16.04 or Windows 10): Ubuntu 16.04
* Did you change the source code? (yes / no): no

closed 2019-10-09 by highlightz

marcusabate, 2019-10-09:
@highlightz were you able to solve this issue? It looks like you may not have installed DBoW2 correctly. Let us know if this persists.

mgrallos, 2023-02-16:
> @highlightz were you able to solve this issue? It looks like you may not have installed DBoW2 correctly. Let us know if this persists.

hi how to properly install DBOW2?

LimHyungTae, 2025-02-07:
This error still occurs like this: 

[image]

LimHyungTae, 2025-02-10:
FYI. This has been fixed in this commit: https://github.com/MIT-SPARK/Kimera-VIO/commit/ce8c59b7b273ab5ac29db7e5572e1623760e19c7