MIT-SPARK/Kimera-VIO #9: What is missing to make Kimera-VIO working for mono?

K0stIa, 2019-10-17:
I am wondering what is missing in current version to make it working for mono as well?
Is it possible to run current version in mono? Providing such scenario would be nice. :) 

Thanks!
-K.

ToniRV, 2019-10-21:
Mono should work already (with some perf differences wrt stereo), but loop closure detection requires stereo images currently.

K0stIa, 2019-10-21:
Can you provide a scenari for mono?

ToniRV, 2019-10-23:
Euroc dataset has stereo images, but you can just use mono if desired.

closed 2019-11-08 by ToniRV

wangshuailpp, 2019-11-22:
How to run monocular dataset please?
thanks!

K0stIa, 2019-11-22:
I have to say I haven't managed to run this code in poor mono. Looking briefly through code gave me impression that there are some code pieces that work only for stereo, e.g. smoother has only stereo visual factors...

ToniRV, 2020-10-24:
I'm a bit late, but answering to @K0stIa, the stereo visual factors are such that if you do not provide the right image keypoint information (so just left image keypoint information, aka mono) it will still work.
Nevertheless, it is true that there are still some sections of the code that assume stereo configuration, we are working on removing those.

For reference have a look at this section of the SmartStereoProjectionFactor in gtsam:
```
  /**
   * This corrects the Jacobians and error vector for the case in which the right pixel in the monocular camera is missing (nan)
   */
  virtual void correctForMissingMeasurements(const Cameras& cameras, Vector& ue,
      boost::optional<typename Cameras::FBlocks&> Fs = boost::none,
      boost::optional<Matrix&> E = boost::none) const
  {
    // when using stereo cameras, some of the measurements might be missing:
    for(size_t i=0; i < cameras.size(); i++){
      const StereoPoint2& z = measured_.at(i);
      if(std::isnan(z.uR())) // if the right pixel is invalid
      {
        if(Fs){ // delete influence of right point on jacobian Fs
          MatrixZD& Fi = Fs->at(i);
          for(size_t ii=0; ii<Dim; ii++)
            Fi(1,ii) = 0.0;
        }
        if(E) // delete influence of right point on jacobian E
          E->row(ZDim * i + 1) = Matrix::Zero(1, E->cols());

        // set the corresponding entry of vector ue to zero
        ue(ZDim * i + 1) = 0.0;
      }
    }
  }
```