MIT-SPARK/Kimera-VIO #15: Docker fails to build

ijeppsen, 2019-10-28:
**Description:**
I'm trying to follow the docker build instructions, and it fails on building the opengv libraries.  I've reproduced this on both Windows 10 and MacOSX

**Command:**
```
docker build --rm -t kimera_vio -f ./scripts/docker/Dockerfile . 
```

**Console output:**
```
[ 39%] Building CXX object CMakeFiles/opengv.dir/src/absolute_pose/methods.cpp.o
CMakeFiles/opengv.dir/build.make:734: recipe for target 'CMakeFiles/opengv.dir/src/relative_pose/modules/main.cpp.o' failed
c++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-7/README.Bugs> for instructions.
make[2]: *** [CMakeFiles/opengv.dir/src/relative_pose/modules/main.cpp.o] Error 4
make[2]: *** Waiting for unfinished jobs....
CMakeFiles/Makefile2:409: recipe for target 'CMakeFiles/opengv.dir/all' failed
make[1]: *** [CMakeFiles/opengv.dir/all] Error 2
make: *** [all] Error 2
Makefile:140: recipe for target 'all' failed

```

**Additional files:**
Please attach all the files needed to reproduce the error.

Please give also the following information:
* SparkVio branch, tag or commit used
Master/head as of 10/26
* GTSAM version used:
* OpenGV version used:
* OpenCV version used: type `opencv_version`
* Operating system and version (e.g. Ubuntu 16.04 or Windows 10): 
Windows 10, Mac OSX 10.14
* Did you change the source code? (yes / no): 
No

amirhosseingholami, 2019-10-28:
I'm trying to follow the docker build instructions, and it fails on building. on macOS Catalina version 10.15 MacBook Pro (Retina, 13-inch, Early 2015)

I attached the Terminal log file 
[Terminal Saved Output.txt](https://github.com/MIT-SPARK/Kimera-VIO/files/3780031/Terminal.Saved.Output.txt)

ToniRV, 2019-10-29:
Checking this!

amirhosseingholami, 2019-10-30:
> Checking this!

Thank you

danieasv, 2019-10-30:
I'm trying to build it with Docker on Ubuntu 18.04: It fails at this point, saying 
`DBoW2.h: No such file or directory`. Did anyone else get a similar issue?

Console log:
`[ 53%] Building CXX object CMakeFiles/KimeraVIO.dir/src/utils/Statistics.cpp.o
[ 54%] Building CXX object CMakeFiles/KimeraVIO.dir/src/pipeline/Pipeline.cpp.o
In file included from /root/Kimera-VIO/include/kimera-vio/pipeline/Pipeline.h:30:0,
                 from /root/Kimera-VIO/src/pipeline/Pipeline.cpp:15:
/root/Kimera-VIO/include/kimera-vio/loopclosure/LoopClosureDetector.h:27:10: fatal error: DBoW2/DBoW2.h: No such file or directory
 #include <DBoW2/DBoW2.h>
          ^~~~~~~~~~~~~~~
compilation terminated.
CMakeFiles/KimeraVIO.dir/build.make:662: recipe for target 'CMakeFiles/KimeraVIO.dir/src/pipeline/Pipeline.cpp.o' failed
make[2]: *** [CMakeFiles/KimeraVIO.dir/src/pipeline/Pipeline.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
CMakeFiles/Makefile2:718: recipe for target 'CMakeFiles/KimeraVIO.dir/all' failed
make[1]: *** [CMakeFiles/KimeraVIO.dir/all] Error 2
make: *** [all] Error 2
Makefile:140: recipe for target 'all' failed
The command '/bin/sh -c cd Kimera-VIO && mkdir build && cd build && cmake .. && make -j$(nproc)' returned a non-zero code: 2`

amirhosseingholami, 2019-11-01:
> Checking this!

another try today! 
[Terminal Saved Output.txt](https://github.com/MIT-SPARK/Kimera-VIO/files/3798590/Terminal.Saved.Output.txt)

ToniRV, 2019-11-01:
Seeing the same on my Mac:
```
[ 38%] Building CXX object gtsam/CMakeFiles/gtsam.dir/base/timing.cpp.o
[ 40%] Building CXX object gtsam/CMakeFiles/gtsam.dir/geometry/Cal3Bundler.cpp.o
c++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-7/README.Bugs> for instructions.
gtsam/CMakeFiles/gtsam.dir/build.make:158: recipe for target 'gtsam/CMakeFiles/gtsam.dir/base/SymmetricBlockMatrix.cpp.o' failed
make[2]: *** [gtsam/CMakeFiles/gtsam.dir/base/SymmetricBlockMatrix.cpp.o] Error 4
make[2]: *** Waiting for unfinished jobs....
Scanning dependencies of target wrap
[ 40%] Building CXX object wrap/CMakeFiles/wrap.dir/wrap.cpp.o
[ 41%] Linking CXX executable wrap
[ 41%] Built target wrap
CMakeFiles/Makefile2:469: recipe for target 'gtsam/CMakeFiles/gtsam.dir/all' failed
make[1]: *** [gtsam/CMakeFiles/gtsam.dir/all] Error 2
Makefile:162: recipe for target 'all' failed
make: *** [all] Error 2
The command '/bin/sh -c cd gtsam &&     git fetch &&     mkdir build &&     cd build &&     cmake -DCMAKE_INSTALL_PREFIX=/usr/local -DGTSAM_BUILD_TESTS=OFF -DGTSAM_BUILD_EXAMPLES_ALWAYS=OFF -DCMAKE_BUILD_TYPE=Release -DGTSAM_BUILD_UNSTABLE=ON .. &&     make -j$(nproc) install' returned a non-zero code: 2
```
I'll investigate this issue further...

ToniRV, 2019-11-01:
It does work on Ubuntu though...

amirhosseingholami, 2019-11-01:
> It does work on Ubuntu though...

can you please help me on this problem ?

danieasv, 2019-11-02:
A build problem occurred when I tried to build opengv with the path to my Eigen folder. Found a Cmake issue posted [here](https://github.com/laurentkneip/opengv/issues/102) with the file FindEigen.cmake in `opengv/modules/FindEigen.cmake`. I changed line 56 to
`if (EIGEN_INCLUDE_DIR)` 
and successfully did the cmake step (with -DEIGEN_INCLUDE_DIR): 

`cd opengv
mkdir build
cd build
# Replace path to your GTSAM's Eigen
cmake .. -DEIGEN_INCLUDE_DIR=/home/tonirv/Code/gtsam/gtsam/3rdparty/Eigen`

I manually installed all the dependencies. Kimera-VIO works perfect now on Ubuntu 16.04.

violetteavi, 2020-04-30:
The Dockerfile failure is likely due to memory usage.

CMake's Kimera build process is a Huge memory and CPU hog. In my experience, these sorts of internal compile errors come when it uses up all the available resources, then asks for more. I fixed this by putting -j 4 after the build command-- this limits the resources the build process can use, so it no longer asks for more than is available.

closed 2020-05-10 by ToniRV