MIT-SPARK/Kimera-VIO #17: IMU Integration

omarosamahu, 2019-11-12:
I have a problem while integrating my sensor yaml files with Kimera-VIO as shown in the below text
-------------------------------------------------------------------------
No vio parameters specified, using default.
Applying IMU timestamp shift of: 3.59449e+246ns.
&&&&&&&&&&&&&&&&&&&& TRACKER PARAMETERS &&&&&&&&&&&&&&&&&&&&&&&&&&&
** Feature tracking parameters **
klt_win_size_: 24
klt_max_iter_: 30
klt_max_level_: 4
klt_eps_: 0.1
maxFeatureAge_: 15
** Feature detection parameters **
maxFeaturesPerFrame_: 800
quality_level_: 0.001
min_distance_: 20
block_size_: 3
use_harris_detector_: 0
k_: 0.04
** Sparse Stereo Matching parameters **
equalize_image_: 0
nominalBaseline_: 0.11
vision_sensor_type_: 0
toleranceTemplateMatching_: 0.15
templ_cols_: 101
templ_rows_: 11
stripe_extra_rows_: 0
minPointDist_: 0.5
maxPointDist_: 10
bidirectionalMatching_: 0
subpixelRefinementStereo_: 0
** Feature selection parameters **
featureSelectionCriterion_: 0
featureSelectionHorizon_: 3
featureSelectionNrCornersToSelect_: 600
featureSelectionImuRate_: 0.005
featureSelectionDefaultDepth_: 2
featureSelectionCosineNeighborhood_: 0.984808
featureSelectionUseLazyEvaluation_: 1
useSuccessProbabilities_: 1
** RANSAC parameters **
useRANSAC_: 1
minNrMonoInliers_: 10
minNrStereoInliers_: 5
ransac_threshold_mono_: 1e-06
ransac_threshold_stereo_: 1
ransac_use_1point_stereo_: 1
ransac_use_2point_mono_: 1
ransac_max_iterations_: 100
ransac_probability_: 0.995
ransac_randomize_: 0
** STEREO tracker parameters **
intra_keyframe_time_: 0.2
minNumberFeatures_: 0
useStereoTracking_: 1
** OTHER parameters **
disparityThreshold_: 0.5
&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&
LoopClosureDetector:: Loading vocabulary from ../vocabulary/ORBvoc.yml
Loaded vocabulary with 971815 visual words.
Spinning Visualizer.
Shutting down VIO pipeline once processing has finished.
Frontend launched (parallel_run set to 1).
------------------- Initialize Pipeline with frame k = 50--------------------
Spinning StereoFrontEnd.
getRectifiedImages: abnormal baseline: 0, nominalBaseline: 0.11(+/-10%) 

 Using Regular VIO with modality 0
[0, 0, 0]';isam_param
type:              ISAM2GaussNewtonParams
wildfireThreshold: -1
relinearizeThreshold:              0.01
relinearizeSkip:                   1
enableRelinearization:             1
evaluateNonlinearError:            0
factorization:                     CHOLESKY
cacheLinearizedFactors:            1
enableDetailedResults:             0
enablePartialRelinearizationCheck: 0
findUnusedFactorSlots:             1
Initial state seed: 
 - Initial pose: 
|0.005, 5.4873e+247, 2.51834e-316|
|0.133927, 6.52016e+252, 2.56383e-316|
|1.61701, 1.60426e-316, 1.00444e-320|

[-nan, 0, 0]';

 - Initial vel: 4.94066e-324 000000000000 000000000000
 - Initial IMU bias: acc = [0, 0, 0]' gyro = [0, 0, 0]'

Indeterminant linear system detected while working near variable
8646911284551352320 (Symbol: x0).

Thrown when a linear system is ill-posed.  The most common cause for this
error is having underconstrained variables.  Mathematically, the system is
underdetermined.  See the GTSAM Doxygen documentation at
http://borg.cc.gatech.edu/ on gtsam::IndeterminantLinearSystemException for
more information.
ERROR: Variable has type 'x' and index 0
Smoother's factors:
[
	size: 3

Factor 0: PriorFactor on x0
  prior mean: R:
[
 	          0.005 5.48729961e+247 2.51834192e-316;
    	     0.13392721 6.52016458e+252 2.56383233e-316;
    	      1.6170107  1.6042604e-316 1.00443546e-320
  ]
[-nan, 0, 0]';  noise model: Gaussian[
 	-nan -nan -nan -nan -nan -nan;
    	   0 -nan -nan -nan -nan -nan;
    	   0    0 -nan -nan -nan -nan;
    	   0    0    0 -nan -nan -nan;
    	   0    0    0    0 -nan -nan;
    	   0    0    0    0    0 -nan
  ]

Factor 1: PriorFactor on v0
  prior mean: [
 	4.94065646e-324;
    	              0;
    	              0
  ]
isotropic dim=3 sigma=0.001

Factor 2: PriorFactor on b0
  prior mean: acc = [0, 0, 0]' gyro = [0, 0, 0]'
  noise model: diagonal sigmas[0.1; 0.1; 0.1; 0.01; 0.01; 0.01];

 ]
State values
[
	Values with 0 values:
 ]
 =============== START:CATCHING EXCEPTION =============== 
Nr of factors in graph (smoother getFactors): 3, with factors:
[

 ]
Nr of new factors to add: 0 with factors:
[
 (slot # wrt to new_factors_tmp graph) 	

 ]
Nr deleted slots: 0, with slots:
[
	

 ]
Nr of values in state_ : 0, with keys:
[
	
 ]
Nr values in new_values_ : 0, with keys:
[
	
 ]
 =============== END: CATCHING EXCEPTION ===============

ToniRV, 2019-11-20:
Can you provide your sensor .yaml files perhaps?

omarosamahu, 2019-11-21:
Uploaded as .txt files but the content is the same as my yaml files with the same file names provided in the repo 

[sensor_cam1.txt](https://github.com/MIT-SPARK/Kimera-VIO/files/3874340/sensor_cam1.txt)

[sensor_imu.txt](https://github.com/MIT-SPARK/Kimera-VIO/files/3874345/sensor_imu.txt)
sensor_cam0.txt](https://github.com/MIT-SPARK/Kimera-VIO/files/3874334/sensor_cam0.txt)

ToniRV, 2019-12-28:
@omarosamahu we have changed extensively the API for Kimera-VIO so the above does not necessarily apply anymore.
In particular, now we parse our own ImuParams.yaml file inside the params folder (so to avoid problems like this one).
One thing that scaped my sight first is that it seems that there is an IMU timeshift being applied (which is only there in case IMU and camera clocks are not perfectly synchronized), but in your param file it seems as if the shift wasn't initialized to 0: I'm referring to this line `Applying IMU timestamp shift of: 3.59449e+246ns.` (the second one in the first message of this thread).

closed 2023-11-03 by marcusabate