MIT-SPARK/Kimera-VIO #23: equidistant (fisheye) distortion model

berndpfrommer, 2020-01-02:
Nice package, thanks for making this open source!

Question: does Kimera-VIO also support equidistant distortion models? Last time I checked GTSAM was not supporting it, but I have implemented it outside of GTSAM:
https://github.com/berndpfrommer/tagslam/tree/master/include/gtsam_distortion

Before I try hacking an equidistant model into Kimera, I wanted to check that nobody has done this already.

berndpfrommer, 2020-01-07:
finished the implementation (at least in Kimera-VIO-ROS and Kimera-VIO). How do I run the tests in Kimera-VIO/tests?

ToniRV, 2020-01-07:
Oh thanks @berndpfrommer ! Kimera-VIO uses gtests, but Kimera-VIO-ROS does not have a test suite (yet).
There are two ways to run the tests:
- Easiest, server: just push your branch to this repo and the Jenkins server will run the tests for you: see [here](http://ci-sparklab.mit.edu:8080/job/MIT-SPARK-Kimera/).
- Locally non-ROS: in the build folder:
i) Using the classical `make test` will run all the tests.
ii) there should be a `./testKimeraVIO`  executable that will run all the tests.
- Locally ROS: that is the trickiest, since I haven't configured the catkin test to use gtest, and if you run the testKimeraVIO in the build folder of catkin workspace the paths won't be correct, perhaps you can override the gflag inside the main testKimeraVIO executable.

But the easiest is definitely just to push your changes to a  branch in this repo and all the unit tests and performance tests should run fine.

ToniRV, 2020-01-07:
We actually tried to handle the equidistant model in the frontend side of things but it was never extensively tested...

marcusabate, 2023-11-03:
For future visitors, equi model is now supported, but does not perform very well. We are working on better spherical camera representations!

closed 2023-11-03 by marcusabate