MIT-SPARK/Kimera-VIO #24: TBB Warnings and Compilation Error

acxz, 2020-01-11:
**Description:**
First off, I'd like to thank you guys for creating an amazing piece of software in the VIO realm, truly amazing work and I hope it gets used more. I maintain the ArchLinux package [here](https://aur.archlinux.org/packages/kimera-vio/) for this codebase, and recently the build started failing. This is due to a compilation error with one of the tests files: `testMesher.cpp`. Also with an updated version of TBB, `tbb/mutex.h` has been deprecated. If this can be fixed as well, it would be nice to stop seeing such a cluttered build output. Having the build be fixed so ArchLinux users can start using this project again would be great.

**Command:**
```
make
```

**Console output:**
```

[ 80%] Building CXX object CMakeFiles/testKimeraVIO.dir/tests/testParallelPlaneRegularTangentSpaceFactor.cpp.o
In file included from /usr/include/gtsam/geometry/Unit3.h:36,
                 from /usr/include/gtsam/geometry/Rot3.h:25,
                 from /usr/include/gtsam/geometry/OrientedPlane3.h:23,
                 from /home/acxz/.cache/yay/kimera-vio/src/Kimera-VIO-5.0/tests/testParallelPlaneRegularBasicFactor.cpp:25:
/usr/include/tbb/mutex.h:21:140: note: #pragma message: TBB Warning: tbb/mutex.h is deprecated. For details, please see Deprecated Features appendix in the TBB reference manual.
   21 | #pragma message("TBB Warning: tbb/mutex.h is deprecated. For details, please see Deprecated Features appendix in the TBB reference manual.")
      |                                                                                                                                            ^
In file included from /usr/include/gtsam/base/types.h:31,
                 from /usr/include/gtsam/inference/Key.h:25,
                 from /usr/include/gtsam/linear/Scatter.h:22,
                 from /usr/include/gtsam/linear/VectorValues.h:20,
                 from /usr/include/gtsam/base/numericalDerivative.h:31,
                 from /home/acxz/.cache/yay/kimera-vio/src/Kimera-VIO-5.0/tests/testParallelPlaneRegularTangentSpaceFactor.cpp:21:
/usr/include/tbb/task_scheduler_init.h:21:154: note: #pragma message: TBB Warning: tbb/task_scheduler_init.h is deprecated. For details, please see Deprecated Features appendix in the TBB reference manual.
   21 | #pragma message("TBB Warning: tbb/task_scheduler_init.h is deprecated. For details, please see Deprecated Features appendix in the TBB reference manual.")
      |                                                                                                                                                          ^
/home/acxz/.cache/yay/kimera-vio/src/Kimera-VIO-5.0/tests/testMesher.cpp:25:15: error: expected constructor, destructor, or type conversion before ‘(’ token
   25 | DECLARE_string(test_data_path);
      |               ^
/home/acxz/.cache/yay/kimera-vio/src/Kimera-VIO-5.0/tests/testMesher.cpp: In constructor ‘VIO::MesherFixture::MesherFixture()’:
/home/acxz/.cache/yay/kimera-vio/src/Kimera-VIO-5.0/tests/testMesher.cpp:38:29: error: ‘FLAGS_test_data_path’ was not declared in this scope
   38 |     img_name_ = std::string(FLAGS_test_data_path) + "/chessboard_small.png";
      |                             ^~~~~~~~~~~~~~~~~~~~
In file included from /usr/include/gtsam/geometry/Unit3.h:36,
                 from /usr/include/gtsam/geometry/Rot3.h:25,
                 from /usr/include/gtsam/geometry/OrientedPlane3.h:23,
                 from /home/acxz/.cache/yay/kimera-vio/src/Kimera-VIO-5.0/tests/testParallelPlaneRegularTangentSpaceFactor.cpp:25:
/usr/include/tbb/mutex.h:21:140: note: #pragma message: TBB Warning: tbb/mutex.h is deprecated. For details, please see Deprecated Features appendix in the TBB reference manual.
   21 | #pragma message("TBB Warning: tbb/mutex.h is deprecated. For details, please see Deprecated Features appendix in the TBB reference manual.")
      |                       

```

**Additional files:**
None needed.

Please give also the following information:
* SparkVio branch, tag or commit used: 5.0
* GTSAM version used: 4.0.2
* OpenGV version used: Master branch, following [commit](https://github.com/laurentkneip/opengv/commit/306a54e6c6b94e2048f820cdf77ef5281d4b48ad) at time of testing
* OpenCV version used: 4.2.0
* Operating system and version (e.g. Ubuntu 16.04 or Windows 10): ArchLinux
* Did you change the source code? (yes / no): no

ToniRV, 2020-02-11:
Uh oh! These warnings looks ugly indeed, but we are not using tbb mutex explicitly but perhaps via gtsam.
Hopefully they have solved this issue.

Concerning the failing testMesher I think it is because I forgot to include gflags there, but I was not seeing any issue neither locally nor in Jenkins... Hopefully now it is fixed.

acxz, 2020-02-11:
Thx so much! Yeah the TBB warnings can be traced to this issue: https://github.com/borglab/gtsam/issues/208

ToniRV, 2020-03-11:
Thank you @acxz, I hope the issue is resolved from gtsam side (I'll try to ping them).

closed 2020-03-11 by ToniRV