MIT-SPARK/Kimera-VIO #26: Scale Drift optimization

omarosamahu, 2020-01-13:
I have tested the estimated trajectory compared to the ground-truth data using 
[https://github.com/uzh-rpg/rpg_trajectory_evaluation] analysis tool before and after the modification of the code and I have noticed that the scale drift err has increased as described in the attached files 
[image]
[image]

ToniRV, 2020-02-11:
Thanks! Do you remember which commits you were comparing?

omarosamahu, 2020-03-02:
I have compared between "d00b5af6f7aabd85dcb2cbab0a908722b203da4f" and "aca75f72a0aa1f7af688b6c49f0d4ef2dc704d20"

closed 2023-11-03 by marcusabate