MIT-SPARK/Kimera-VIO #27: Euroc example fails to start

m-pilia, 2020-01-14:
**Description:**
It seems the script launching the Euroc example contains some unsupported flags. After removing the flags from the bash script, the example runs correctly.

**Command:**
```
# bash ./scripts/stereoVIOEuroc.bash -p /home/martino/Data/euroc/mav0/
```

**Console output:**
```

Using dataset at path: /home/martino/Data/euroc/mav0/
 Launching:

            ██╗  ██╗██╗███╗   ███╗███████╗██████╗  █████╗
            ██║ ██╔╝██║████╗ ████║██╔════╝██╔══██╗██╔══██╗
            █████╔╝ ██║██╔████╔██║█████╗  ██████╔╝███████║
            ██╔═██╗ ██║██║╚██╔╝██║██╔══╝  ██╔══██╗██╔══██║
            ██║  ██╗██║██║ ╚═╝ ██║███████╗██║  ██║██║  ██║
            ╚═╝  ╚═╝╚═╝╚═╝     ╚═╝╚══════╝╚═╝  ╚═╝╚═╝  ╚═╝

 
ERROR: unknown command line flag 'colorlogtostderr'
ERROR: unknown command line flag 'log_prefix'
ERROR: unknown command line flag 'logtostderr'
ERROR: unknown command line flag 'v'
ERROR: unknown command line flag 'vmodule'

```

**Additional files:**
Please attach all the files needed to reproduce the error.

Please give also the following information:
* SparkVio branch, tag or commit used: bceed7203f
* GTSAM version used: 4.0.2
* OpenGV version used: commit 306a54e
* OpenCV version used: 3.4.7
* Operating system and version (e.g. Ubuntu 16.04 or Windows 10): Arch Linux, kernel 4.19.95
* Did you change the source code? (yes / no): no

ToniRV, 2020-02-11:
This might be because glog or gflags are not correctly linked.
Did you install them?
```
sudo apt-get install libgoogle-glog-dev libgflags-dev
```

m-pilia, 2020-02-11:
I am sure glog and gflags were installed. Also, other command line flags for the program were working, just the five I mentioned in the issue were not recognised by the application.

Sadly I have wiped the machine where I encountered the issue, and I am not working on this any longer, so it is fine for me if you close the issue in case you cannot reproduce it.

closed 2020-03-11 by ToniRV

ToniRV, 2020-03-16:
I actually encountered this issue just recently. It seems like Kimera was linking to a local installation of gflags, which was different than the gflags used by glog.
Modifying the variable: ` ` in the cmake cache solved the issue.
Running `cmake-gui .` in the build folder, one should see the following entries:
[image]

Where originally one of the cmake flags was pointing to a local installation of gflags.
(Btw this only applies to Kimera-VIO (no ROS), since with Kimera-VIO-ROS, Kimera should be linking with gflags provided by gflags_catkin).

Leaving this here just for the sake of completeness.

buqing2009, 2020-08-12:
i have the same problem, i have change the source to commit  [f32943c](https://github.com/MIT-SPARK/Kimera-VIO/commit/f32943c6117006fc6678949d1410827628f95231), but when i **ldd** the output library,
`ldd libkimera_vio.so | grep libglog` , it return nothing. But the print the cmake seems ok,
```
-- ==============================================================
-- ====================  Dependencies ===========================
-- Boost version: 1.58.0
-- Found the following Boost libraries:
--   filesystem
--   serialization
--   regex
--   timer
--   date_time
--   thread
--   system
--   chrono
--   atomic
-- Found glog    (include: /usr/include, library: /usr/lib/x86_64-linux-gnu/libglog.so)
-- GTSAM include directory:  /usr/local/lib/cmake/GTSAM/../../../include
-- Vocabulary file exists, will not download.
-- ===============================================================
-- ================  Configuration Options  ======================
-- CMAKE_CXX_COMPILER_ID type                : GNU
-- CMAKE_CXX_COMPILER_VERSION                : 7.5.0
-- Build flags                                               
--   Build type                              : Release
--   C compilation flags (Release)           :  -O3 -DNDEBUG
--   C++ compilation flags (Release)         :  -O3 -DNDEBUG
-- Configuring done
-- Generating done
```