MIT-SPARK/Kimera-VIO #29: Exporting 3D world model

ruffsl, 2020-02-11:
Many downstream robotic applications are not only reliant on the current pose estimate provided, but also world model collected by SLAM pipelines. This is relevant for planning domains such as in navigation stacks, but also exploration mapping techniques as well, e.g. frontier expansion.

Conventionally, exporting a SLAM framework's internal world representation often consists of projections into discretized occupancy models, either 2D maps or 3D or voxel grids, probabilistic or discrete depending upon application requirements or information available from internal world representation. Such projections could be computed online at runtime or offline ex post facto.

A LIDAR based pipelines, such as [Cartographer](https://github.com/googlecartographer/cartographer_ros), [Slam Toolbox](https://github.com/SteveMacenski/slam_toolbox), often provide such online and offline methods.  

## Online

> https://google-cartographer-ros.readthedocs.io/en/latest/assets_writer.html#exploiting-the-map-generated-by-cartographer-ros

> #### Occupancy grid Node
> The occupancy_grid_node listens to the submaps published by SLAM, builds an ROS occupancy_grid out of them and publishes it. This tool is useful to keep old nodes that require a single monolithic map to work happy until new nav stacks can deal with Cartographer’s submaps directly.
> https://google-cartographer-ros.readthedocs.io/en/latest/ros_api.html#occupancy-grid-node

## Offline

> #### Pbstream Map Publisher Node
> The pbstream_map_publisher is a simple node that creates a static occupancy grid out of a serialized Cartographer state (pbstream format). It is an efficient alternative to the occupancy grid node if live updates are not important.
> https://google-cartographer-ros.readthedocs.io/en/latest/ros_api.html#pbstream-map-publisher-node

That said, we could also see how to adjust navigation2 to deal with Kimera_VIO's world representation directly, but I think support exporting for conventional volumetric occupancy model could benefit for legacy or traditional planning approaches. A related discussion ticket about using mesh models directly for navigation: https://github.com/ros-planning/navigation2/issues/1461

Related:
* https://github.com/MIT-SPARK/Kimera-VIO/issues/12

ToniRV, 2020-03-11:
So relating to this issue, and as I was implying in https://github.com/ros-planning/navigation2/issues/1461, exporting the global 3D Mesh is already possible via Kimera-Semantics by using Voxblox API:  just call the rosservice
```
rosservice call /kimera_semantics_node/generate_mesh "{}"
```

satyamSF, 2022-07-05:
Can I know what format would this exported 3d mesh have? I want to use this map further for AR applications.

dt1729, 2022-12-07:
Hi @ToniRV, trying to generate the full mesh here, is it possible to get the ESDF/TSDF file with this command? Also, is there any documentation for this service call?

marcusabate, 2023-11-03:
Voxblox does support this as one of the services, and Kimera-Semantics will give you the esdf/tsdf file:
```
rosservice call /kimera_semantics_node/generate_mesh
rosservice call /kimera_semantics_node/save_map
```
And so on. For more info, see [here](https://github.com/MIT-SPARK/voxblox/blob/master/docs/pages/The-Voxblox-Node.rst).

closed 2023-11-03 by marcusabate