MIT-SPARK/Kimera-VIO #30: Support for an efficient Localization Mode

ruffsl, 2020-02-11:
For scenarios where robots may return to, or continuously navigate, priorly explored environments, many SLAM pipelines are capable of loading/importing models of previous maped environments, e.g. a serialized pose graph, for pure localization purposes. For applications focused on navigation, rather than mere exploration, this offers a means to reduce computation overhead, as well as prior world map for global path planning. 

LIDAR based pipelines, such as [Slam Toolbox](https://github.com/SteveMacenski/slam_toolbox) in 2D or [Cartographer](https://github.com/googlecartographer/cartographer_ros) in 3D, often provide such importing of prior graphs and efficient localization mode features:

> ### Localization
>
> Localization mode consists of 3 things:
> * Loads existing serialized map into the node
> * Maintains a rolling buffer of recent scans in the pose-graph
> * After expiring from the buffer scans are removed and the underlying map is not affected
>
> https://github.com/SteveMacenski/slam_toolbox#localization

> ### Localization only
> If you have a map you are happy with and want to reduce computations, you can use the localization-only mode of Cartographer which will run SLAM against the existing map and won’t build a new one. This is enabled by running `cartographer_node` with a `-load_state_filename` argument and by defining the following line in your lua config:
> ```
> TRAJECTORY_BUILDER.pure_localization_trimmer = {
>    max_submaps_to_keep = 3,
> }
> ```
> https://google-cartographer-ros.readthedocs.io/en/latest/going_further.html#localization-only
> https://google-cartographer-ros.readthedocs.io/en/latest/assets_writer.html#exploiting-the-map-generated-by-cartographer-ros

I'm not as familiar with the internal architecture of Kimera-VIO, but was wondering if it would be possible to similarly load graph state from disk, and expose a parameter for trimming new pose, and landmark factors when operating under a localization mode runtime; similar to how Cartographer and SLAM Toolbox simply keep a rolling buffer of submaps/scans that are most recent to the current trajectory. I'm not sure how visual encoded landmarks could be retained, but am guessing there might already be some similar export/serialization function around for offline benchmarking.

SteveMacenski, 2020-05-05:
Would localization mode make more sense to be in the Kimera SLAM repo? 

I think this would be a great feature to add. We're considering this over in the ROS2 navigation-land as a potential supported VIO source for odometry fusion, as well as V-SLAM for positioning (https://github.com/ros-planning/navigation2). 

Beyond a typically functional SLAM, we'd also need a way to localize in a prebuilt SLAM session so that users can position themselves in a space over time (days, weeks, months). We'd love to do some demos of Kimera and ROS2 Navigation with integration guides and such.

Does your lab plan on supporting / maintaining this project for the immediate future?

ToniRV, 2020-05-10:
Hi @SteveMacenski,

Yes! Actually, we are working on making Kimera a multi-robot pipeline.
Right now, we only support loop-closure detection and pose-graph optimization in Kimera-VIO which uses Kimera-RPGO.

A full-fledged re-localization pipeline with map serialization and so on seems to be further in the future though. Perhaps, the quickest way would be to use [maplab](https://github.com/ethz-asl/maplab).

robot-mh, 2021-11-09:
@ToniRV 
When will Kimera-Multi opensourced？

Kimera-Multi: Robust, Distributed, Dense Metric-Semantic SLAM for Multi-Robot Systems

dt1729, 2022-11-27:
Just checking in, is this implemented or do we need to use something like DBoW2 for place recognition and then pull the corresponding transform? Any other easier way to do this in Kimera RPGO?