MIT-SPARK/Kimera-VIO #33: Error with CMake version during build process

kartiksonu, 2020-02-19:
**Description:**

**Command:**
```
 catkin build --cmake-args -DCMAKE_BUILD_TYPE=Release -DGTSAM_USE_SYSTEM_EIGEN=ON
```

**Console output:**
```
Errors     << kimera_vio_ros:cmake /home/natsu/catkin_ws/logs/kimera_vio_ros/build.cmake.001.log            
CMake Error at /usr/share/cmake-3.5/Modules/CMakeFindDependencyMacro.cmake:45 (message):
  Invalid arguments to find_dependency
Call Stack (most recent call first):
  /home/natsu/catkin_ws/devel/lib/cmake/kimera_vio/kimera_vioConfig.cmake:14 (find_dependency)
  /home/natsu/catkin_ws/devel/share/catkin_simple/cmake/catkin_simple-extras.cmake:40 (find_package)
  CMakeLists.txt:16 (catkin_simple)

cd /home/natsu/catkin_ws/build/kimera_vio_ros; catkin build --get-env kimera_vio_ros | catkin env -si  /usr/bin/cmake /home/natsu/catkin_ws/src/Kimera-VIO-ROS --no-warn-unused-cli -DCATKIN_DEVEL_PREFIX=/home/natsu/catkin_ws/devel -DCMAKE_INSTALL_PREFIX=/home/natsu/catkin_ws/install -DCMAKE_BUILD_TYPE=Release -DCMAKE_BUILD_TYPE=Release -DGTSAM_USE_SYSTEM_EIGEN=ON; cd -
............................................................................................................
Failed     << kimera_vio_ros:cmake                         [ Exited with code 1 ]                           
Failed    <<< kimera_vio_ros                               [ 0.4 seconds ]                                  
[build] Summary: 13 of 14 packages succeeded.                                                               
[build]   Ignored:   None.                                                                                  
[build]   Warnings:  None.                                                                                  
[build]   Abandoned: None.                                                                                  
[build]   Failed:    1 packages failed.                                                                     
[build] Runtime: 5.1 seconds total. 
# remove this line and paste your console output HERE - no screenshots please

```

**Additional files:**
Please attach all the files needed to reproduce the error.

Please give also the following information:
* SparkVio branch, tag or commit used
* GTSAM version used:
* OpenGV version used:
* OpenCV version used: type `opencv_version`
* Operating system and version (e.g. Ubuntu 16.04 or Windows 10): 
* Did you change the source code? (yes / no):

ToniRV, 2020-02-19:
Can you try removing the 0.3.5 in this line?:
 https://github.com/MIT-SPARK/Kimera-VIO/blob/bceed7203f824c1bcffc0c195a186d13c5ea2b22/cmake/kimera_vioConfig.cmake.in#L14

ToniRV, 2020-03-11:
The fix is now merged.

closed 2020-03-11 by ToniRV