MIT-SPARK/Kimera-VIO #35: Build Failure without fatal error

xin-peng, 2020-03-09:
**Description:**
Hello, I am trying to install Kimera-VIO in Mac Mojave. I installed all of lastest dependents. But when I run 'make -j4' for Kimera-VIO, the process terminate without fatal error.

**Command:**
```
cd Kimera-VIO
mkdir build
cd build
cmake ..
make -j4
```

**Console output:**
```

[  2%] Built target gtest
[  3%] Building CXX object CMakeFiles/kimera_vio.dir/src/dataprovider/KittiDataProvider.cpp.o
[  4%] Building CXX object CMakeFiles/kimera_vio.dir/src/frontend/StereoImuSyncPacket.cpp.o
[  6%] Building CXX object CMakeFiles/kimera_vio.dir/src/frontend/StereoVisionFrontEnd.cpp.o
[  7%] Building CXX object CMakeFiles/kimera_vio.dir/src/frontend/VisionFrontEndModule.cpp.o
In file included from /usr/local/include/Kimera-VIO/src/frontend/VisionFrontEndModule.cpp:15:
In file included from /usr/local/include/Kimera-VIO/include/kimera-vio/frontend/VisionFrontEndModule.h:17:
In file included from /usr/local/include/Kimera-VIO/include/kimera-vio/frontend/StereoVisionFrontEnd-definitions.h:19:
In file included from /usr/local/include/Kimera-VIO/include/kimera-vio/frontend/Tracker.h:32:
In file included from /usr/local/include/Kimera-VIO/include/kimera-vio/frontend/FeatureSelector.h:33:
In file included from /usr/local/include/Kimera-VIO/include/kimera-vio/backend/VioBackEndParams.h:30:
/usr/local/include/Kimera-VIO/include/kimera-vio/dataprovider/DataProviderInterface-definitions.h:74:16: warning: 
      'VIO::VioNavStateTimestamped::equals' hides overloaded virtual function
      [-Woverloaded-virtual]
  virtual bool equals(const VioNavStateTimestamped& rhs) const;
               ^
/usr/local/include/Kimera-VIO/include/kimera-vio/dataprovider/DataProviderInterface-definitions.h:54:16: note: 
      hidden overloaded virtual function 'VIO::VioNavState::equals' declared
      here: type mismatch at 1st parameter ('const VIO::VioNavState &' vs 'const
      VIO::VioNavStateTimestamped &')
  virtual bool equals(const VioNavState& rhs) const;
               ^
In file included from /usr/local/include/Kimera-VIO/src/frontend/VisionFrontEndModule.cpp:15:
In file included from /usr/local/include/Kimera-VIO/include/kimera-vio/frontend/VisionFrontEndModule.h:17:
In file included from /usr/local/include/Kimera-VIO/include/kimera-vio/frontend/StereoVisionFrontEnd-definitions.h:19:
In file included from /usr/local/include/Kimera-VIO/include/kimera-vio/frontend/Tracker.h:32:
In file included from /usr/local/include/Kimera-VIO/include/kimera-vio/frontend/FeatureSelector.h:36:
/usr/local/include/Kimera-VIO/include/kimera-vio/frontend/VioFrontEndParams.h:41:9: warning: 
      field 'featureSelectionHorizon_' will be initialized after field
      'featureSelectionNrCornersToSelect_' [-Wreorder]
        featureSelectionHorizon_(3),  // in seconds
        ^
In file included from /usr/local/include/Kimera-VIO/src/frontend/StereoVisionFrontEnd.cpp:16:
In file included from /usr/local/include/Kimera-VIO/include/kimera-vio/frontend/StereoVisionFrontEnd.h:27:
In file included from /usr/local/include/Kimera-VIO/include/kimera-vio/backend/VioBackEnd-definitions.h:29:
In file included from /usr/local/include/Kimera-VIO/include/kimera-vio/frontend/StereoVisionFrontEnd-definitions.h:19:
In file included from /usr/local/include/Kimera-VIO/include/kimera-vio/frontend/Tracker.h:32:
In file included from /usr/local/include/Kimera-VIO/include/kimera-vio/frontend/FeatureSelector.h:33:
In file included from /usr/local/include/Kimera-VIO/include/kimera-vio/backend/VioBackEndParams.h:30:
/usr/local/include/Kimera-VIO/include/kimera-vio/dataprovider/DataProviderInterface-definitions.h:74:16: warning: 
      'VIO::VioNavStateTimestamped::equals' hides overloaded virtual function
      [-Woverloaded-virtual]
  virtual bool equals(const VioNavStateTimestamped& rhs) const;
               ^
/usr/local/include/Kimera-VIO/include/kimera-vio/dataprovider/DataProviderInterface-definitions.h:54:16: note: 
      hidden overloaded virtual function 'VIO::VioNavState::equals' declared
      here: type mismatch at 1st parameter ('const VIO::VioNavState &' vs 'const
      VIO::VioNavStateTimestamped &')
  virtual bool equals(const VioNavState& rhs) const;
               ^
In file included from /usr/local/include/Kimera-VIO/src/frontend/StereoVisionFrontEnd.cpp:16:
In file included from /usr/local/include/Kimera-VIO/include/kimera-vio/frontend/StereoVisionFrontEnd.h:27:
In file included from /usr/local/include/Kimera-VIO/include/kimera-vio/backend/VioBackEnd-definitions.h:29:
In file included from /usr/local/include/Kimera-VIO/include/kimera-vio/frontend/StereoVisionFrontEnd-definitions.h:19:
In file included from /usr/local/include/Kimera-VIO/include/kimera-vio/frontend/Tracker.h:32:
[... 89 lines cut ...]
            ^
In file included from /usr/local/include/Kimera-VIO/src/frontend/StereoVisionFrontEnd.cpp:16:
In file included from /usr/local/include/Kimera-VIO/include/kimera-vio/frontend/StereoVisionFrontEnd.h:36:
/usr/local/include/Kimera-VIO/include/kimera-vio/logging/Logger.h:30:13: warning: 
      unused function 'OpenFile' [-Wunused-function]
static void OpenFile(const std::string& output_filename,
            ^
3 warnings generated.
2 warnings generated.
3 warnings generated.
make[1]: *** [CMakeFiles/kimera_vio.dir/all] Error 2
make: *** [all] Error 2

```

**Additional files:**
Please attach all the files needed to reproduce the error.

Please give also the following information:
* SparkVio branch, tag or commit used
* GTSAM version used:
* OpenGV version used:
* OpenCV version used: type `opencv_version`
* Operating system and version (e.g. Ubuntu 16.04 or Windows 10): 
* Did you change the source code? (yes / no):

FangGet, 2020-03-09:
```
/usr/local/include/Kimera-VIO/src/dataprovider/KittiDataProvider.cpp:45:38: error: 
      use of undeclared identifier 'CV_LOAD_IMAGE_UNCHANGED'
  cv::Mat img = cv::imread(img_name, CV_LOAD_IMAGE_UNCHANGED);
```

xin-peng, 2020-03-09:
Thanks @FangGet I just checked, this error is caused by name changing in OpenCv 4. Right now I have no idea about how to solve this problem except reinstall lower version OpenCv. Do you have any idea? https://stackoverflow.com/questions/28534070/cv-load-image-unchanged-undeclared-identifier

FangGet, 2020-03-09:
Lower down the opencv version to 3.x or just revise the code, change the flag from "CV_LOAD_IMAGE_UNCHANGED" to "IMREAD_UNCHANGED" since you just need to compile it success. but I'm not sure if there are any other features changed between opencv 3.x and 4.x.

ToniRV, 2020-03-10:
Hi, thanks for the issue report. Since we are not supporting Kitti right now, I think the easiest for you would be to not comoile the KittiDataProvider file. You can do so commenting out any Kitti related file in the CMakeLists.txt inside ‘include/kimera-vio/dataprovider’ and ‘src/dataprovider’.
These Kitti files will be removed anyways for next release.

xin-peng, 2020-03-10:
Hello, thanks for your reply. Here is another error about opencv. Should I just switch to opencv 3.4.4 instead of opencv 4?
[  2%] Built target gtest
[  3%] Building CXX object CMakeFiles/kimera_vio.dir/src/pipeline/Pipeline.cpp.o
[  5%] Building CXX object CMakeFiles/kimera_vio.dir/src/loopclosure/LoopClosureDetector.cpp.o
[  6%] Building CXX object CMakeFiles/kimera_vio.dir/src/loopclosure/LoopClosureDetectorParams.cpp.o
[  7%] Building CXX object CMakeFiles/kimera_vio.dir/src/loopclosure/LcdThirdPartyWrapper.cpp.o
In file included from /usr/local/include/Kimera-VIO/src/loopclosure/LoopClosureDetector.cpp:34:
/usr/local/include/Kimera-VIO/include/kimera-vio/loopclosure/LoopClosureDetector.h:26:10: fatal error: 
      'opencv/cv.hpp' file not found
#include <opencv/cv.hpp>
         ^~~~~~~~~~~~~~~
1 error generated.
make[2]: *** [CMakeFiles/kimera_vio.dir/src/loopclosure/LoopClosureDetector.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
In file included from /usr/local/include/Kimera-VIO/src/pipeline/Pipeline.cpp:15:
In file included from /usr/local/include/Kimera-VIO/include/kimera-vio/pipeline/Pipeline.h:25:
In file included from /usr/local/include/Kimera-VIO/include/kimera-vio/backend/VioBackEnd-definitions.h:29:
In file included from /usr/local/include/Kimera-VIO/include/kimera-vio/frontend/StereoVisionFrontEnd-definitions.h:19:
In file included from /usr/local/include/Kimera-VIO/include/kimera-vio/frontend/Tracker.h:32:
In file included from /usr/local/include/Kimera-VIO/include/kimera-vio/frontend/FeatureSelector.h:33:
In file included from /usr/local/include/Kimera-VIO/include/kimera-vio/backend/VioBackEndParams.h:30:
/usr/local/include/Kimera-VIO/include/kimera-vio/dataprovider/DataProviderInterface-definitions.h:74:16: warning: 
      'VIO::VioNavStateTimestamped::equals' hides overloaded virtual function
      [-Woverloaded-virtual]
  virtual bool equals(const VioNavStateTimestamped& rhs) const;
               ^
/usr/local/include/Kimera-VIO/include/kimera-vio/dataprovider/DataProviderInterface-definitions.h:54:16: note: 
      hidden overloaded virtual function 'VIO::VioNavState::equals' declared
      here: type mismatch at 1st parameter ('const VIO::VioNavState &' vs 'const
      VIO::VioNavStateTimestamped &')
  virtual bool equals(const VioNavState& rhs) const;
               ^
In file included from /usr/local/include/Kimera-VIO/src/pipeline/Pipeline.cpp:15:
In file included from /usr/local/include/Kimera-VIO/include/kimera-vio/pipeline/Pipeline.h:25:
In file included from /usr/local/include/Kimera-VIO/include/kimera-vio/backend/VioBackEnd-definitions.h:29:
In file included from /usr/local/include/Kimera-VIO/include/kimera-vio/frontend/StereoVisionFrontEnd-definitions.h:19:
In file included from /usr/local/include/Kimera-VIO/include/kimera-vio/frontend/Tracker.h:32:
In file included from /usr/local/include/Kimera-VIO/include/kimera-vio/frontend/FeatureSelector.h:36:
/usr/local/include/Kimera-VIO/include/kimera-vio/frontend/VioFrontEndParams.h:41:9: warning: 
      field 'featureSelectionHorizon_' will be initialized after field
      'featureSelectionNrCornersToSelect_' [-Wreorder]
        featureSelectionHorizon_(3),  // in seconds
        ^
In file included from /usr/local/include/Kimera-VIO/src/pipeline/Pipeline.cpp:15:
In file included from /usr/local/include/Kimera-VIO/include/kimera-vio/pipeline/Pipeline.h:33:
/usr/local/include/Kimera-VIO/include/kimera-vio/loopclosure/LoopClosureDetector.h:26:10: fatal error: 
      'opencv/cv.hpp' file not found
#include <opencv/cv.hpp>
         ^~~~~~~~~~~~~~~
2 warnings and 1 error generated.
make[2]: *** [CMakeFiles/kimera_vio.dir/src/pipeline/Pipeline.cpp.o] Error 1
make[1]: *** [CMakeFiles/kimera_vio.dir/all] Error 2
make: *** [all] Error 2

xin-peng, 2020-03-10:
**Description:**
Hello, sorry to bother again. I install opencv-4.2.0 and corresponding contribute, and run c++ 11 support by 
**Command:**
```
cmake -DCMAKE_CXX_FLAGS:STRING=-std=c++11 ..
make -j4
```
**Console output:**
```
[  1%] Building CXX object external/googletest-build/googletest/CMakeFiles/gtest.dir/src/gtest-all.cc.o
[  2%] Building CXX object CMakeFiles/kimera_vio.dir/src/dataprovider/DataProviderInterface-definitions.cpp.o
[  3%] Building CXX object CMakeFiles/kimera_vio.dir/src/dataprovider/DataProviderInterface.cpp.o
[  5%] Building CXX object CMakeFiles/kimera_vio.dir/src/dataprovider/DataProviderModule.cpp.o
In file included from /usr/local/include/Kimera-VIO/src/dataprovider/DataProviderInterface-definitions.cpp:14:
/usr/local/include/Kimera-VIO/include/kimera-vio/dataprovider/DataProviderInterface-definitions.h:74:16: warning: 
      'VIO::VioNavStateTimestamped::equals' hides overloaded virtual function
      [-Woverloaded-virtual]
  virtual bool equals(const VioNavStateTimestamped& rhs) const;
               ^
/usr/local/include/Kimera-VIO/include/kimera-vio/dataprovider/DataProviderInterface-definitions.h:54:16: note: 
      hidden overloaded virtual function 'VIO::VioNavState::equals' declared
      here: type mismatch at 1st parameter ('const VIO::VioNavState &' vs 'const
      VIO::VioNavStateTimestamped &')
  virtual bool equals(const VioNavState& rhs) const;
               ^
[  6%] Linking CXX static library ../../../lib/libgtest.a
[  6%] Built target gtest
1 warning generated.
[  7%] Building CXX object CMakeFiles/kimera_vio.dir/src/dataprovider/EurocDataProvider.cpp.o
[  8%] Building CXX object CMakeFiles/kimera_vio.dir/src/frontend/CameraParams.cpp.o
In file included from /usr/local/include/Kimera-VIO/src/dataprovider/DataProviderModule.cpp:16:
In file included from /usr/local/include/Kimera-VIO/include/kimera-vio/dataprovider/DataProviderModule.h:23:
In file included from /usr/local/include/Kimera-VIO/include/kimera-vio/frontend/StereoImuSyncPacket.h:22:
In file included from /usr/local/include/Kimera-VIO/include/kimera-vio/backend/VioBackEnd-definitions.h:29:
In file included from /usr/local/include/Kimera-VIO/include/kimera-vio/frontend/StereoVisionFrontEnd-definitions.h:19:
In file included from /usr/local/include/Kimera-VIO/include/kimera-vio/frontend/Tracker.h:32:
In file included from /usr/local/include/Kimera-VIO/include/kimera-vio/frontend/FeatureSelector.h:33:
In file included from /usr/local/include/Kimera-VIO/include/kimera-vio/backend/VioBackEndParams.h:30:
/usr/local/include/Kimera-VIO/include/kimera-vio/dataprovider/DataProviderInterface-definitions.h:74:16: warning: 
      'VIO::VioNavStateTimestamped::equals' hides overloaded virtual function
      [-Woverloaded-virtual]
  virtual bool equals(const VioNavStateTimestamped& rhs) const;
               ^
/usr/local/include/Kimera-VIO/include/kimera-vio/dataprovider/DataProviderInterface-definitions.h:54:16: note: 
      hidden overloaded virtual function 'VIO::VioNavState::equals' declared
      here: type mismatch at 1st parameter ('const VIO::VioNavState &' vs 'const
      VIO::VioNavStateTimestamped &')
  virtual bool equals(const VioNavState& rhs) const;
               ^
In file included from /usr/local/include/Kimera-VIO/src/dataprovider/DataProviderModule.cpp:16:
In file included from /usr/local/include/Kimera-VIO/include/kimera-vio/dataprovider/DataProviderModule.h:23:
In file included from /usr/local/include/Kimera-VIO/include/kimera-vio/frontend/StereoImuSyncPacket.h:22:
In file included from /usr/local/include/Kimera-VIO/include/kimera-vio/backend/VioBackEnd-definitions.h:29:
In file included from /usr/local/include/Kimera-VIO/include/kimera-vio/frontend/StereoVisionFrontEnd-definitions.h:19:
In file included from /usr/local/include/Kimera-VIO/include/kimera-vio/frontend/Tracker.h:32:
In file included from /usr/local/include/Kimera-VIO/include/kimera-vio/frontend/FeatureSelector.h:36:
/usr/local/include/Kimera-VIO/include/kimera-vio/frontend/VioFrontEndParams.h:41:9: warning: 
      field 'featureSelectionHorizon_' will be initialized after field
      'featureSelectionNrCornersToSelect_' [-Wreorder]
        featureSelectionHorizon_(3),  // in seconds
        ^
In file included from /usr/local/include/Kimera-VIO/src/dataprovider/DataProviderInterface.cpp:15:
In file included from /usr/local/include/Kimera-VIO/include/kimera-vio/dataprovider/DataProviderInterface.h:21:
In file included from /usr/local/include/Kimera-VIO/include/kimera-vio/frontend/StereoImuSyncPacket.h:22:
In file included from /usr/local/include/Kimera-VIO/include/kimera-vio/backend/VioBackEnd-definitions.h:29:
In file included from /usr/local/include/Kimera-VIO/include/kimera-vio/frontend/StereoVisionFrontEnd-definitions.h:19:
In file included from /usr/local/include/Kimera-VIO/include/kimera-vio/frontend/Tracker.h:32:
In file included from /usr/local/include/Kimera-VIO/include/kimera-vio/frontend/FeatureSelector.h:33:
In file included from /usr/local/include/Kimera-VIO/include/kimera-vio/backend/VioBackEndParams.h:30:
/usr/local/include/Kimera-VIO/include/kimera-vio/dataprovider/DataProviderInterface-definitions.h:74:16: warning: 
[... 964 lines cut ...]
In file included from /usr/local/include/Kimera-VIO/include/kimera-vio/backend/VioBackEnd-definitions.h:29:
In file included from /usr/local/include/Kimera-VIO/include/kimera-vio/frontend/StereoVisionFrontEnd-definitions.h:19:
[... cut ...]
      private field 'open_file_in_append_mode' is not used
      [-Wunused-private-field]
  const bool open_file_in_append_mode = false;
             ^
/usr/local/include/Kimera-VIO/include/kimera-vio/logging/Logger.h:90:10: warning: 
      private field 'timing_loggerBackend_' is not used [-Wunused-private-field]
  double timing_loggerBackend_;
         ^
4 warnings generated.
make[1]: *** [CMakeFiles/kimera_vio.dir/all] Error 2
make: *** [all] Error 2
```
Is it possible for me to get any help? I am sorry if I ask a stupid question...

ToniRV, 2020-03-11:
I think the easiest would be to use opencv 3.3.1
Can you perhaps print smaller compilation output, just with the error?
Usually, when you try to compile the second time, most of the warnings are gone, and only the errors appear.

xin-peng, 2020-03-11:
Thanks for your reply. I compile again. Here is the result:
**Console output:**
```
[  2%] Built target gtest
[  3%] Building CXX object CMakeFiles/kimera_vio.dir/src/loopclosure/LoopClosureDetector.cpp.o
In file included from /usr/local/include/Kimera-VIO/src/loopclosure/LoopClosureDetector.cpp:34:
In file included from /usr/local/include/Kimera-VIO/include/kimera-vio/loopclosure/LoopClosureDetector.h:33:
In file included from /usr/local/include/Kimera-VIO/include/kimera-vio/logging/Logger.h:23:
In file included from /usr/local/include/Kimera-VIO/include/kimera-vio/backend/VioBackEnd-definitions.h:29:
In file included from /usr/local/include/Kimera-VIO/include/kimera-vio/frontend/StereoVisionFrontEnd-definitions.h:19:
In file included from /usr/local/include/Kimera-VIO/include/kimera-vio/frontend/Tracker.h:32:
In file included from /usr/local/include/Kimera-VIO/include/kimera-vio/frontend/FeatureSelector.h:33:
In file included from /usr/local/include/Kimera-VIO/include/kimera-vio/backend/VioBackEndParams.h:30:
/usr/local/include/Kimera-VIO/include/kimera-vio/dataprovider/DataProviderInterface-definitions.h:74:16: warning: 
      'VIO::VioNavStateTimestamped::equals' hides overloaded virtual function
      [-Woverloaded-virtual]
  virtual bool equals(const VioNavStateTimestamped& rhs) const;
               ^
/usr/local/include/Kimera-VIO/include/kimera-vio/dataprovider/DataProviderInterface-definitions.h:54:16: note: 
      hidden overloaded virtual function 'VIO::VioNavState::equals' declared
      here: type mismatch at 1st parameter ('const VIO::VioNavState &' vs 'const
      VIO::VioNavStateTimestamped &')
  virtual bool equals(const VioNavState& rhs) const;
               ^
In file included from /usr/local/include/Kimera-VIO/src/loopclosure/LoopClosureDetector.cpp:34:
In file included from /usr/local/include/Kimera-VIO/include/kimera-vio/loopclosure/LoopClosureDetector.h:33:
In file included from /usr/local/include/Kimera-VIO/include/kimera-vio/logging/Logger.h:23:
In file included from /usr/local/include/Kimera-VIO/include/kimera-vio/backend/VioBackEnd-definitions.h:29:
In file included from /usr/local/include/Kimera-VIO/include/kimera-vio/frontend/StereoVisionFrontEnd-definitions.h:19:
In file included from /usr/local/include/Kimera-VIO/include/kimera-vio/frontend/Tracker.h:32:
In file included from /usr/local/include/Kimera-VIO/include/kimera-vio/frontend/FeatureSelector.h:36:
/usr/local/include/Kimera-VIO/include/kimera-vio/frontend/VioFrontEndParams.h:41:9: warning: 
      field 'featureSelectionHorizon_' will be initialized after field
      'featureSelectionNrCornersToSelect_' [-Wreorder]
        featureSelectionHorizon_(3),  // in seconds
        ^
/usr/local/include/Kimera-VIO/src/loopclosure/LoopClosureDetector.cpp:83:43: error: 
      cannot initialize a parameter of type 'ORB::ScoreType' with an lvalue of
      type 'int'
                                          lcd_params_.score_type_,
                                          ^~~~~~~~~~~~~~~~~~~~~~~
/usr/local/include/opencv4/opencv2/features2d.hpp:351:55: note: passing argument
      to parameter 'scoreType' here
        int firstLevel=0, int WTA_K=2, ORB::ScoreType scoreType=ORB::HAR...
                                                      ^
/usr/local/include/Kimera-VIO/src/loopclosure/LoopClosureDetector.cpp:89:7: error: 
      no matching function for call to 'create'
      cv::DescriptorMatcher::create(lcd_params_.matcher_type_);
      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/local/include/opencv4/opencv2/features2d.hpp:1048:43: note: candidate
      function not viable: no known conversion from 'int' to 'const cv::String'
      (aka 'const basic_string<char, char_traits<char>, allocator<char> >') for
      1st argument
    CV_WRAP static Ptr<DescriptorMatcher> create( const String& descript...
                                          ^
/usr/local/include/opencv4/opencv2/features2d.hpp:1050:43: note: candidate
      function not viable: no known conversion from 'int' to 'const
      DescriptorMatcher::MatcherType' for 1st argument
    CV_WRAP static Ptr<DescriptorMatcher> create( const DescriptorMatche...
                                          ^
In file included from /usr/local/include/Kimera-VIO/src/loopclosure/LoopClosureDetector.cpp:34:
In file included from /usr/local/include/Kimera-VIO/include/kimera-vio/loopclosure/LoopClosureDetector.h:37:
In file included from /usr/local/include/Kimera-VIO/include/kimera-vio/pipeline/PipelineModule.h:28:
In file included from /usr/local/include/Kimera-VIO/include/kimera-vio/pipeline/QueueSynchronizer.h:24:
/usr/local/include/Kimera-VIO/include/kimera-vio/utils/ThreadsafeQueue.h:208:7: warning: 
      field 'mutex_' will be initialized after field 'queue_id_' [-Wreorder]
    : mutex_(),
      ^
/usr/local/include/Kimera-VIO/include/kimera-vio/utils/ThreadsafeQueue.h:216:7: note: 
      in instantiation of member function
[... 10 lines cut ...]
In file included from /usr/local/include/Kimera-VIO/src/loopclosure/LoopClosureDetector.cpp:34:
In file included from /usr/local/include/Kimera-VIO/include/kimera-vio/loopclosure/LoopClosureDetector.h:37:
In file included from /usr/local/include/Kimera-VIO/include/kimera-vio/pipeline/PipelineModule.h:28:
In file included from /usr/local/include/Kimera-VIO/include/kimera-vio/pipeline/QueueSynchronizer.h:24:
/usr/local/include/Kimera-VIO/include/kimera-vio/utils/ThreadsafeQueue.h:208:7: warning: 
[... cut ...]
/usr/local/include/Kimera-VIO/include/kimera-vio/loopclosure/LoopClosureDetector.h:437:9: note: 
      in instantiation of member function
      'VIO::ThreadsafeQueue<std::__1::shared_ptr<VIO::BackendOutput>
      >::ThreadsafeQueue' requested here
        backend_queue_("lcd_backend_queue"),
        ^
4 warnings and 2 errors generated.
make[2]: *** [CMakeFiles/kimera_vio.dir/src/loopclosure/LoopClosureDetector.cpp.o] Error 1
make[1]: *** [CMakeFiles/kimera_vio.dir/all] Error 2
make: *** [all] Error 2
```

ToniRV, 2020-03-11:
I see! Ok, let me ping @marcusabate, but this seems to be an incompatibility with opencv versions.
We will have to spend some time to support both versions which I am afraid is not in our top list of priorities. Would you mind proposing a fix/pull request? Otw, the best would be to downgrade your opencv version to 3.3.1

xin-peng, 2020-03-11:
Thanks, I will try to install opencv 3.3.1

closed 2023-11-03 by marcusabate