MIT-SPARK/Kimera-VIO #36: How to run in Kitti dataset

michaelnguyen11, 2020-03-10:
Hi authors,

Thank you, guys for awesome project.

I run Kimera-VIO in Euroc's dataset, everything is okay, but when run in Kitti dataset, PipelineModule stuck at Module: Visualizer - Spinning. Other spin modules didn't start.
```
I0310 17:10:02.398468 12352 KittiDataProvider.cpp:530] -------------------------------------------------------------
-------------------------------------------------------------
-------------------------------------------------------------
I0310 17:10:02.399873 12352 RegularVioBackEnd.cpp:113] Using Regular VIO backend.
I0310 17:10:02.402761 12352 LoopClosureDetector.cpp:98] LoopClosureDetector:: Loading vocabulary from ../vocabulary/ORBvoc.yml
I0310 17:10:10.057924 12352 LoopClosureDetector.cpp:101] Loaded vocabulary with 971815 visual words.
I0310 17:10:10.170249 12404 PipelineModule.h:163] Module: Data Provider - Spinning.
I0310 17:10:10.170296 12405 Pipeline.cpp:348] Shutting down VIO pipeline once processing has finished.
I0310 17:10:10.179666 12352 PipelineModule.h:163] Module: Visualizer - Spinning.
```
I tested the [Kimera-VIO repository](https://github.com/ToniRV/Kimera-VIO) of @ToniRV , example demo works well in Kitti dataset.

Could you help me run the Kimera-VIO in Kitti dataset ? Many thanks in advance !

Regards,
Michael

ToniRV, 2020-03-11:
Hi @michaelnguyen11, we don't support Kitti dataset right now as we had some issues using their IMU data (it is very sparse, one measurement per GPS msg or something on this range).
One could interpolate or fake IMU messages between GPS positions but we didn't do that in the end.
Also, mind that Kimera has had some heavy refactoring so the Kitti data provider might be outdated.

closed 2020-03-11 by ToniRV

michaelnguyen11, 2020-03-11:
For anyone who want to run the demo in KITTI-raw dataset:
- Based on EurocDatasetProvider, I created this patch as a workaround to run in Kitti dataset. (note: change to patch file)
[kimera_vio_and_kitti.txt](https://github.com/MIT-SPARK/Kimera-VIO/files/4317197/kimera_vio_and_kitti.txt)

The demo works well with Kitti-raw resident video 2011_09_26_drive_0023.

omarosamahu, 2020-03-31:
@michaelnguyen11  I tried to apply the patch you provided for kitti but I got the following err 
error: patch failed: src/dataprovider/KittiDataProvider.cpp:38
error: src/dataprovider/KittiDataProvider.cpp: patch does not apply

Boyuan-Tian, 2022-06-22:
Hi @michaelnguyen11, just out of curiosity, how did you initialize Kimera on KITTI?
Is that sequence begins with static data that allows static IMU initialization?
If not, how did you specify the bias values when initialized using the ground truth file?
Thanks a lot!