MIT-SPARK/Kimera-VIO #37: Infinite loop when no IMU data between two frames

ruffsl, 2020-03-10:
**Description:**

As firsted pointed out in https://github.com/MIT-SPARK/Kimera-VIO-ROS2/pull/5#discussion_r386776118 , it seems the Kimera-VIO is liable to fall into an infinite loop when no IMU data between two frames is received. Such can occur when IMU messages are lost or dropped by the middleware QoS, or when the sample rates from the IMU and Stereo frames happened to alias such that two Stereo frames are timestamped between two consecutively received IMU samples.

**Command:**

See launch command and context from linked PR:

```
ros2 launch kimera_vio_ros kimera_ros_realsense.launch.xml 
```
 https://github.com/MIT-SPARK/Kimera-VIO-ROS2/pull/5#discussion_r386776118

**Console output:**
```
$ ros2 launch kimera_vio_ros kimera_ros_realsense.launch.xml 
[INFO] [launch]: All log files can be found below /home/ruffsl/.ros/log/2020-03-06-11-59-33-116779-cog-rw07-6169
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [kimera_vio_ros-1]: process started with pid [6180]
[kimera_vio_ros-1] I0306 11:59:33.421044  6180 CameraParams.cpp:31] Parsing camera parameters for: left_cam
[kimera_vio_ros-1] I0306 11:59:33.582267  6180 CameraParams.cpp:31] Parsing camera parameters for: right_cam
[kimera_vio_ros-1] I0306 11:59:33.582564  6180 DataProviderInterface.cpp:96] Running dataset between frame 50 and frame 10000
[kimera_vio_ros-1] I0306 11:59:33.589473  6180 RegularVioBackEnd.cpp:113] Using Regular VIO backend.
[kimera_vio_ros-1] I0306 11:59:33.596930  6194 PipelineModule.h:163] Module: Data Provider - Spinning.
[kimera_vio_ros-1] W0306 11:59:33.672981  6194 DataProviderModule.cpp:97] No IMU data from last frame timestamp: 1583524773595843072 to timestamp: 1583524773629287936
[kimera_vio_ros-1] W0306 11:59:33.673069  6194 DataProviderModule.cpp:97] No IMU data from last frame timestamp: 1583524773595843072 to timestamp: 1583524773629287936
[kimera_vio_ros-1] W0306 11:59:33.673132  6194 DataProviderModule.cpp:97] No IMU data from last frame timestamp: 1583524773595843072 to timestamp: 1583524773629287936
[kimera_vio_ros-1] W0306 11:59:33.673167  6194 DataProviderModule.cpp:97] No IMU data from last frame timestamp: 1583524773595843072 to timestamp: 1583524773629287936
[kimera_vio_ros-1] W0306 11:59:33.673197  6194 DataProviderModule.cpp:97] No IMU data from last frame timestamp: 1583524773595843072 to timestamp: 1583524773629287936
[kimera_vio_ros-1] W0306 11:59:33.673235  6194 DataProviderModule.cpp:97] No IMU data from last frame timestamp: 1583524773595843072 to timestamp: 1583524773629287936
[kimera_vio_ros-1] W0306 11:59:33.673280  6194 DataProviderModule.cpp:97] No IMU data from last frame timestamp: 1583524773595843072 to timestamp: 1583524773629287936
[kimera_vio_ros-1] W0306 11:59:33.673322  6194 DataProviderModule.cpp:97] No IMU data from last frame timestamp: 1583524773595843072 to timestamp: 1583524773629287936
[kimera_vio_ros-1] W0306 11:59:33.673369  6194 DataProviderModule.cpp:97] No IMU data from last frame timestamp: 1583524773595843072 to timestamp: 1583524773629287936
[kimera_vio_ros-1] W0306 11:59:33.673408  6194 DataProviderModule.cpp:97] No IMU data from last frame timestamp: 1583524773595843072 to timestamp: 1583524773629287936
[kimera_vio_ros-1] W0306 11:59:33.673439  6194 DataProviderModule.cpp:97] No IMU data from last frame timestamp: 1583524773595843072 to timestamp: 1583524773629287936
```

The infinite looping is encountered most of the time, but something one can get lucky with the race condition and avoid it for a few seconds:

<details>

```
$ ros2 launch kimera_vio_ros kimera_ros_realsense.launch.xml 
[INFO] [launch]: All log files can be found below /home/ruffsl/.ros/log/2020-03-06-11-44-36-116764-cog-rw07-4477
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [kimera_vio_ros-1]: process started with pid [4490]
[kimera_vio_ros-1] I0306 11:44:36.403184  4490 CameraParams.cpp:31] Parsing camera parameters for: left_cam
[kimera_vio_ros-1] I0306 11:44:36.537226  4490 CameraParams.cpp:31] Parsing camera parameters for: right_cam
[kimera_vio_ros-1] I0306 11:44:36.537469  4490 DataProviderInterface.cpp:96] Running dataset between frame 50 and frame 10000
[kimera_vio_ros-1] I0306 11:44:36.542948  4490 RegularVioBackEnd.cpp:113] Using Regular VIO backend.
[kimera_vio_ros-1] I0306 11:44:36.551772  4504 PipelineModule.h:163] Module: Data Provider - Spinning.
[kimera_vio_ros-1] I0306 11:44:36.733357  4504 Pipeline.cpp:780] Frontend launched (parallel_run set to 1).
[kimera_vio_ros-1] I0306 11:44:36.733377  4544 PipelineModule.h:163] Module: VioFrontEnd - Spinning.
[kimera_vio_ros-1] I0306 11:44:36.733757  4504 Pipeline.cpp:511] ------------------- Initialize Pipeline with frame k = 1--------------------
[kimera_vio_ros-1] W0306 11:44:36.733768  4504 InitializationFromImu.cpp:26] InitializationFromImu: assumes that the vehicle is stationary and upright along some axis,and gravity vector is along a single axis!
[kimera_vio_ros-1] I0306 11:44:36.802474  4504 VioBackEnd.cpp:217] Initial state seed: 
[kimera_vio_ros-1]  - Initial pose: 
[kimera_vio_ros-1] |0.999984, -0.0027797, 0.0049191|
[kimera_vio_ros-1] |-0.0027797, 0.515936, 0.856622|
[kimera_vio_ros-1] |-0.0049191, -0.856622, 0.51592|
[kimera_vio_ros-1] 
[kimera_vio_ros-1] [0, 0, 0]';
[kimera_vio_ros-1] 
[kimera_vio_ros-1]  - Initial vel: 0 0 0
[kimera_vio_ros-1]  - Initial IMU bias: acc = [-0.00123735, -0.215474, 0.129774]' gyro = [-0.00703574, -2.63912e-05, 6.80661e-06]'
[kimera_vio_ros-1] I0306 11:44:36.807591  4504 Pipeline.cpp:288] Before launching threads.
[kimera_vio_ros-1] I0306 11:44:36.807698  4504 Pipeline.cpp:809] Backend, mesher and visualizer launched (parallel_run set to 1).
[kimera_vio_ros-1] I0306 11:44:36.807705  4504 Pipeline.cpp:290]  launching threads.
[kimera_vio_ros-1] I0306 11:44:36.807698  4564 PipelineModule.h:163] Module: VioBackEnd - Spinning.
[kimera_vio_ros-1] I0306 11:44:36.807870  4565 PipelineModule.h:163] Module: MesherModule - Spinning.
[... 56 lines cut ...]
[kimera_vio_ros-1] W0306 11:44:37.634687  4544 PipelineModule.h:186] Module: VioFrontEnd - frequency: 166.667 Hz. (6 ms).
[kimera_vio_ros-1] I0306 11:44:37.656633  4544 StereoVisionFrontEnd.cpp:62] ------------------- Processing frame k = 23--------------------
[kimera_vio_ros-1] W0306 11:44:37.661800  4544 PipelineModule.h:186] Module: VioFrontEnd - frequency: 200 Hz. (5 ms).
[kimera_vio_ros-1] I0306 11:44:37.689841  4544 StereoVisionFrontEnd.cpp:62] ------------------- Processing frame k = 24--------------------
[kimera_vio_ros-1] I0306 11:44:37.730131  4544 StereoVisionFrontEnd.cpp:124] Keyframe 24 with: 461 smart measurements
[kimera_vio_ros-1] W0306 11:44:37.730195  4544 PipelineModule.h:186] Module: VioFrontEnd - frequency: 25 Hz. (40 ms).
[kimera_vio_ros-1] I0306 11:44:37.730213  4544 StereoVisionFrontEnd.cpp:62] ------------------- Processing frame k = 25--------------------
```

</details>

**Additional files:**

For reference, I'm using an updated realsense package for ROS2:
https://github.com/ruffsl/ros2_intel_realsense/tree/devel
(The repo's devel branch includes a Dockerfile for how to build)
And the Kimera-VIO-ROS2 branch for the current work in progress PR: 
https://github.com/MIT-SPARK/Kimera-VIO-ROS2/pull/5

* OS: Ubuntu 18.04
* Kimera-VIO: bceed7203f824c1bcffc0c195a186d13c5ea2b22
* Downstream Libraries: See `.repo` files in respective fork/repo
* Source code changed: yes
 * Just trivial build system fixes
 * https://github.com/MIT-SPARK/Kimera-VIO/compare/master...ruffsl:patch-3

ToniRV, 2020-03-10:
About to merge fix

ToniRV, 2020-03-11:
Ok, fix merged. Let us know if it works for you @ruffsl 
Also, this issue is related: https://github.com/MIT-SPARK/Kimera-VIO-ROS/issues/49

ruffsl, 2020-03-13:
I've rebased off the latest push to the public repo, and no longer encounter this issue. Though the diff from the recent set of commits pushed to the public repo is rather large so I couldn't easily grok which exact changes may have fixed this. I'd might suggest making more regular steady syncs to the open repos, as large infrequent pushes of commits can dissuade external collaboration.

closed 2020-03-13 by ruffsl

ToniRV, 2020-03-13:
Thank you @ruffsl, I totally agree! We should make smaller PRs and encourage regular syncs.
The semester is turning out to be a bit hectic.