MIT-SPARK/Kimera-VIO #38: Error when running Kimera-VIO in Euroc's dataset

Kecheng-Chen, 2020-03-12:
**Description:**

**Command:**
```
bash ./scripts/stereoVIOEuroc.bash -p "$HOME/Euroc/V1_01_easy"
```

**Console output:**
```
Using dataset at path: /home/kechengchen/Euroc/V1_01_easy
 Launching:

            ██╗  ██╗██╗███╗   ███╗███████╗██████╗  █████╗
            ██║ ██╔╝██║████╗ ████║██╔════╝██╔══██╗██╔══██╗
            █████╔╝ ██║██╔████╔██║█████╗  ██████╔╝███████║
            ██╔═██╗ ██║██║╚██╔╝██║██╔══╝  ██╔══██╗██╔══██║
            ██║  ██╗██║██║ ╚═╝ ██║███████╗██║  ██║██║  ██║
            ╚═╝  ╚═╝╚═╝╚═╝     ╚═╝╚══════╝╚═╝  ╚═╝╚═╝  ╚═╝

 
*** Error in `../build/stereoVIOEuroc': double free or corruption (out): 0x000000000192d540 ***
======= Backtrace: =========
/lib/x86_64-linux-gnu/libc.so.6(+0x777e5)[0x7efd9ee257e5]
/lib/x86_64-linux-gnu/libc.so.6(+0x8037a)[0x7efd9ee2e37a]
/lib/x86_64-linux-gnu/libc.so.6(cfree+0x4c)[0x7efd9ee3253c]
/usr/local/lib/libgtsam.so.4(_ZN5gtsam10noiseModel8Diagonal6SigmasERKN5Eigen6MatrixIdLin1ELi1ELi0ELin1ELi1EEEb+0x18d)[0x7efd9fb5fa5d]
/usr/local/lib/libgtsam.so.4(+0x113dad)[0x7efd9fa27dad]
/lib64/ld-linux-x86-64.so.2(+0x106ca)[0x7efda196e6ca]
/lib64/ld-linux-x86-64.so.2(+0x107db)[0x7efda196e7db]
/lib64/ld-linux-x86-64.so.2(+0xc6a)[0x7efda195ec6a]
======= Memory map: ========
00400000-00416000 r-xp 00000000 08:08 173546                             /home/kechengchen/Kimera-VIO/build/stereoVIOEuroc
00615000-00616000 r--p 00015000 08:08 173546                             /home/kechengchen/Kimera-VIO/build/stereoVIOEuroc
00616000-00617000 rw-p 00016000 08:08 173546                             /home/kechengchen/Kimera-VIO/build/stereoVIOEuroc
018c4000-01948000 rw-p 00000000 00:00 0                                  [heap]
7efd70000000-7efd70021000 rw-p 00000000 00:00 0 
7efd70021000-7efd74000000 ---p 00000000 00:00 0 
7efd753a1000-7efd755a1000 rw-p 00000000 00:00 0 
7efd755a1000-7efd76e57000 r-xp 00000000 08:07 666626                     /usr/lib/x86_64-linux-gnu/libicudata.so.55.1
7efd76e57000-7efd77056000 ---p 018b6000 08:07 666626                     /usr/lib/x86_64-linux-gnu/libicudata.so.55.1
7efd77056000-7efd77057000 r--p 018b5000 08:07 666626                     /usr/lib/x86_64-linux-gnu/libicudata.so.55.1
7efd77057000-7efd77058000 rw-p 018b6000 08:07 666626                     /usr/lib/x86_64-linux-gnu/libicudata.so.55.1
7efd77058000-7efd77061000 r-xp 00000000 08:07 1185238                    /lib/x86_64-linux-gnu/libcrypt-2.23.so
7efd77061000-7efd77260000 ---p 00009000 08:07 1185238                    /lib/x86_64-linux-gnu/libcrypt-2.23.so
7efd77260000-7efd77261000 r--p 00008000 08:07 1185238                    /lib/x86_64-linux-gnu/libcrypt-2.23.so
7efd77261000-7efd77262000 rw-p 00009000 08:07 1185238                    /lib/x86_64-linux-gnu/libcrypt-2.23.so
7efd77262000-7efd77290000 rw-p 00000000 00:00 0 
7efd77290000-7efd772d7000 r-xp 00000000 08:07 666607                     /usr/lib/x86_64-linux-gnu/libhx509.so.5.0.0
7efd772d7000-7efd774d6000 ---p 00047000 08:07 666607                     /usr/lib/x86_64-linux-gnu/libhx509.so.5.0.0
7efd774d6000-7efd774d8000 r--p 00046000 08:07 666607                     /usr/lib/x86_64-linux-gnu/libhx509.so.5.0.0
7efd774d8000-7efd774da000 rw-p 00048000 08:07 666607                     /usr/lib/x86_64-linux-gnu/libhx509.so.5.0.0
7efd774da000-7efd774db000 rw-p 00000000 00:00 0 
7efd774db000-7efd774e9000 r-xp 00000000 08:07 666589                     /usr/lib/x86_64-linux-gnu/libheimbase.so.1.0.0
7efd774e9000-7efd776e8000 ---p 0000e000 08:07 666589                     /usr/lib/x86_64-linux-gnu/libheimbase.so.1.0.0
7efd776e8000-7efd776e9000 r--p 0000d000 08:07 666589                     /usr/lib/x86_64-linux-gnu/libheimbase.so.1.0.0
7efd776e9000-7efd776ea000 rw-p 0000e000 08:07 666589                     /usr/lib/x86_64-linux-gnu/libheimbase.so.1.0.0
7efd776ea000-7efd77711000 r-xp 00000000 08:07 667270                     /usr/lib/x86_64-linux-gnu/libwind.so.0.0.0
7efd77711000-7efd77911000 ---p 00027000 08:07 667270                     /usr/lib/x86_64-linux-gnu/libwind.so.0.0.0
7efd77911000-7efd77912000 r--p 00027000 08:07 667270                     /usr/lib/x86_64-linux-gnu/libwind.so.0.0.0
7efd77912000-7efd77913000 rw-p 00028000 08:07 667270                     /usr/lib/x86_64-linux-gnu/libwind.so.0.0.0
7efd77913000-7efd7791c000 r-xp 00000000 08:07 666726                     /usr/lib/x86_64-linux-gnu/libltdl.so.7.3.1
7efd7791c000-7efd77b1b000 ---p 00009000 08:07 666726                     /usr/lib/x86_64-linux-gnu/libltdl.so.7.3.1
7efd77b1b000-7efd77b1c000 r--p 00008000 08:07 666726                     /usr/lib/x86_64-linux-gnu/libltdl.so.7.3.1
7efd77b1c000-7efd77b1d000 rw-p 00009000 08:07 666726                     /usr/lib/x86_64-linux-gnu/libltdl.so.7.3.1
7efd77b1d000-7efd77b3e000 r-xp 00000000 08:07 666708                     /usr/lib/x86_64-linux-gnu/libgomp.so.1.0.0
7efd77b3e000-7efd77d3d000 ---p 00021000 08:07 666708                     /usr/lib/x86_64-linux-gnu/libgomp.so.1.0.0
7efd77d3d000-7efd77d3e000 r--p 00020000 08:07 666708                     /usr/lib/x86_64-linux-gnu/libgomp.so.1.0.0
7efd77d3e000-7efd77d3f000 rw-p 00021000 08:07 666708                     /usr/lib/x86_64-linux-gnu/libgomp.so.1.0.0
7efd77d3f000-7efd77ebe000 r-xp 00000000 08:07 666640                     /usr/lib/x86_64-linux-gnu/libicuuc.so.55.1
7efd77ebe000-7efd780be000 ---p 0017f000 08:07 666640                     /usr/lib/x86_64-linux-gnu/libicuuc.so.55.1
[... 659 lines cut ...]
7efd947ca000-7efd947cb000 r--p 00057000 08:07 657582                     /usr/lib/x86_64-linux-gnu/libjpeg.so.8.0.2
7efd947cb000-7efd947cc000 rw-p 00058000 08:07 657582                     /usr/lib/x86_64-linux-gnu/libjpeg.so.8.0.2
7efd947cc000-7efd948db000 r-xp 00000000 08:07 1185113                    /lib/x86_64-linux-gnu/libglib-2.0.so.0.4800.2
7efd948db000-7efd94ada000 ---p 0010f000 08:07 1185113                    /lib/x86_64-linux-gnu/libglib-2.0.so.0.4800.2
7efd94ada000-7efd94adb000 r--p 0010e000 08:07 1185113                    /lib/x86_64-linux-gnu/libglib-2.0.so.0.4800.2
7efd94adb000-7efd94adc000 rw-p 0010f000 08:07 1185113                    /lib/x86_64-linux-gnu/libglib-2.0.so.0.4800.2
7efd94adc000-7efd94add000 rw-p 00000000 00:00 0 
7efd94add000-7efd94b2f000 r-xp 00000000 08:07 655643                     /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0.4800.2
7efd94b2f000-7efd94d2e000 ---p 00052000 08:07 655643                     /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0.4800.2
7efd94d2e000-7efd94d2f000 r--p 00051000 08:07 655643                     /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0.4800.2
7efd94d2f000-7efd94d30000 rw-p 00052000 08:07 655643                     /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0.4800.2
7efd94d30000-7efd94d51000 r-xp 00000000 08:07 657573                     /usr/lib/x86_64-linux-gnu/libgdk_pixbuf-2./scripts/stereoVIOEuroc.bash: line 137: 14665 Aborted                 (core dumped) $BUILD_PATH/stereoVIOEuroc --dataset_type="$DATASET_TYPE" --dataset_path="$DATASET_PATH" --initial_k=50 --final_k=2000 --backend_type="$BACKEND_TYPE" --left_cam_params_path="$PARAMS_PATH/LeftCameraParams.yaml" --right_cam_params_path="$PARAMS_PATH/RightCameraParams.yaml" --imu_params_path="$PARAMS_PATH/ImuParams.yaml" --backend_params_path="$PARAMS_PATH/regularVioParameters.yaml" --frontend_params_path="$PARAMS_PATH/trackerParameters.yaml" --use_lcd="$USE_LCD" --lcd_params_path="$PARAMS_PATH/LCDParameters.yaml" --vocabulary_path="$VOCABULARY_PATH/ORBvoc.yml" --flagfile="$PARAMS_PATH/flags/stereoVIOEuroc.flags" --flagfile="$PARAMS_PATH/flags/Mesher.flags" --flagfile="$PARAMS_PATH/flags/VioBackEnd.flags" --flagfile="$PARAMS_PATH/flags/RegularVioBackEnd.flags" --flagfile="$PARAMS_PATH/flags/Visualizer3D.flags" --parallel_run="$PARALLEL_RUN" --logtostderr=1 --colorlogtostderr=1 --log_prefix=1 --v=0 --vmodule=Pipeline*=00 --log_output="$LOG_OUTPUT" --save_frontend_images=1 --visualize_frontend_images=1 --output_path="$OUTPUT_PATH"

```

**Additional files:**
Please attach all the files needed to reproduce the error.

Please give also the following information:
* SparkVio branch, tag or commit used
* GTSAM version used: 4.0
* OpenGV version used: 1.0
* OpenCV version used: 3.3.1
* Operating system and version (e.g. Ubuntu 16.04 or Windows 10): Ubuntu 16.04
* Did you change the source code? (yes / no): no

ToniRV, 2020-03-12:
Hi @Kecheng-Chen,
Thanks for raising the issue.
Which gtsam commit are you using?

Kecheng-Chen, 2020-03-12:
> Hi @Kecheng-Chen,
> Thanks for raising the issue.
> Which gtsam commit are you using?

Hi Thank you for the quick reply. The commit is ee069286b447ff58b809423cc77c777a02abdfe5.

ToniRV, 2020-03-12:
Interesting, Kimera-VIO should not be able to compile with this commit, can you try with:
`0c3e05f375c03c5ff5218e708db416b38f4113c8`
Thanks!

Kecheng-Chen, 2020-03-12:
> Interesting, Kimera-VIO should not be able to compile with this commit, can you try with:
> `0c3e05f375c03c5ff5218e708db416b38f4113c8`
> Thanks!

Thank you. I will try it.

hperik1, 2020-03-24:
@ToniRV , I am experiencing the same problem.
Hopefully the following will help ...

Here is my setup:
    **SparkVio** branch, tag or commit used: (19 march)
          commit: 5f4bc95fb00728aa41f051ac7bd4d2e4708d0154
    **GTSAM** version used: 4.0.2
          commit: 342f30d148fae84c92ff71705c9e50e0a3683bda
          reverted also to 4.0.0 commit: 0c3e05f375c03c5ff5218e708db416b38f4113c8
    **OpenGV** version used: 1.0
          commit: 306a54e6c6b94e2048f820cdf77ef5281d4b48ad
    **OpenCV** version used: 3.3.1
          commit: a871f9e4f7d83dc5851f965bdba5cd01bb7527fe
    **Operating system and version** (e.g. Ubuntu 16.04 or Windows 10): 
          Ubuntu 16.04 / amd64
    **Glog** version:    0.3.4-0.1
    **Gflags** version:  2.1.2-3
    **Gtest** version:    1.7.0-4ubuntu1
    **DBoW2** dorian3d
          commit: 4d08e9fc751fac9063874d85a43c1ccdcda8b401
    **Kimera-RPGO** (19 march)
          commit: 65ce225647fed0463796913c17fbf488167317ef
 
   **shell commands** from within _Kimera-VIO_:
     _cd scripts_
     _./stereoVIOEuroc.bash -lcd -p ~/benchmark/datasets/Euroc/V1_01_easy_ and for
     _./stereoVIOEuroc.bash -lcd -p ~/benchmark/datasets/Euroc/V1_01_easy/mav0_
   note: datasets are downloaded from google drive: euroc-spark/V1_01_easy

    Did you change the source code? (yes / no): yes
         Next change in Kimera-VIO was necessary to make it compile
         changed std::mt19937 into boost::mt19937 in line 85 of tests/testlogger.cpp

hperik1, 2020-03-24:
@ToniRV @Kecheng-Chen 
In addition to the previous comment, I ran the backtrace in gdb:

#0  0x00007ffff4857428 in __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:54
#1  0x00007ffff485902a in __GI_abort () at abort.c:89
#2  0x00007ffff48997ea in __libc_message (do_abort=do_abort@entry=2, 
    fmt=fmt@entry=0x7ffff49b2ed8 "*** Error in `%s': %s: 0x%s ***\n") at ../sysdeps/posix/libc_fatal.c:175
#3  0x00007ffff48a237a in malloc_printerr (ar_ptr=<optimized out>, ptr=<optimized out>, 
    str=0x7ffff49b2fe8 "double free or corruption (out)", action=3) at malloc.c:5006
#4  _int_free (av=<optimized out>, p=<optimized out>, have_lock=0) at malloc.c:3867
#5  0x00007ffff48a653c in __GI___libc_free (mem=<optimized out>) at malloc.c:2968
#6  0x00007ffff561595d in gtsam::noiseModel::Diagonal::Sigmas(Eigen::Matrix<double, -1, 1, 0, -1, 1> const&, bool) ()
   from /usr/local/lib/libgtsam.so.4
#7  0x00007ffff549cb9d in _GLOBAL__sub_I_lago.cpp () from /usr/local/lib/libgtsam.so.4
#8  0x00007ffff7de76ca in call_init (l=<optimized out>, argc=argc@entry=29, argv=argv@entry=0x7fffffffd938, 
    env=env@entry=0x7fffffffda28) at dl-init.c:72
#9  0x00007ffff7de77db in call_init (env=0x7fffffffda28, argv=0x7fffffffd938, argc=29, l=<optimized out>) at dl-init.c:30
#10 _dl_init (main_map=0x7ffff7ffe168, argc=29, argv=0x7fffffffd938, env=0x7fffffffda28) at dl-init.c:120
#11 0x00007ffff7dd7c6a in _dl_start_user () from /lib64/ld-linux-x86-64.so.2
#12 0x000000000000001d in ?? ()
etc...

hperik1, 2020-03-24:
@ToniRV @Kecheng-Chen 
one more update ... 
I have rebuild _gtsam_ using the following _cmake_ arguments:

`    cmake -DCMAKE_INSTALL_PREFIX=/usr/local -DCMAKE_BUILD_TYPE=Release -DGTSAM_USE_SYSTEM_EIGEN=OFF -DGTSAM_POSE3_EXPMAP=ON -DGTSAM_ROT3_EXPMAP=ON -DGTSAM_WITH_TBB=ON **-DGTSAM_BUILD_WITH_MARCH_NATIVE=ON** ..`

Specially the _march_native_ was set to ON.

Consequently _Kimera-RPGO_ and _Kimera-VIO_ had to be rebuild as well.

It start running, however I get a lot of 'yellow' warnings ... may I assume this normal ?

ToniRV, 2020-03-24:
Warnings should be fine! Especially the timing ones, I just find them useful to check that modules are working.

hperik1, 2020-04-02:
@ToniRV, in addition it throws a reproducable OpenCV errors pretty much to the end of the EuRoc V1_01_easy dataset:

`OpenCV Error: One of arguments' values is out of range () in locate, file /tmp/binarydeb/ros-kinetic-opencv3-3.3.1/modules/imgproc/src/subdivision2d.cpp, line 288`

Is this a single point issue which can be omitted?

Herewith a part of the std output logging: [RunTimeError_200402.log](https://github.com/MIT-SPARK/Kimera-VIO/files/4420808/RunTimeError_200402.log)

closed 2023-11-03 by marcusabate