MIT-SPARK/Kimera-VIO #40: Out-door path failure

fouad1995, 2020-03-16:
**Description:**
I am using ZED-2 Camera with its internal imu , and trying to record outdoor but the path is not correct 

**Command:**
```
#TRACKER PARAMETERS
klt_win_size: 24
klt_max_iter: 30
klt_max_level: 4
klt_eps: 0.1
maxFeatureAge: 15
maxFeaturesPerFrame: 800  #it was 800
quality_level: 0.001
min_distance: 20
block_size: 3
use_harris_detector: 0
k: 0.04
equalizeImage: 0
nominalBaseline: 0.179622
toleranceTemplateMatching: 0.15
templ_cols: 101 #must be odd
templ_rows: 11
stripe_extra_rows: 0
minPointDist: 0.5
maxPointDist: 10
bidirectionalMatching: 0
subpixelRefinementStereo: 0
featureSelectionCriterion: 0  #(QUALITY, MIN_EIG, LOGDET, RANDOM)
featureSelectionHorizon: 3
featureSelectionNrCornersToSelect: 600
featureSelectionImuRate: 0.0025    #it was 0.005
featureSelectionDefaultDepth: 3
featureSelectionCosineNeighborhood: 0.984807753012208 # Rad
featureSelectionUseLazyEvaluation: 1
useSuccessProbabilities: 1
useRANSAC: 1
minNrMonoInliers: 10
minNrStereoInliers: 5
ransac_threshold_mono: 1e-06
ransac_threshold_stereo: 1
ransac_use_1point_stereo: 1
ransac_use_2point_mono: 1
ransac_max_iterations: 100
ransac_probability: 0.995
ransac_randomize: 0
intra_keyframe_time: 0.001 #it was 0.2
minNumberFeatures: 0
useStereoTracking: 1
display_time: 100
disparityThreshold: 0.2   #it was 0.5
--------------------------------------------------------------------------------------------------------

```

**Console output:**
```
I0316 17:37:10.589303  1896 VioBackEnd.cpp:1615]  =============== START:CATCHING EXCEPTION ===============
I0316 17:37:10.589303  1896 VioBackEnd.cpp:1643] Nr of factors in graph (graph before optimization): 4036, with factors:
I0316 17:37:10.589303  1896 VioBackEnd.cpp:1645] [
        Slot # 4035: Linear Container Factor:   Factor  b10 v10 x10 x11

I0316 17:37:10.590301  1896 VioBackEnd.cpp:1652]  ]
I0316 17:37:10.590301  1896 VioBackEnd.cpp:1656] Nr of new factors to add: 196 with factors:
I0316 17:37:10.590301  1896 VioBackEnd.cpp:1658] [
 (slot # wrt to new_factors_tmp graph)

I0316 17:37:10.590301  1896 VioBackEnd.cpp:1665]  ]
I0316 17:37:10.590301  1896 VioBackEnd.cpp:1669] Nr deleted slots: 22, with slots:
I0316 17:37:10.590301  1896 VioBackEnd.cpp:1671] [

I0316 17:37:10.590301  1896 VioBackEnd.cpp:1698]  ]
I0316 17:37:10.590301  1896 VioBackEnd.cpp:1702] Nr of values in state_ : 69, with keys:
[
        b10 b11 b12 b13 b14 b15 b16 b17 b18 b19 b20 b21 b22 b23 b24 b25 b26 b27 b28 b29 b30 b31 b32 v10 v11 v12 v13 v14 v15 v16 v17 v18 v19 v20 v21 v22 v23 v24 v25 v26 v27 v28 v29 v30 v31 v32 x10 x11 x12 x13 x14 x15 x16 x17 x18 x19 x20 x21 x22 x23 x24 x25 x26 x27 x28 x29 x30 x31 x32
I0316 17:37:10.590301  1896 VioBackEnd.cpp:1708]  ]
I0316 17:37:10.590301  1896 VioBackEnd.cpp:1711] Nr values in new_values_ : 3, with keys:
[
         b33  v33  x33
I0316 17:37:10.590301  1896 VioBackEnd.cpp:1719]  ]
I0316 17:37:10.590301  1896 VioBackEnd.cpp:1730]  =============== END: CATCHING EXCEPTION ===============

```

**Additional files:**
Please attach all the files needed to reproduce the error.

Please give also the following information:
* SparkVio branch, tag or commit used
* GTSAM version used:
* OpenGV version used:
* OpenCV version used: type `opencv_version`
* Operating system and version (e.g. Ubuntu 16.04 or Windows 10):  Windows 10
* Did you change the source code? (yes / no): no

ToniRV, 2020-03-16:
Hi @fouad1995, thanks for reporting.

We haven't really tested ZED 2 camera, did you update the ImuParams.yaml, Left/RightCameraParams.yaml?
I would suggest using [Kalibr](https://github.com/ethz-asl/kalibr) to get these right.

fouad1995, 2020-03-17:
Hi , thanks for your reply.
i already update ImuParams.yaml , Left/RightCameraParms.yaml and also used Kalibr but there is something wrong and i didn't understand Console output to debug 
---------------------------------------------------------------------------------------------------------
ImuParams.yaml 
%YAML:1.0
# Type of IMU preintegration:
# 0: CombinedImuFactor
# 1: ImuFactor
imu_preintegration_type: 1

T_BS:
  cols: 4
  rows: 4
  data: [1.0, 0.0, 0.0, 0.0,
         0.0, 1.0, 0.0, 0.0,
         0.0, 0.0, 1.0, 0.0,
         0.0, 0.0, 0.0, 1.0]
rate_hz: 400

# inertial sensor noise model parameters (static)
gyroscope_noise_density: 8.6e-05     # [ rad / s / sqrt(Hz) ]   ( gyro "white noise" )
gyroscope_random_walk: 2.2e-06       # [ rad / s^2 / sqrt(Hz) ] ( gyro bias diffusion )
accelerometer_noise_density: 0.0014   # [ m / s^2 / sqrt(Hz) ]   ( accel "white noise" )
accelerometer_random_walk: 8.0e-05    # [ m / s^3 / sqrt(Hz) ].  ( accel bias diffusion )

# Extra parameters
imu_integration_sigma: 1.0e-8
imu_time_shift: -0.0004913018933950657
n_gravity: [0.0, 0.0, -9.81]
---------------------------------------------------------------------------------------------------------
 LeftCameraParams.yaml
camera_id: left_cam

# Sensor extrinsics wrt. the body-frame.
T_BS:
  cols: 4
  rows: 4
  data: [0.9998933873072151, 0.0010550571826621227, 0.014563683381797836, -0.18194255916575514,
         -0.0010853817294038297, 0.9999972592852897, 0.002074456172353754, -0.0067353521261741415,
        -0.014561454797011793, -0.00209004216485058, 0.9998917920244891, 0.001699595719823257,
         0.0, 0.0, 0.0, 1.0]

# Camera specific definitions.
rate_hz: 15
resolution: [1280, 720]
camera_model: pinhole
intrinsics: [530.4083419772949, 532.1056971857535, 668.0350967019064, 340.4321280694387] #fu, fv, cu, cv
distortion_model: equidistant
distortion_coefficients: [0.21939477849952854, 0.4355303682977106, -0.2340650394424158, 0.09711016463789046]
--------------------------------------------------------------------------------------------------------
RightCameraParams.yaml
camera_id: right_cam

# Sensor extrinsics wrt. the body-frame.
T_BS:
  cols: 4
  rows: 4
  data: [0.9998868149065552, 0.0002386139324634518, 0.01504328552600487, -0.0023094675842962797,
         -0.00026629263481123055, 0.9999982754921434, 0.0018379610913340232,  -0.006873311343661757,
        -0.015042821020617063, -0.0018417589776750887, 0.9998851541350192, 0.001154459918399746,
         0.0, 0.0, 0.0, 1.0]

# Camera specific definitions.
rate_hz: 15
resolution: [1280, 720]
camera_model: pinhole
intrinsics: [529.7404242221473, 531.4905125317819, 668.0580730801421, 340.4265368191226] #fu, fv, cu, cv
distortion_model: equidistant
distortion_coefficients: [0.243326180870939,  0.28193873057203617, 0.1016150397927026, -0.1380494424459571]
-------------------------------------------------------------------------------------------------------

fouad1995, 2020-03-17:
Is there any tips and tricks for tuning parameters for outdoor ?

marcusabate, 2023-11-03:
For future visitors or if still relevant, for outdoor scenes we find that vision parameters need to be adjusted slightly to support farther keypoints. Increase `min_distance` in `FrontendParams.yaml`, increase `maxFeaturesPerFrame` while maintaining acceptable latency, use non-max suppression (also in `FrontendParams.yaml`), and ensure the `BackendParams.yaml` parameter related to outlier rejection (eg `outlierRejection`, `smartNoiseSigma`, etc) are properly set for your dataset.

closed 2023-11-03 by marcusabate