MIT-SPARK/Kimera-VIO #50: Nominal baseline parameter is redundant

jingnanshi, 2020-04-14:
**Description:**

The `nominalBaseline` parameter in the `trackerParameters.yml` files is redundant since we already have the extrinsics in the camera parameter yaml files.

fouad1995, 2020-04-16:
In ```StereoFrame.cpp``` you can see that the base line is calculated already from the extrinsics in the **camera parameter yaml files** and somehow it compares this base line with the one you insert in the ```trackerPrarameters.yml```  , i think **Kimera** makes this to make sure that the calibration of your cameras is correct because you know the base line of your stereo camera explicitly from its data-sheet or whatever the method so if your calibration is correct then the calculation of base line will be same or almost the same of the one you insert , but if not this means that there is a problem in your calibration or maybe the calibration is correct but the base line is not

jingnanshi, 2020-04-16:
These are valid points. However in some situations you don't know the baseline beforehand, in which case the `norminalBaseline` parameter will be redundant.

ToniRV, 2020-05-10:
We removed the nominal baseline as it seemed to give more headaches than to help. Thanks!

closed 2020-05-10 by ToniRV