MIT-SPARK/Kimera-VIO #56: Kimera parameters

rttariverdi67, 2020-04-30:
Dear Repository,
I want to experiment Kimera performance with different parameters and see the differences. Also, try to obtain the set of 3D landmarks generated this way. The problem is I don't know how to access this parameters. Can you guide me from where I should start in Kimera?

- also I'm wondering how can I get point clouds of given frames in Kimera!

Tank you for your help in advance,
Rahim

violetteavi, 2020-04-30:
1. Parameters are in .yaml files in the /params folder, organized by dataset. We use Euroc for testing and evaluation. `FrontEndParams.yaml` is probably where you want to focus. I would recommend against touching the Imu or Camera parameter files, as those are calibrated for the hardware.

2. Are you using ROS? In ROS, the 3D landmarks are published as A) a pointcloud of the currently tracked features (including some from the past) on the topic `/kimera_vio_ros/time_horizon_pointcloud` and B) a mesh of all the active feature points in the current frame on the topic `/kimera_vio_ros/mesh`. If you are looking to access them programmatically from C++, you can run the pipeline with [this example](https://github.com/MIT-SPARK/Kimera-VIO/blob/d677a3dedad2da605c84a4b7f51154cd7e38a98d/examples/KimeraVIO.cpp), then register callbacks with the VioPipeline (see [here](https://github.com/MIT-SPARK/Kimera-VIO-ROS/blob/master/src/KimeraVioRos.cpp#L169)).

rttariverdi67, 2020-04-30:
Thank you for your quick response,It as helpful. I didn't use ROS but I might need to use.
Best wishes,
Rahim

closed 2020-05-10 by ToniRV