MIT-SPARK/Kimera-VIO #62: Why is biasAccOmegaInt so large?

andre-nguyen, 2020-05-05:
Full disclosure I haven't actually tried out Kimera yet (nice work btw). But I was interested in understanding why here https://github.com/MIT-SPARK/Kimera-VIO/blob/d677a3dedad2da605c84a4b7f51154cd7e38a98d/src/imu-frontend/ImuFrontEnd.cpp#L200

`biasAccOmegaInt` is set to such a large value when in the gtsam example it is set to `1e-5`. So I've read that it models the covariance of the bias used during preintegration which would mean that you really don't trust the biases you are estimating on your IMU? Why is that?

ToniRV, 2020-05-05:
Good catch! We are actually not using the Combined IMU factor, but I'll take a note on that:
https://github.com/MIT-SPARK/Kimera-VIO/blob/a5109c7157482bb1203918512ef6593e226507f7/params/Euroc/ImuParams.yaml#L5

We allowed using either the ImuFactor or the CombinedImuFactor from gtsam, I didn't notice any significant difference at the time though, I might as well try again with a lower `biasAccOmegaInt`.

andre-nguyen, 2020-05-05:
Thanks for the quick reply, as a side note it says on [bitbucket](https://bitbucket.org/gtborg/gtsam/issues/254/combinedimufactor-question-and-example#comment-28366634) that if you set `biasAccOmegaInt` to 0 it will be the same.

closed 2020-05-05 by andre-nguyen