MIT-SPARK/Kimera-VIO #66: Dataset to test loopclosures

WouterJansen, 2020-05-07:
I've been experimenting with Kimera and while the state estimation works great I was wondering  how I could best to test the loop closure detection and PGO. As when I run Kimera (I use the ROS Wrapper) with the euroc dataset and verbosity set to print out loop closures it doesn't seem to find any. Is there any dataset available to test this better or a publicly recommended dataset that I could use?

ToniRV, 2020-05-10:
Hi @WouterJansen, the Euroc dataset is the only one we support right now for loop closures.
Kitti would be nice, but it doesn't have real IMU data.
Alternatively, we could upload a rosbag doing a big loop-closure in a simulator (given current lockdown it might be the safest/quickest).

gkiri, 2020-05-18:
Hi @ToniRV ,
I have been testing Kimera with intel real sense D435i hardware.Just wondering if Loop closures will work on live captured data or not?

WouterJansen, 2020-05-29:
See #72 for I think the same issue I'm experiencing. Perhaps it's not a dataset issue but more an issue with the LCD.

closed 2020-05-29 by WouterJansen