MIT-SPARK/Kimera-VIO #72: No loop closure detected

kdr193, 2020-05-13:


ToniRV, 2020-05-19:
Hi @kdr193,
Alpha is what controls the number of loop closures detected, try to increase it, as it is now low: `alpha: 0.001`
Check in [here](https://github.com/MIT-SPARK/Kimera-VIO/blob/master/params/Euroc/LcdParams.yaml).

WouterJansen, 2020-05-29:
I'm experiencing the same issue. Changed the alpha value to 0.1 but still no LC detected;

WouterJansen, 2020-08-03:
@ToniRV 

I'm continuing to experience many issues regarding loop closures and (R)PGO in the latest version. In the V1 euroc set I have to set the LCD parameters to
```
alpha: 1
min_temporal_matches: 0
```
to detect a few closures. And I don't believe they are in any way correct.

I know this is in the wrong repository to say but on top of that none of the LC/RPGO related ROS topics does anything get published. 
```
/kimera_vio_ros/graph_nodes
/kimera_vio_ros/loop_edges
/kimera_vio_ros/optimized_trajectory
/kimera_vio_ros/pose_graph
/kimera_vio_ros/rejected_loop_edges
```
All those topics are empty. So it's also rather difficult to test the system this way.

Is this something you guys are experiencing yourselves or does this continue to just be an issue for certain users? I'm using the current master updated and nothing changed except LCD turned on (and those LCD 2 parameters tweaked as stated above).

WouterJansen, 2020-09-04:
As stated above, currently can get a few loop closures in the euroc dataset if I set `alpha` to 0.001 and `min_temporal_matches`  to 0. However, I am not certain these are detected correctly now or are false positives. Hard to tell in Euroc where there should be closures detected just visually.

In my own dataset I also get similar results but granted it is not the best dataset (taken in simulation, low rates of IMU and camera) but it does not a bad job at the state estimation with Kimera-VIO. However, the LCD did not detect some very obvious loops. The biggest reason seems to be _LOW_NSS_FACTOR_. I saw this meant normalized similarity score but didn't really understand what it meant in terms of LCD specifically in Kimera. Judging by the code it analyzes mostly if you are standing still by comparing current to previous DBOW vector? I might be interpreting that wrong. However, when turning it off (`use_nss: 0`) for testing the obvious loop point was detected correctly. 

Just thought I'd share this with others having these issues with their datasets

closed 2023-11-03 by marcusabate