MIT-SPARK/Kimera-VIO #73: Reduce the number of compiler warnings

ruffsl, 2020-05-13:
Compiling the project generates a large number of compiler warnings, making it difficult to contribute while being careful not in introduce more warnings oneself or picking out compilation errors during development. The root causes for these warnings should be resolved as not to overwhelm contributors with walls of warnings in build logs and CI.

ruffsl, 2020-05-13:
List of flags to suppress current list warnings:

```
-DCMAKE_CXX_FLAGS="\
    -Wno-comment \
    -Wno-maybe-uninitialized \
    -Wno-parentheses \
    -Wno-pragma-once-outside-header \
    -Wno-reorder \
    -Wno-return-type \
    -Wno-sign-compare \
    -Wno-unused-but-set-variable \
    -Wno-unused-function \
    -Wno-unused-parameter \
    -Wno-unused-value \
    -Wno-unused-variable"
```

Example build log without any warning suppression:
(about 2.5k lines of warnings)
> https://pastebin.com/zQ9r3Lec

ToniRV, 2020-05-13:
I might use these in our CI, thanks! 
Most of these warnings are from legacy code, but I strive to keep reducing them...
There are far too many though, agree.

closed 2023-11-03 by marcusabate