MIT-SPARK/Kimera-VIO #81: Mono-only capabilities release date

rastinrastini, 2020-05-25:
Dear,
thanks for this code.
have any release date for mono-only capability while you wrote "Mono-only capabilities soon to be released"?
Thankful.

ToniRV, 2020-05-29:
@rastinrastini unfortunately I am caught with some deadlines so I won't be able to release this before the summer ends :(

closed 2020-05-29 by ToniRV

aseyfi, 2020-09-08:
Hi @ToniRV ,

Thank you for sharing this code. Any updates on releasing mono-only feature?

Thanks,
Ahmad

ToniRV, 2020-09-08:
We are currently working on it.
First week of October would be a reasonable deadline if all goes well.

aseyfi, 2020-09-08:
Thanks @ToniRV for the quick response. Based on what I see in [issue-9](https://github.com/MIT-SPARK/Kimera-VIO/issues/9), it seems we already can use mono+IMU odometry. Would you please confirm?

jaadelgren, 2020-10-23:
Also interested in this, any update?

ToniRV, 2020-10-24:
> Thanks @ToniRV for the quick response. Based on what I see in issue-9, it seems we already can use mono+IMU odometry. Would you please confirm?

Technically yes, but there are some sections of the code that still assume Stereo: we are currently working on making the whole pipeline also work in the mono-only case.

sathishkumark27, 2020-11-24:
@ToniRV 
Thanks for the great work. does mono-only means mono without IMU? Do you have any update on the mono-only?

Thanks & Regards
Sathish

danigr99727, 2020-12-13:
any updates on this? I have tried to run the code with mono-only euroc, without success.

DamDsj, 2021-01-20:
@ToniRV 
Is there any update concerningthe mono+imu feature? Is there a branch/patch where we could try this out?
Thanks
Regards
Damien

neo-knight-td, 2021-03-08:
@ToniRV, 
Any update concerning monocular ?

ToniRV, 2021-03-08:
Hi everyone, yes, we will be pushing the mono+imu version soon (this week!).

bednarhonza, 2021-03-17:
I believe here #152 is the pull request for the mono feature.

ToniRV, 2021-03-17:
Yes, PR is #152 :+1:

aseyfi, 2021-04-06:
@ToniRV , thank you very much for releasing this code. I have been able to run the dockers  you have provided, but I would like to use kimera mono with ROS. Does the ROS wrapper support kimera mono?

Thank You,
Ahmad

marcusabate, 2021-04-06:
@aseyfi The associated ROS PR is on its way, maybe another day or two.

yhabib29, 2021-04-08:
Could you give more details on which functionalities won't be available in monocular mode please ?

johnsmakar, 2021-12-05:
Is loop closure available for mono?