MIT-SPARK/Kimera-VIO #82: Build failure

bennywijaya55, 2020-05-29:
**Description:**
Hello, I am trying to install Kimera-VIO in Ubuntu 16.04. I installed all the dependencies. But when I run 'sudo make -j4' for Kimera-VIO, the process terminate without fatal error.

**Command:**
cd Kimera-VIO-master
mkdir build
cd build
cmake ..
make -j4

**The configuration result shows no error whatsoever:**
-- The CXX compiler identification is GNU 5.4.0
-- Check for working CXX compiler: /usr/lib/ccache/c++
-- Check for working CXX compiler: /usr/lib/ccache/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- ==============================================================
-- ====================  Dependencies ===========================
-- Looking for C++ include pthread.h
-- Looking for C++ include pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Boost version: 1.58.0
-- Found the following Boost libraries:
--   filesystem
--   serialization
--   regex
--   timer
--   date_time
--   thread
--   system
--   chrono
--   atomic
-- Failed to find installed gflags CMake configuration, searching for gflags build directories exported with CMake.
-- Found exported gflags build directory: /home/benny/maplab_ws/build/gflags_catkin/gflags_src-prefix/src/gflags_src
-- Detected broken gflags install in: /home/benny/maplab_ws/build/gflags_catkin/gflags_src-prefix/src/gflags_src, version: 2.1.2 <= 2.1.2 which defines gflags_LIBRARIES = gflags which is not an imported CMake target, see: https://github.com/gflags/gflags/issues/110.  Attempting to fix by detecting correct gflags target.
-- Found valid gflags target: gflags-shared, updating gflags_LIBRARIES.
-- Detected gflags version: 2.1.2
-- Found Gflags: /home/benny/maplab_ws/build/gflags_catkin/gflags_src-prefix/src/gflags_src/include  
-- Failed to find installed glog CMake configuration, searching for glog build directories exported with CMake.
-- Failed to find an installed/exported CMake configuration for glog, will perform search for installed glog components.
-- Found Glog: /usr/include (Required is at least version "0.3.5") 
-- GTSAM include directory:  /usr/local/lib/cmake/GTSAM/../../../include;/usr/include;/usr/local/include/gtsam/3rdparty/Eigen/
-- Found OpenCV: /opt/ros/kinetic (found suitable version "3.3.1", minimum required is "3.3.1") 
-- Vocabulary file exists, will not download.
Cloning into 'googletest-src'...
Note: checking out 'v1.10.x'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by performing another checkout.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -b with the checkout command again. Example:

  git checkout -b <new-branch-name>

HEAD is now at 703bd9c... Googletest export
-- The C compiler identification is GNU 5.4.0
-- Check for working C compiler: /usr/lib/ccache/cc
-- Check for working C compiler: /usr/lib/ccache/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
[... 450 lines cut ...]
/home/benny/kimeraVIO/Kimera-VIO-master/include/kimera-vio/dataprovider/EurocDataProvider.h:192:11: warning: ‘VIO::EurocDataProvider::final_k_’ will be initialized after [-Wreorder]
   FrameId final_k_;    // end frame
           ^
/home/benny/kimeraVIO/Kimera-VIO-master/include/kimera-vio/dataprovider/EurocDataProvider.h:177:13: warning:   ‘VIO::VioParams VIO::EurocDataProvider::pipeline_params_’ [-Wreorder]
   VioParams pipeline_params_;
             ^
/home/benny/kimeraVIO/Kimera-VIO-master/src/dataprovider/EurocDataProvider.cpp:51:1: warning:   when initialized here [-Wreorder]
 EurocDataProvider::EurocDataProvider(const std::string& dataset_path,
 ^
/home/benny/kimeraVIO/Kimera-VIO-master/src/dataprovider/EurocDataProvider.cpp: In member function ‘const VIO::InitializationPerformance VIO::EurocDataProvider::getInitializationPerformance(const std::vector<long int>&, const std::vector<gtsam::Pose3>&, const VIO::VioNavState&, const Vector3&)’:
/home/benny/kimeraVIO/Kimera-VIO-master/src/dataprovider/EurocDataProvider.cpp:602:21: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
   for (int i = 1; i < timestamps.size(); i++) {
                     ^
In file included from /home/benny/kimeraVIO/Kimera-VIO-master/include/kimera-vio/dataprovider/EurocDataProvider.h:35:0,
                 from /home/benny/kimeraVIO/Kimera-VIO-master/src/dataprovider/EurocDataProvider.cpp:17:
/home/benny/kimeraVIO/Kimera-VIO-master/include/kimera-vio/logging/Logger.h: At global scope:
/home/benny/kimeraVIO/Kimera-VIO-master/include/kimera-vio/logging/Logger.h:31:13: warning: ‘void VIO::OpenFile(const string&, std::ofstream*, bool)’ defined but not used [-Wunused-function]
 static void OpenFile(const std::string& output_filename,
             ^
CMakeFiles/kimera_vio.dir/build.make:134: recipe for target 'CMakeFiles/kimera_vio.dir/src/dataprovider/EurocDataProvider.cpp.o' failed
make[2]: *** [CMakeFiles/kimera_vio.dir/src/dataprovider/EurocDataProvider.cpp.o] Error 1
CMakeFiles/Makefile2:520: recipe for target 'CMakeFiles/kimera_vio.dir/all' failed
make[1]: *** [CMakeFiles/kimera_vio.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2

bennywijaya55, 2020-05-29:
**Dependencies:**
GTSAM: 4.0.0
OpenCV: 3.3.1

Also I havent make any change for the source code

ToniRV, 2020-05-29:
The error you are seeing is because GTSAM version is too old, I would recommend using the newest changes in the `develop` branch of gtsam: `git fetch && git checkout develop`

Sushant-Chavan, 2020-06-01:
Could the rosinstall files [kimera_vio_ros_ssh](https://github.com/MIT-SPARK/Kimera-VIO-ROS/blob/c0fe1caa0aecc4f141980affd94b22fdb4e93153/install/kimera_vio_ros_ssh.rosinstall#L16) and [kimera_vio_ros_https](https://github.com/MIT-SPARK/Kimera-VIO-ROS/blob/c0fe1caa0aecc4f141980affd94b22fdb4e93153/install/kimera_vio_ros_https.rosinstall#L16) in Kimera-VIO-ROS be updated to use the develop version instead of master?

bennywijaya55, 2020-06-01:
Hi, thanks for the reply! I have now use the latest version of GTSAM from the develop branch. It solved the previous problem. However now I have another problem and it is related to GTSAM again I think! Do you have any suggestions?

**This is the cmake configuration output:**
-- The CXX compiler identification is GNU 5.4.0
-- Check for working CXX compiler: /usr/lib/ccache/c++
-- Check for working CXX compiler: /usr/lib/ccache/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- ==============================================================
-- ====================  Dependencies ===========================
-- Looking for C++ include pthread.h
-- Looking for C++ include pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Boost version: 1.58.0
-- Found the following Boost libraries:
--   filesystem
--   serialization
--   regex
--   timer
--   date_time
--   thread
--   system
--   chrono
--   atomic
-- Failed to find installed gflags CMake configuration, searching for gflags build directories exported with CMake.
-- Found exported gflags build directory: /home/benny/maplab_ws/build/gflags_catkin/gflags_src-prefix/src/gflags_src
-- Detected broken gflags install in: /home/benny/maplab_ws/build/gflags_catkin/gflags_src-prefix/src/gflags_src, version: 2.1.2 <= 2.1.2 which defines gflags_LIBRARIES = gflags which is not an imported CMake target, see: https://github.com/gflags/gflags/issues/110.  Attempting to fix by detecting correct gflags target.
-- Found valid gflags target: gflags-shared, updating gflags_LIBRARIES.
-- Detected gflags version: 2.1.2
-- Found Gflags: /home/benny/maplab_ws/build/gflags_catkin/gflags_src-prefix/src/gflags_src/include  
-- Failed to find installed glog CMake configuration, searching for glog build directories exported with CMake.
-- Failed to find an installed/exported CMake configuration for glog, will perform search for installed glog components.
-- Found Glog: /usr/include (Required is at least version "0.3.5") 
-- Boost version: 1.58.0
-- Found the following Boost libraries:
--   serialization
--   system
--   filesystem
--   thread
--   program_options
--   date_time
--   timer
--   chrono
--   regex
--   atomic
-- GTSAM include directory:  /usr/local/lib/cmake/GTSAM/../../../include
-- Found OpenCV: /opt/ros/kinetic (found suitable version "3.3.1", minimum required is "3.3.1") 
-- Vocabulary file exists, will not download.
Cloning into 'googletest-src'...
Note: checking out 'v1.10.x'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by performing another checkout.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -b with the checkout command again. Example:

  git checkout -b <new-branch-name>

HEAD is now at 703bd9c... Googletest export
-- The C compiler identification is GNU 5.4.0
[... 1939 lines cut ...]
                 ^
/usr/local/include/gtsam/geometry/Rot3.h:124:17: note:   no known conversion for argument 1 from ‘std::mt19937 {aka std::mersenne_twister_engine<long unsigned int, 32ul, 624ul, 397ul, 31ul, 2567483615ul, 11ul, 4294967295ul, 7ul, 2636928640ul, 15ul, 4022730752ul, 18ul, 1812433253ul>}’ to ‘boost::random::mt19937& {aka boost::random::mersenne_twister_engine<unsigned int, 32ul, 624ul, 397ul, 31ul, 2567483615u, 11ul, 4294967295u, 7ul, 2636928640u, 15ul, 4022730752u, 18ul, 1812433253u>&}’
/home/benny/kimeraVIO/Kimera-VIO-master/tests/testLogger.cpp:414:31: error: no matching function for call to ‘gtsam::Rot3::Random(std::mt19937&)’
       gtsam::Rot3::Random(rng_), gtsam::Point3::Random());
                               ^
In file included from /usr/local/include/gtsam/geometry/Pose3.h:24:0,
                 from /usr/local/include/gtsam/geometry/CalibratedCamera.h:23,
                 from /usr/local/include/gtsam/geometry/CameraSet.h:22,
                 from /usr/local/include/gtsam/slam/RegularImplicitSchurFactor.h:10,
                 from /usr/local/include/gtsam/slam/SmartFactorBase.h:25,
                 from /usr/local/include/gtsam_unstable/slam/SmartStereoProjectionFactor.h:23,
                 from /usr/local/include/gtsam_unstable/slam/SmartStereoProjectionPoseFactor.h:23,
                 from /home/benny/kimeraVIO/Kimera-VIO-master/include/kimera-vio/backend/VioBackEnd-definitions.h:26,
                 from /home/benny/kimeraVIO/Kimera-VIO-master/tests/testLogger.cpp:28:
/usr/local/include/gtsam/geometry/Rot3.h:124:17: note: candidate: static gtsam::Rot3 gtsam::Rot3::Random(boost::random::mt19937&)
     static Rot3 Random(boost::mt19937 & rng);
                 ^
/usr/local/include/gtsam/geometry/Rot3.h:124:17: note:   no known conversion for argument 1 from ‘std::mt19937 {aka std::mersenne_twister_engine<long unsigned int, 32ul, 624ul, 397ul, 31ul, 2567483615ul, 11ul, 4294967295ul, 7ul, 2636928640ul, 15ul, 4022730752ul, 18ul, 1812433253ul>}’ to ‘boost::random::mt19937& {aka boost::random::mersenne_twister_engine<unsigned int, 32ul, 624ul, 397ul, 31ul, 2567483615u, 11ul, 4294967295u, 7ul, [... cut ...] 40u, 15ul, 4022730752u, 18ul, 1812433253u>&}’
CMakeFiles/testKimeraVIO.dir/build.make:374: recipe for target 'CMakeFiles/testKimeraVIO.dir/tests/testLogger.cpp.o' failed
make[2]: *** [CMakeFiles/testKimeraVIO.dir/tests/testLogger.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
CMakeFiles/Makefile2:846: recipe for target 'CMakeFiles/testKimeraVIO.dir/all' failed
make[1]: *** [CMakeFiles/testKimeraVIO.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2

bennywijaya55, 2020-06-01:
Also, this is the configuration of the GTSAM:

-- The CXX compiler identification is GNU 5.4.0
-- The C compiler identification is GNU 5.4.0
-- Check for working CXX compiler: /usr/lib/ccache/c++
-- Check for working CXX compiler: /usr/lib/ccache/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /usr/lib/ccache/cc
-- Check for working C compiler: /usr/lib/ccache/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- GTSAM_SOURCE_ROOT_DIR: [/home/benny/kimeraVIO/gtsam-develop]
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Found Boost: /usr/include (found suitable version "1.58.0", minimum required is "1.43") found components:  serialization system filesystem thread program_options date_time timer chrono regex atomic 
Ignoring Boost restriction on optional lvalue assignment from rvalues
-- Found TBB: /usr/include (found suitable version "4.4", minimum required is "4.4") found components:  tbb tbbmalloc 
-- Found MKL: /opt/intel/mkl/include  
-- Try OpenMP C flag = [-fopenmp]
-- Performing Test OpenMP_FLAG_DETECTED
-- Performing Test OpenMP_FLAG_DETECTED - Success
-- Try OpenMP CXX flag = [-fopenmp]
-- Performing Test OpenMP_FLAG_DETECTED
-- Performing Test OpenMP_FLAG_DETECTED - Success
-- Found OpenMP: -fopenmp  
-- Found Eigen3: /usr/include/eigen3 (Required is at least version "2.91.0") 
-- Found Eigen version: 3.2.92
-- Building 3rdparty
-- Looking for execinfo.h
-- Looking for execinfo.h - found
-- Looking for getline
-- Looking for getline - found
-- checking for thread-local storage - found
-- Could NOT find GeographicLib (missing:  GeographicLib_LIBRARY_DIRS GeographicLib_LIBRARIES GeographicLib_INCLUDE_DIRS) 
-- Building base
-- Building geometry
-- Building inference
-- Building symbolic
-- Building discrete
-- Building linear
-- Building nonlinear
-- Building sam
-- Building sfm
-- Building slam
-- Building smart
-- Building navigation
-- GTSAM Version: 4.0.2
-- Install prefix: /usr/local
-- Building GTSAM - shared: ON
-- Building base_unstable
-- Building geometry_unstable
-- Building linear_unstable
-- Building discrete_unstable
-- Building dynamics_unstable
-- Building nonlinear_unstable
-- Building slam_unstable
-- Building partition_unstable
-- GTSAM_UNSTABLE Version: 4.0.2
[... 52 lines cut ...]
--   Eigen will use MKL             : Yes
--   Eigen will use MKL and OpenMP  : Yes
--   Default allocator              : TBB
--   Build with ccache              : Yes
-- Packaging flags                                               
--   CPack Source Generator         : TGZ
--   CPack Generator                : TGZ
-- GTSAM flags                                               
--   Quaternions as default Rot3     : Disabled
--   Runtime consistency checking    : Disabled
--   Rot3 retract is full ExpMap     : Disabled
--   Pose3 retract is full ExpMap    : Disabled
--   Deprecated in GTSAM 4 allowed   : Enabled
--   Point3 is typedef to Vector3    : Disabled
--   Metis-based Nested Dissection   : Enabled
--   Use tangent-space preintegration: Enabled
--   Build Wrap                     : Enabled
-- MATLAB toolbox flags                                      
--   Install matlab toolbox         : Disabled
-- Cython toolbox flags                                      
--   Install Cython toolbox         : Disabled
-- ===============================================================
-- Configuring done
-- Generating done
-- Build files have been written to: /home/benny/kimeraVIO/gtsam-develop/build

closed 2023-11-03 by marcusabate