MIT-SPARK/Kimera-VIO #84: is it possible to use Kimera-VIO as a odometry for Dense 3D reconstruction system?

GucciPrada, 2020-06-03:
It seems that Kimera-VIO only produce an optimized pose for each keyframe, and only fuse the keyframes to Kimera-semantics module. So the map will not be as dense as general 3d reconstruction, right?

Is it possible to make Kimera-VIO to estimate a 3d pose for every consecutive frame?

Thanks!

ToniRV, 2020-10-15:
Hi @Mad-Thanos, good point! Right now we integrate pointclouds even for non-keyframes but we use the ROS TF interpolation between keyframes, which is definitely sub-optimal.
At some point I was sending pose estimates at IMU rate to ROS, but with the latest heavy refactor I haven't been able to re-implement that feature.

apache8080, 2020-10-19:
When you integrate every pointcloud does the pose get updated also? Right now, I use registerBackendOutputCallback to setup a callback that gets the pose after every keyframe but I want to get the pose after each camera frame.

apache8080, 2020-10-19:
https://github.com/MIT-SPARK/Kimera-VIO/blob/8012623e0780a9d59ab2e27bbed68b163a94d1b4/src/pipeline/Pipeline.cpp#L134
I see that this where the check for if it is a key frame happens. If I just had the backend run for every frame how much would that ruin the accuracy?

ToniRV, 2020-10-19:
The accuracy won't get much worst, but you may not achieve real-time performance.

ToniRV, 2020-10-19:
You can try to set the param `intra_keyframe_time` to `0.0` in `params/Euroc/FrontendParams.yaml` to have a pose per frame.

apache8080, 2020-10-20:
Thanks. Do you plan on ever releasing the version that publishes poses at IMU rate?

marcusabate, 2023-11-03:
Currently there is no plan to reimplement the IMU rate pose estimation

closed 2023-11-03 by marcusabate