MIT-SPARK/Kimera-VIO #89: Error running example in yaml parser

esteimle, 2020-06-15:
**Description**
I built kimeria-vio and downloaded your version of V1_01_easy.  But it fails when I try to run it.

**Command:**
```
sudo bash scripts/stereoVIOEuroc.bash -p "/home/sam/V1_01_easy"
```

**Console output:**
```

Using dataset at path: /home/sam/V1_01_easy
 Launching:

            ██╗  ██╗██╗███╗   ███╗███████╗██████╗  █████╗
            ██║ ██╔╝██║████╗ ████║██╔════╝██╔══██╗██╔══██╗
            █████╔╝ ██║██╔████╔██║█████╗  ██████╔╝███████║
            ██╔═██╗ ██║██║╚██╔╝██║██╔══╝  ██╔══██╗██╔══██║
            ██║  ██╗██║██║ ╚═╝ ██║███████╗██║  ██║██║  ██║
            ╚═╝  ╚═╝╚═╝╚═╝     ╚═╝╚══════╝╚═╝  ╚═╝╚═╝  ╚═╝

 
F0615 10:57:24.729456  5682 YamlParser.h:44] Check failed: file_handle.type() != cv::FileNode::NONE (0 vs. 0) Missing parameter: maxFeaturesPerFrame in file: ���!IV
*** Check failure stack trace: ***
    @     0x7ff8985bd0cd  google::LogMessage::Fail()
    @     0x7ff8985bef33  google::LogMessage::SendToLog()
    @     0x7ff8985bcc28  google::LogMessage::Flush()
    @     0x7ff8985bf999  google::LogMessageFatal::~LogMessageFatal()
    @     0x7ff89999c077  VIO::FeatureDetectorParams::parseYAML()
    @     0x7ff899987d0b  VIO::FrontendParams::parseYAML()
    @     0x7ff899a9b3ad  VIO::VioParams::parseYAML()
    @     0x7ff899a9c8c1  VIO::VioParams::VioParams()
    @     0x7ff899a9cae3  VIO::VioParams::VioParams()
    @     0x564921a8ae05  main
    @     0x7ff897dc1b6b  __libc_start_main
    @     0x564921a8b5aa  _start
scripts/stereoVIOEuroc.bash: line 109:  5682 Aborted                 (core dumped) $BUILD_PATH/stereoVIOEuroc --dataset_type="$DATASET_TYPE" --dataset_path="$DATASET_PATH" --initial_k=50 --final_k=2000 --params_folder_path="$PARAMS_PATH" --use_lcd="$USE_LCD" --vocabulary_path="$VOCABULARY_PATH/ORBvoc.yml" --flagfile="$PARAMS_PATH/flags/stereoVIOEuroc.flags" --flagfile="$PARAMS_PATH/flags/Mesher.flags" --flagfile="$PARAMS_PATH/flags/VioBackEnd.flags" --flagfile="$PARAMS_PATH/flags/RegularVioBackEnd.flags" --flagfile="$PARAMS_PATH/flags/Visualizer3D.flags" --logtostderr=1 --colorlogtostderr=1 --log_prefix=1 --v=0 --vmodule=Pipeline*=00 --log_output="$LOG_OUTPUT" --log_euroc_gt_data="$LOG_OUTPUT" --save_frontend_images=1 --visualize_frontend_images=1 --output_path="$OUTPUT_PATH"
sam@sam-MS-7A70:~/Kimera-VIO$ sudo bash scripts/stereoVIOEuroc.bash -p "/home/sam/V1_01_easy" > error.txt
F0615 10:57:31.471714  5709 YamlParser.h:44] Check failed: file_handle.type() != cv::FileNode::NONE (0 vs. 0) Missing parameter: maxFeaturesPerFrame in file: �f솤U
*** Check failure stack trace: ***
    @     0x7f3e9dcba0cd  google::LogMessage::Fail()
    @     0x7f3e9dcbbf33  google::LogMessage::SendToLog()
    @     0x7f3e9dcb9c28  google::LogMessage::Flush()
    @     0x7f3e9dcbc999  google::LogMessageFatal::~LogMessageFatal()
    @     0x7f3e9f099077  VIO::FeatureDetectorParams::parseYAML()
    @     0x7f3e9f084d0b  VIO::FrontendParams::parseYAML()
    @     0x7f3e9f1983ad  VIO::VioParams::parseYAML()
    @     0x7f3e9f1998c1  VIO::VioParams::VioParams()
    @     0x7f3e9f199ae3  VIO::VioParams::VioParams()
    @     0x55a48584ae05  main
    @     0x7f3e9d4beb6b  __libc_start_main
    @     0x55a48584b5aa  _start
scripts/stereoVIOEuroc.bash: line 109:  5709 Aborted                 (core dumped) $BUILD_PATH/stereoVIOEuroc --dataset_type="$DATASET_TYPE" --dataset_path="$DATASET_PATH" --initial_k=50 --final_k=2000 --params_folder_path="$PARAMS_PATH" --use_lcd="$USE_LCD" --vocabulary_path="$VOCABULARY_PATH/ORBvoc.yml" --flagfile="$PARAMS_PATH/flags/stereoVIOEuroc.flags" --flagfile="$PARAMS_PATH/flags/Mesher.flags" --flagfile="$PARAMS_PATH/flags/VioBackEnd.flags" --flagfile="$PARAMS_PATH/flags/RegularVioBackEnd.flags" --flagfile="$PARAMS_PATH/flags/Visualizer3D.flags" --logtostderr=1 --colorlogtostderr=1 --log_prefix=1 --v=0 --vmodule=Pipeline*=00 --log_output="$LOG_OUTPUT" --log_euroc_gt_data="$LOG_OUTPUT" --save_frontend_images=1 --visualize_frontend_images=1 --output_path="$OUTPUT_PATH"

```

**Additional files:**
Please attach all the files needed to reproduce the error.

Please give also the following information:
* Kimera-Vio branch, tag or commit used git clone on 6/15/2020
* GTSAM version used: git clone 6/15
* OpenGV version used:  git clone 6/15
* OpenCV version used: type `opencv_version` 3.4.10
* Operating system and version (e.g. Ubuntu 16.04 or Windows 10): ubuntu 19
* Did you change the source code? (yes / no): no

ToniRV, 2020-10-15:
@esteimle do you still have this error? It seems like it can't find the params folder.

ToniRV, 2020-10-24:
Closing for now, but feel free to re-open if it doesn't work.

closed 2020-10-24 by ToniRV

paucarre, 2021-04-04:
@ToniRV This is also happening to me. It does seem the problem is that the filename is either a dangling pointer or an uninitialized pointer.