MIT-SPARK/Kimera-VIO #94: Recommended GTSAM version number?

mcamurri, 2020-06-30:
**Description:**
Thank you for realising package opensource. This and the other works from the SPARK labs are really interesting and useful.
 
The README reports that the GTSAM version supported is `>= 4.0.0`. 
I have version `4.0.0` installed on my system as a custom Debian package, which I use for other projects. 

However, I get the following error when compiling:
**Command:**
```
make -j4
```

**Console output:**
```

/home/mcamurri/mygit/Kimera-VIO/tests/testLogger.cpp: In member function ‘virtual void VIO::BackendLoggerFixture_logBackendOutput_Test::TestBody()’:
/home/mcamurri/mygit/Kimera-VIO/tests/testLogger.cpp:136:31: error: no matching function for call to ‘gtsam::Rot3::Random(std::mt19937&)’
       gtsam::Rot3::Random(rng_), gtsam::Point3::Random());
                               ^
In file included from /usr/local/include/gtsam/geometry/Pose3.h:24:0,
                 from /usr/local/include/gtsam/geometry/CalibratedCamera.h:23,
                 from /usr/local/include/gtsam/geometry/CameraSet.h:22,
                 from /usr/local/include/gtsam/slam/RegularImplicitSchurFactor.h:10,
                 from /usr/local/include/gtsam/slam/SmartFactorBase.h:25,
                 from /usr/local/include/gtsam_unstable/slam/SmartStereoProjectionFactor.h:23,
                 from /usr/local/include/gtsam_unstable/slam/SmartStereoProjectionPoseFactor.h:23,
                 from /home/mcamurri/mygit/Kimera-VIO/include/kimera-vio/backend/VioBackEnd-definitions.h:26,
                 from /home/mcamurri/mygit/Kimera-VIO/tests/testLogger.cpp:28:
/usr/local/include/gtsam/geometry/Rot3.h:124:17: note: candidate: static gtsam::Rot3 gtsam::Rot3::Random(boost::random::mt19937&)
     static Rot3 Random(boost::mt19937 & rng);
                 ^~~~~~
/usr/local/include/gtsam/geometry/Rot3.h:124:17: note:   no known conversion for argument 1 from ‘std::mt19937 {aka std::mersenne_twister_engine<long unsigned int, 32, 624, 397, 31, 2567483615, 11, 4294967295, 7, 2636928640, 15, 4022730752, 18, 1812433253>}’ to ‘boost::random::mt19937& {aka boost::random::mersenne_twister_engine<unsigned int, 32, 624, 397, 31, 2567483615, 11, 4294967295, 7, 2636928640, 15, 4022730752, 18, 1812433253>&}’
/home/mcamurri/mygit/Kimera-VIO/tests/testLogger.cpp: In member function ‘virtual void VIO::FrontendLoggerFixture_logFrontendRansac_Test::TestBody()’:
/home/mcamurri/mygit/Kimera-VIO/tests/testLogger.cpp:328:31: error: no matching function for call to ‘gtsam::Rot3::Random(std::mt19937&)’
       gtsam::Rot3::Random(rng_), gtsam::Point3::Random());
                               ^
In file included from /usr/local/include/gtsam/geometry/Pose3.h:24:0,
                 from /usr/local/include/gtsam/geometry/CalibratedCamera.h:23,
                 from /usr/local/include/gtsam/geometry/CameraSet.h:22,
                 from /usr/local/include/gtsam/slam/RegularImplicitSchurFactor.h:10,
                 from /usr/local/include/gtsam/slam/SmartFactorBase.h:25,
                 from /usr/local/include/gtsam_unstable/slam/SmartStereoProjectionFactor.h:23,
                 from /usr/local/include/gtsam_unstable/slam/SmartStereoProjectionPoseFactor.h:23,
                 from /home/mcamurri/mygit/Kimera-VIO/include/kimera-vio/backend/VioBackEnd-definitions.h:26,
                 from /home/mcamurri/mygit/Kimera-VIO/tests/testLogger.cpp:28:
/usr/local/include/gtsam/geometry/Rot3.h:124:17: note: candidate: static gtsam::Rot3 gtsam::Rot3::Random(boost::random::mt19937&)
     static Rot3 Random(boost::mt19937 & rng);
                 ^~~~~~
/usr/local/include/gtsam/geometry/Rot3.h:124:17: note:   no known conversion for argument 1 from ‘std::mt19937 {aka std::mersenne_twister_engine<long unsigned int, 32, 624, 397, 31, 2567483615, 11, 4294967295, 7, 2636928640, 15, 4022730752, 18, 1812433253>}’ to ‘boost::random::mt19937& {aka boost::random::mersenne_twister_engine<unsigned int, 32, 624, 397, 31, 2567483615, 11, 4294967295, 7, 2636928640, 15, 4022730752, 18, 1812433253>&}’
/home/mcamurri/mygit/Kimera-VIO/tests/testLogger.cpp:330:31: error: no matching function for call to ‘gtsam::Rot3::Random(std::mt19937&)’
       gtsam::Rot3::Random(rng_), gtsam::Point3::Random());
                               ^
In file included from /usr/local/include/gtsam/geometry/Pose3.h:24:0,
                 from /usr/local/include/gtsam/geometry/CalibratedCamera.h:23,
                 from /usr/local/include/gtsam/geometry/CameraSet.h:22,
                 from /usr/local/include/gtsam/slam/RegularImplicitSchurFactor.h:10,
                 from /usr/local/include/gtsam/slam/SmartFactorBase.h:25,
                 from /usr/local/include/gtsam_unstable/slam/SmartStereoProjectionFactor.h:23,
                 from /usr/local/include/gtsam_unstable/slam/SmartStereoProjectionPoseFactor.h:23,
                 from /home/mcamurri/mygit/Kimera-VIO/include/kimera-vio/backend/VioBackEnd-definitions.h:26,
                 from /home/mcamurri/mygit/Kimera-VIO/tests/testLogger.cpp:28:
/usr/local/include/gtsam/geometry/Rot3.h:124:17: note: candidate: static gtsam::Rot3 gtsam::Rot3::Random(boost::random::mt19937&)
     static Rot3 Random(boost::mt19937 & rng);
                 ^~~~~~
/usr/local/include/gtsam/geometry/Rot3.h:124:17: note:   no known conversion for argument 1 from ‘std::mt19937 {aka std::mersenne_twister_engine<long unsigned int, 32, 624, 397, 31, 2567483615, 11, 4294967295, 7, 2636928640, 15, 4022730752, 18, 1812433253>}’ to ‘boost::random::mt19937& {aka boost::random::mersenne_twister_engine<unsigned int, 32, 624, 397, 31, 2567483615, 11, 4294967295, 7, 2636928640, 15, 4022730752, 18, 1812433253>&}’
/home/mcamurri/mygit/Kimera-VIO/tests/testLogger.cpp: In member function ‘virtual void VIO::LoopClosureDetectorLoggerFixture_logOptimizedTraj_Test::TestBody()’:
/home/mcamurri/mygit/Kimera-VIO/tests/testLogger.cpp:410:31: error: no matching function for call to ‘gtsam::Rot3::Random(std::mt19937&)’
       gtsam::Rot3::Random(rng_), gtsam::Point3::Random());
                               ^
In file included from /usr/local/include/gtsam/geometry/Pose3.h:24:0,
[... 43 lines cut ...]
/usr/local/include/gtsam/geometry/Rot3.h:124:17: note:   no known conversion for argument 1 from ‘std::mt19937 {aka std::mersenne_twister_engine<long unsigned int, 32, 624, 397, 31, 2567483615, 11, 4294967295, 7, 2636928640, 15, 4022730752, 18, 1812433253>}’ to ‘boost::random::mt19937& {aka boost::random::mersenne_twister_engine<unsigned int, 32, 624, 397, 31, 2567483615, 11, 4294967295, 7, 2636928640, 15, 4022730752, 18, 1812433253>&}’
[ 81%] Building CXX object CMakeFiles/testKimeraVIO.dir/tests/testParallelPlaneRegularBasicFactor.cpp.o
CMakeFiles/testKimeraVIO.dir/build.make:374: recipe for target 'CMakeFiles/testKimeraVIO.dir/tests/testLogger.cpp.o' failed
make[2]: *** [CMakeFiles/testKimeraVIO.dir/tests/testLogger.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
CMakeFiles/Makefile2:1006: recipe for target 'CMakeFiles/testKimeraVIO.dir/all' failed
make[1]: *** [CMakeFiles/testKimeraVIO.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2

```
This is because the GTSAM `development` branch must be transitioning from `boost` to `std`, and in this particular case the build fails because GTSAM is expecting a `boost::random::mt19937` but it is given a `std::mt19937`.

Can you please recommend a specific release version or commit hash of GTSAM that is compatible with Kimera-VIO and update the README accordingly? 
I would prefer not to clone the `development` branch of GTSAM because that changes over time.

Note that also the `4.0.2` still expects a `boost::random::mt19937`.

Please give also the following information:
* Kimera-Vio branch, tag or commit used: master
* GTSAM version used: `4.0.0`
* OpenGV version used: `master`
* OpenCV version used: `3.3.1`
* Operating system and version: `Ubuntu 18.04`
* Did you change the source code?  `no`

ToniRV, 2020-07-26:
Hi @mcamurri, thanks for reporting this! I might change from std to boost after all if 4.0.2 and 4.0.0 of gtsam stick to it.

ToniRV, 2020-07-26:
Linking to issues #82 and PR #78

ToniRV, 2020-10-24:
FYI, we are using gtsam `master` branch now.

closed 2020-10-24 by ToniRV