MIT-SPARK/Kimera-VIO #95: Mesher dies

fatrybl, 2020-07-03:
**Description:**
The mesher dies when i run Kimera on Euroc MH_05 dataset. Sometimes it works 10 seconds, sometime 2

**Command:**
bash ./scripts/stereoVIOEuroc.bash -p "/home/mark/Desktop/kimera/datasets/MH_05_difficult"
# replace this line with the command(s) you used
```

**Console output:**
 bash ./scripts/stereoTUM.bash -p "/home/mark/Desktop/kimera/datasets/MH_05_difficult"
Using dataset at path: /home/mark/Desktop/kimera/datasets/MH_05_difficult
 Launching:

            ██╗  ██╗██╗███╗   ███╗███████╗██████╗  █████╗
            ██║ ██╔╝██║████╗ ████║██╔════╝██╔══██╗██╔══██╗
            █████╔╝ ██║██╔████╔██║█████╗  ██████╔╝███████║
            ██╔═██╗ ██║██║╚██╔╝██║██╔══╝  ██╔══██╗██╔══██║
            ██║  ██╗██║██║ ╚═╝ ██║███████╗██║  ██║██║  ██║
            ╚═╝  ╚═╝╚═╝╚═╝     ╚═╝╚══════╝╚═╝  ╚═╝╚═╝  ╚═╝

 
I0703 16:01:56.053722  4900 RegularVioBackEnd.cpp:114] Using Regular VIO backend.
I0703 16:01:56.194212  4900 PipelineModule.h:450] MISO Pipeline Module: Display has no output queue registered.
I0703 16:01:56.194495  4904 EurocDataProvider.cpp:98] Parsing Euroc dataset...
I0703 16:01:56.194551  4906 Pipeline.cpp:375] Shutting down VIO pipeline once processing has finished.
I0703 16:01:56.194730  4906 Pipeline.cpp:449] Statistics
-----------                                  #	Log Hz	{avg     +- std    }	[min,max]
Display [ms]                            	    0	
backend_input_queue Size [#]            	    0	
data_provider_left_frame_queue Size [#] 	    0	
data_provider_right_frame_queue Size [#]	    0	
display_input_queue Size [#]            	    0	
mesher_backend Size [#]                 	    0	
mesher_frontend Size [#]                	    0	
stereo_frontend_input_queue Size [#]    	    0	
visualizer_backend_queue Size [#]       	    0	
visualizer_frontend_queue Size [#]      	    0	
visualizer_mesher_queue Size [#]        	    0	
I0703 16:01:56.194581  4905 Pipeline.h:184] Spinning Kimera-VIO.
I0703 16:01:56.363860  4904 EurocDataProvider.cpp:107] Running dataset between frame 50 and frame 2000
I0703 16:01:56.465602  4905 Pipeline.cpp:787] Frontend launched (parallel_run set to 1).
I0703 16:01:56.465659  4905 Pipeline.cpp:580] ------------------- Initialize Pipeline with frame k = 51--------------------
I0703 16:01:56.695399  4906 Pipeline.cpp:449] Statistics
-----------                                  #	Log Hz	{avg     +- std    }	[min,max]
Data Provider [ms]                      	    0	
Display [ms]                            	    0	
VioFrontEnd [ms]                        	    0	
backend_input_queue Size [#]            	    0	
data_provider_left_frame_queue Size [#] 	    6	8.99924	{2.00000 +- 1.26491}	[1,4]
data_provider_right_frame_queue Size [#]	    5	7.86014	{1.60000 +- 0.894427}	[1,3]
display_input_queue Size [#]            	    0	
mesher_backend Size [#]                 	    0	
mesher_frontend Size [#]                	    0	
stereo_frontend_input_queue Size [#]    	    0	
visualizer_backend_queue Size [#]       	    0	
visualizer_frontend_queue Size [#]      	    0	
visualizer_mesher_queue Size [#]        	    0	
I0703 16:01:57.195822  4906 Pipeline.cpp:449] Statistics
-----------                                  #	Log Hz	{avg     +- std    }	[min,max]
Data Provider [ms]                      	    0	
Display [ms]                            	    0	
VioFrontEnd [ms]                        	    0	
backend_input_queue Size [#]            	    0	
data_provider_left_frame_queue Size [#] 	   13	11.5786	{5.23077 +- 3.56263}	[1,11]
data_provider_right_frame_queue Size [#]	   13	11.2076	{5.23077 +- 3.56263}	[1,11]
display_input_queue Size [#]            	    0	
mesher_backend Size [#]                 	    0	
mesher_frontend Size [#]                	    0	
stereo_frontend_input_queue Size [#]    	    0	
visualizer_backend_queue Size [#]       	    0	
[... 129 lines cut ...]
    @     0x7fc4f1f79b0d  google::LogMessage::Fail()
    @     0x7fc4f1f7b9b1  google::LogMessage::SendToLog()
    @     0x7fc4f1f7963d  google::LogMessage::Flush()
    @     0x7fc4f1f7c389  google::LogMessageFatal::~LogMessageFatal()
    @     0x7fc4f34f9392  VIO::VisualizerModule::getInputPacket()
    @     0x7fc4f34b2f2e  VIO::PipelineModule<>::spin()
    @     0x7fc4f144e6df  (unknown)
    @     0x7fc4f17216db  start_thread
    @     0x7fc4f0ea988f  clone
./scripts/stereoTUM.bash: line 109:  4900 Aborted                 (core dumped) $BUILD_PATH/stereoVIOEuroc --dataset_type="$DATASET_TYPE" --dataset_path="$DATASET_PATH" --initial_k=50 --final_k=2000 --params_folder_path="$PARAMS_PATH" --use_lcd="$USE_LCD" --vocabulary_path="$VOCABULARY_PATH/ORBvoc.yml" --flagfile="$PARAMS_PATH/flags/stereoVIOEuroc.flags" --flagfile="$PARAMS_PATH/flags/Mesher.flags" --flagfile="$PARAMS_PATH/flags/VioBackEnd.flags" --flagfile="$PARAMS_PATH/flags/RegularVioBackEnd.flags" --flagfile="$PARAMS_PATH/flags/Visualizer3D.flags" --logtostderr=1 --colorlogtostderr=1 --log_prefix=1 --v=0 --vmodule=Pipeline*=00 --log_output="$LOG_OUTPUT" --log_euroc_gt_data="$LOG_OUTPUT" --save_frontend_images=1 --visualize_frontend_images=1 --output_path="$OUTPUT_PATH"

# remove this line and paste your console output HERE - no screenshots please

```

**Additional files:**
Please attach all the files needed to reproduce the error.

Please give also the following information:
* Kimera-Vio branch, tag or commit used
* GTSAM version used:
* OpenGV version used:
* OpenCV version used: type `opencv_version`
* Operating system and version (e.g. Ubuntu 16.04 or Windows 10): 
* Did you change the source code? (yes / no):

ToniRV, 2020-07-03:
Hi @fatrybl does that happen consistently on each run?

fatrybl, 2020-07-03:
> 
> 
> Hi @fatrybl does that happen consistently on each run?

Yes, unfortunatelly, every time. Here is the viedo demonstration on V1_01_easy dataset. But with machine hall the situation is the same
[video](https://yadi.sk/i/E7czBt4ZsPiyeA)

ToniRV, 2020-07-03:
Thanks for the video! I don't think it is the mesher that dies?
Can you try downloading and running our Euroc dataset?
I think it's because Euroc has some datasets for which there is no right frame for a left frame...
Also, their timestamps might be unordered.
Checkout this: 
> Datasets MH_04 and V2_03 have different number of left/right frames. We suggest using instead our version of Euroc here.

I think V1_01_easy should work out of the box though...

fatrybl, 2020-07-04:
> Thanks for the video! I don't think it is the mesher that dies?
> Can you try downloading and running our Euroc dataset?
> I think it's because Euroc has some datasets for which there is no right frame [...]

I`m already using your version of Euroc. I downloaded from your google drive. So this error comes with your dataset unfortunatelly

fatrybl, 2020-07-10:
> Thanks for the video! I don't think it is the mesher that dies?
> Can you try downloading and running our Euroc dataset?
> I think it's because Euroc has some datasets for which there is no right frame [...]

The problem has been solved. I ran Kimera on Virtual Machine (VMware) and this was an issue. Reran it on PC with ubuntu 18 and the problem disappered

closed 2020-10-24 by ToniRV