MIT-SPARK/Kimera-VIO #96: Explanation autoInitialize set to 2

rocomanmelli, 2020-07-08:
Hi!

First of all, thank you for sharing Kimera-VIO with the community. I have done some tests with another dataset and it seems to work pretty well.

Sadly, some sequences of this dataset don't start stationary and I suspect that using autoInitialize set to 2 could solve the problem (in other cases I used it set to 1). However, I'm getting "Check failed: frontend_output" and, the truth is that, I don't understand how autoinitialization works. When autoInitialize is set to 0, it uses data from ground truth, when is set to 1, it uses the current IMU measurements I guess, and when it is set to 2, how is the initialization done? I don't expect a long answer, just some ideas to find out more by myself.

And... Is there a way to run Kimera-VIO when the data doesn't correspond to a stationary state at the beginning?

Thank you in advance.

ToniRV, 2020-07-26:
Hi @rocomanmelli,

It is a bit tricky indeed to initialize the pipeline when the dataset is not static.
Autoinitialize:
0: is using ground-truth
1: is using IMU measurements assuming static
2: is using nonlinear optimization using both IMU and Visual information. I am currently modifying the initialization to make it play nicely with the overall Kimera architecture, so I won't be able to fix that issue right now, but I would bet it gets solved once I push these changes.

closed 2020-07-26 by ToniRV

rocomanmelli, 2020-07-26:
Thank you, Toni!