MIT-SPARK/Kimera-VIO #97: Docker image installation

YiChenCityU, 2020-07-08:
**Description:**
I want to use docker to run the system, but I got this error.
**Command:**
```
sudo docker build --rm -t kimera_vio -f ./scripts/docker/Dockerfile .

```

**Console output:**
```

Sending build context to Docker daemon  311.5MB
Step 1/43 : FROM ubuntu:18.04
 ---> d27b9ffc5667
Step 2/43 : MAINTAINER Antoni Rosinol "arosinol@mit.edu"
 ---> Using cache
 ---> 3e06ad1a42f3
Step 3/43 : ENV DEBIAN_FRONTEND noninteractive
 ---> Using cache
 ---> 3da07689ac54
Step 4/43 : ENV DIRPATH /root
 ---> Using cache
 ---> af5cded217bd
Step 5/43 : WORKDIR $DIRPATH
 ---> Using cache
 ---> c0e75e7a0c79
Step 6/43 : RUN apt-get update && apt-get install -y --no-install-recommends apt-utils
 ---> Using cache
 ---> cda1a800f0c9
Step 7/43 : RUN apt-get update && apt-get install -y git cmake
 ---> Using cache
 ---> cb9df84a0e3e
Step 8/43 : RUN apt-get update && apt-get install -y xvfb
 ---> Using cache
 ---> 2d33c5188a5c
Step 9/43 : RUN apt-get update && apt-get install -y libboost-all-dev
 ---> Using cache
 ---> 46b26fd4c86f
Step 10/43 : RUN git clone https://github.com/borglab/gtsam.git
 ---> Running in 2b16930dd252
Cloning into 'gtsam'...
error: RPC failed; curl 18 transfer closed with outstanding read data remaining
fatal: The remote end hung up unexpectedly
fatal: early EOF
fatal: index-pack failed
The command '/bin/sh -c git clone https://github.com/borglab/gtsam.git' returned a non-zero code: 128

```

**Additional files:**
Please attach all the files needed to reproduce the error.

Please give also the following information:
* Kimera-Vio branch, tag or commit used
* GTSAM version used:
* OpenGV version used:
* OpenCV version used: type `opencv_version`
* Operating system and version (e.g. Ubuntu 16.04 or Windows 10): 
* Did you change the source code? (yes / no):

zainmehdi, 2020-07-28:
Hi @ToniRV any update on it. I am also unable to build it.

gaotao19890725, 2020-08-09:
According to the author's method, I am also unable to build it. Anyone who can help me.

ToniRV, 2020-10-15:
@YiChenCityU I think that it is just that gtsam is a bit heavy, so if your connection is not stable, it may break? Not sure though...

ToniRV, 2020-10-24:
Looks like these two issues are related, https://github.com/MIT-SPARK/Kimera-VIO/issues/107
My bet is that it is some sort of connection error?
Closing for now.

closed 2020-10-24 by ToniRV