MIT-SPARK/Kimera-VIO #107: docker error about opencv

gaotao19890725, 2020-08-20:
**Description:**
Step 13/43 : RUN git clone https://github.com/opencv/opencv.git
 ---> Running in 8f5f605d2c0b
Cloning into 'opencv'...
Checking out files: 100% (6503/6503), done.
Removing intermediate container 8f5f605d2c0b
 ---> 7eaea3ebaa71
Step 14/43 : RUN cd opencv &&       git checkout tags/3.3.1 &&       mkdir build
 ---> Running in 5443c0a44da5
Note: checking out 'tags/3.3.1'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by performing another checkout.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -b with the checkout command again. Example:

  git checkout -b <new-branch-name>

HEAD is now at a871f9e4f7 Merge branch 'update_version' into release
Removing intermediate container 5443c0a44da5
 ---> 871cfa63a6d9
Step 15/43 : RUN git clone https://github.com/opencv/opencv_contrib.git
 ---> Running in ab7ec730c58d
Cloning into 'opencv_contrib'...
error: RPC failed; curl 56 GnuTLS recv error (-54): Error in the pull function.
fatal: The remote end hung up unexpectedly
fatal: early EOF
fatal: index-pack failed
The command '/bin/sh -c git clone https://github.com/opencv/opencv_contrib.git' returned a non-zero code: 128
gaotao@gaotao-desktop:~/data/Kimera-VIO-docker$ ^C

**Command:**
```
# replace this line with the command(s) you used
```

**Console output:**
```

# remove this line and paste your console output HERE - no screenshots please

```

**Additional files:**
Please attach all the files needed to reproduce the error.

Please give also the following information:
* Kimera-Vio branch, tag or commit used
* GTSAM version used:
* OpenGV version used:
* OpenCV version used: type `opencv_version`
* Operating system and version (e.g. Ubuntu 16.04 or Windows 10): Ubuntu 18.04
* Did you change the source code? (yes / no): No

ToniRV, 2020-10-24:
That seems more like a connection error? Could you figure out what it was?

ToniRV, 2020-10-24:
Closing for now, as it seems to be independent from Kimera-VIO this issue, see #97

closed 2020-10-24 by ToniRV