MIT-SPARK/Kimera-VIO #111: Docker Build Issue (Ubuntu 18.04)

fishberg, 2020-09-25:
**Description:**
Step 29/43 fails during Docker build on Ubuntu 18.04. Appears to be a Python versioning issue within the Docker image?

Specific Error:
```
pyrsistent requires Python '>=3.5' but the running Python is 2.7.17
```

Following the [docker build instructions](https://github.com/MIT-SPARK/Kimera-VIO/blob/master/docs/kimera_vio_install.md#From-Dockerfile). Only modification to the instructions was cloning with https instead of ssh. Kimera code was not modified.

**Command:**
```
sudo docker build --rm -t kimera_vio -f ./scripts/docker/Dockerfile .
```

**Console output:**
Note: Only showing console output starting at Step 29/43.
```
Step 29/43 : RUN cd spark_vio_evaluation && pip install .
 ---> Running in 5a36effa245d
Processing /root/spark_vio_evaluation
Collecting chart_studio (from spark-vio-evaluation==0.2)
  Downloading https://files.pythonhosted.org/packages/d3/06/cfe81ee02b37eeacb3c74efd2a3280d68d97835c6d55419f89d5a3f63934/chart-studio-1.1.0.tar.gz (51kB)
Collecting evo-1 (from spark-vio-evaluation==0.2)
  Downloading https://files.pythonhosted.org/packages/0b/ef/c79a37588548872a512b9d9618930f0573e615225799fab2826603744ad7/evo-1-1.1.2.tar.gz (118kB)
Collecting glog (from spark-vio-evaluation==0.2)
  Downloading https://files.pythonhosted.org/packages/39/42/d2502683e9d9086cba5fb741c9b66c72ab1a852d046e1d54d2b80221b69f/glog-0.3.1-py2.py3-none-any.whl
Collecting numpy (from spark-vio-evaluation==0.2)
  Downloading https://files.pythonhosted.org/packages/3a/5f/47e578b3ae79e2624e205445ab77a1848acdaa2929a00eeef6b16eaaeb20/numpy-1.16.6-cp27-cp27mu-manylinux1_x86_64.whl (17.0MB)
Collecting open3d-python (from spark-vio-evaluation==0.2)
  Downloading https://files.pythonhosted.org/packages/5f/5c/a86082dc5efc3d22585e8aa22f9840667d9faa5e727b47c43137090caed4/open3d_python-0.7.0.0-cp27-cp27mu-manylinux1_x86_64.whl (3.7MB)
Collecting pandas (from spark-vio-evaluation==0.2)
  Downloading https://files.pythonhosted.org/packages/db/83/7d4008ffc2988066ff37f6a0bb6d7b60822367dcb36ba5e39aa7801fda54/pandas-0.24.2-cp27-cp27mu-manylinux1_x86_64.whl (10.1MB)
Collecting plotly (from spark-vio-evaluation==0.2)
  Downloading https://files.pythonhosted.org/packages/04/20/c2d77eef33dbd40c5e3263a9a8763ffca610a4c3d2b2da21c5601e5fc5d8/plotly-4.10.0-py2.py3-none-any.whl (13.0MB)
Collecting ruamel.yaml (from spark-vio-evaluation==0.2)
  Downloading https://files.pythonhosted.org/packages/7e/39/186f14f3836ac5d2a6a042c8de69988770e8b9abb537610edc429e4914aa/ruamel.yaml-0.16.12-py2.py3-none-any.whl (111kB)
Collecting tqdm (from spark-vio-evaluation==0.2)
  Downloading https://files.pythonhosted.org/packages/73/d5/f220e0c69b2f346b5649b66abebb391df1a00a59997a7ccf823325bd7a3e/tqdm-4.49.0-py2.py3-none-any.whl (69kB)
Collecting requests (from chart_studio->spark-vio-evaluation==0.2)
  Downloading https://files.pythonhosted.org/packages/45/1e/0c169c6a5381e241ba7404532c16a21d86ab872c9bed8bdcd4c423954103/requests-2.24.0-py2.py3-none-any.whl (61kB)
Collecting retrying>=1.3.3 (from chart_studio->spark-vio-evaluation==0.2)
  Downloading https://files.pythonhosted.org/packages/44/ef/beae4b4ef80902f22e3af073397f079c96969c69b2c7d52a57ea9ae61c9d/retrying-1.3.3.tar.gz
Requirement already satisfied: six in /usr/lib/python2.7/dist-packages (from chart_studio->spark-vio-evaluation==0.2)
Collecting argcomplete (from evo-1->spark-vio-evaluation==0.2)
  Downloading https://files.pythonhosted.org/packages/89/4d/b8e035cca2c9b2484ac12d20e0fb68019e17f0b09918f2765e0a381127fb/argcomplete-1.12.0-py2.py3-none-any.whl
Requirement already satisfied: colorama in /usr/lib/python2.7/dist-packages (from evo-1->spark-vio-evaluation==0.2)
Requirement already satisfied: enum34 in /usr/lib/python2.7/dist-packages (from evo-1->spark-vio-evaluation==0.2)
Collecting matplotlib (from evo-1->spark-vio-evaluation==0.2)
  Downloading https://files.pythonhosted.org/packages/9d/40/5ba7d4a3f80d39d409f21899972596bf62c8606f1406a825029649eaa439/matplotlib-2.2.5-cp27-cp27mu-manylinux1_x86_64.whl (12.8MB)
Collecting natsort (from evo-1->spark-vio-evaluation==0.2)
  Downloading https://files.pythonhosted.org/packages/4a/4e/ead3dd6b7c3b6011c04ed9e526b53a2e2b96f1205fb207f73374cb015810/natsort-6.2.1-py2.py3-none-any.whl
Collecting pygments (from evo-1->spark-vio-evaluation==0.2)
  Downloading https://files.pythonhosted.org/packages/be/39/32da3184734730c0e4d3fa3b2b5872104668ad6dc1b5a73d8e477e5fe967/Pygments-2.5.2-py2.py3-none-any.whl (896kB)
Collecting pyyaml (from evo-1->spark-vio-evaluation==0.2)
  Downloading https://files.pythonhosted.org/packages/64/c2/b80047c7ac2478f9501676c988a5411ed5572f35d1beff9cae07d321512c/PyYAML-5.3.1.tar.gz (269kB)
Collecting scipy (from evo-1->spark-vio-evaluation==0.2)
  Downloading https://files.pythonhosted.org/packages/24/40/11b12af7f322c1e20446c037c47344d89bab4922b8859419d82cf56d796d/scipy-1.2.3-cp27-cp27mu-manylinux1_x86_64.whl (24.8MB)
Collecting seaborn (from evo-1->spark-vio-evaluation==0.2)
  Downloading https://files.pythonhosted.org/packages/b2/86/43b8c9138ef4c2a1c492fee92792c83c13799d0e2061ff810d3826d06cd1/seaborn-0.9.1-py2.py3-none-any.whl (216kB)
Collecting python-gflags>=3.1 (from glog->spark-vio-evaluation==0.2)
  Downloading https://files.pythonhosted.org/packages/df/ec/e31302d355bcb9d207d9b858adc1ecc4a6d8c855730c8ba4ddbdd3f8eb8d/python-gflags-3.1.2.tar.gz (52kB)
Collecting notebook (from open3d-python->spark-vio-evaluation==0.2)
  Downloading https://files.pythonhosted.org/packages/7a/fb/6b1735e8ff43f68e867928526134cd6ba22554d596862a7fe71092ba8fc8/notebook-5.7.10-py2.py3-none-any.whl (9.6MB)
Collecting widgetsnbextension (from open3d-python->spark-vio-evaluation==0.2)
  Downloading https://files.pythonhosted.org/packages/6c/7b/7ac231c20d2d33c445eaacf8a433f4e22c60677eb9776c7c5262d7ddee2d/widgetsnbextension-3.5.1-py2.py3-none-any.whl (2.2MB)
Collecting ipywidgets (from open3d-python->spark-vio-evaluation==0.2)
  Downloading https://files.pythonhosted.org/packages/56/a0/dbcf5881bb2f51e8db678211907f16ea0a182b232c591a6d6f276985ca95/ipywidgets-7.5.1-py2.py3-none-any.whl (121kB)
Collecting pytz>=2011k (from pandas->spark-vio-evaluation==0.2)
  Downloading https://files.pythonhosted.org/packages/4f/a4/879454d49688e2fad93e59d7d4efda580b783c745fd2ec2a3adf87b0808d/pytz-2020.1-py2.py3-none-any.whl (510kB)
[... 87 lines cut ...]
  Downloading https://files.pythonhosted.org/packages/c5/10/369f50bcd4621b263927b0a1519987a04383d4a98fb10438042ad410cf88/singledispatch-3.4.0.3-py2.py3-none-any.whl
Collecting simplegeneric>0.8 (from ipython<6.0.0,>=4.0.0; python_version < "3.3"->ipywidgets->open3d-python->spark-vio-evaluation==0.2)
  Downloading https://files.pythonhosted.org/packages/3d/57/4d9c9e3ae9a255cd4e1106bb57e24056d3d0709fc01b2e3e345898e49d5b/simplegeneric-0.8.1.zip
Collecting pickleshare (from ipython<6.0.0,>=4.0.0; python_version < "3.3"->ipywidgets->open3d-python->spark-vio-evaluation==0.2)
  Downloading https://files.pythonhosted.org/packages/9a/41/220f49aaea88bc6fa6cba8d05ecf24676326156c23b991e80b3f2fc24c77/pickleshare-0.7.5-py2.py3-none-any.whl
Collecting backports.shutil-get-terminal-size; python_version == "2.7" (from ipython<6.0.0,>=4.0.0; python_version < "3.3"->ipywidgets->open3d-python->spark-vio-evaluation==0.2)
  Downloading https://files.pythonhosted.org/packages/7d/cd/1750d6c35fe86d35f8562091737907f234b78fdffab42b29c72b1dd861f4/backports.shutil_get_terminal_size-1.0.0-py2.py3-none-any.whl
Collecting pexpect; sys_platform != "win32" (from ipython<6.0.0,>=4.0.0; python_version < "3.3"->ipywidgets->open3d-python->spark-vio-evaluation==0.2)
  Downloading https://files.pythonhosted.org/packages/39/7b/88dbb785881c28a102619d46423cb853b46dbccc70d3ac362d99773a78ce/pexpect-4.8.0-py2.py3-none-any.whl (59kB)
Collecting prompt-toolkit<2.0.0,>=1.0.4 (from ipython<6.0.0,>=4.0.0; python_version < "3.3"->ipywidgets->open3d-python->spark-vio-evaluation==0.2)
  Downloading https://files.pythonhosted.org/packages/9d/d2/2f099b5cd62dab819ce7a9f1431c09a9032fbfbb6474f442722e88935376/prompt_toolkit-1.0.18-py2-none-any.whl (245kB)
Collecting scandir; python_version < "3.5" (from pathlib2; python_version < "3"->importlib-metadata<2,>=0.23; python_version == "2.7"->argcomplete->evo-1->spark-vio-evaluation==0.2)
  Downloading https://files.pythonhosted.org/packages/df/f5/9c052db7bd54d0cbf1bc0bb6554362bba1012d03e5888950a4f5c5dadc4e/scandir-1.10.0.tar.gz
Collecting webencodings (from bleach->nbconvert<6.0->notebook->open3d-python->spark-vio-evaluation==0.2)
[... cut ...]
  Downloading https://files.pythonhosted.org/packages/c5/60/6ac26ad05857c601308d8fb9e87fa36d0ebf889423f47c3502ef034365db/functools32-3.2.3-2.tar.gz
Collecting pyrsistent>=0.14.0 (from jsonschema!=2.5.0,>=2.4->nbformat->notebook->open3d-python->spark-vio-evaluation==0.2)
  Downloading https://files.pythonhosted.org/packages/7d/ae/90ddcf28fb8eee5d4990920586d2856342e42faa95f39223f0b9762ef264/pyrsistent-0.17.2.tar.gz (106kB)
pyrsistent requires Python '>=3.5' but the running Python is 2.7.17
The command '/bin/sh -c cd spark_vio_evaluation && pip install .' returned a non-zero code: 1
```

ToniRV, 2020-10-04:
Hi @fishberg, this dockerfile is a bit outdated now, the newest dockerfiles are https://github.com/MIT-SPARK/Kimera-VIO/blob/master/Dockerfile_16_04 and https://github.com/MIT-SPARK/Kimera-VIO/blob/master/Dockerfile_18_04
Unfortunately, these two only install the Kimera dependencies for Jenkins (our CI server) but not Kimera-VIO itself.
I guess I should make a docker base image and have a dockerfile import one of the two above + install Kimera-VIO.

ToniRV, 2020-10-15:
Should be fixed in @119

closed 2020-10-24 by ToniRV

Cheng-Hsiung, 2020-11-30:
Hi @ToniRV , I also get the `pyrsistent requires Python '>=3.5' but the running Python is 2.7.17` by following the docker image building instruction [here](https://github.com/MIT-SPARK/Kimera-VIO/blob/master/docs/kimera_vio_install.md#From-Dockerfile) (commit f7748e10). May I ask what is the right step to build the docker image at this moment? Is the instruction outdated, and should switch to this [docker file](https://github.com/MIT-SPARK/Kimera-VIO/blob/master/Dockerfile_18_04)?

ToniRV, 2020-11-30:
See #133, and fixes are in https://github.com/MIT-SPARK/Kimera-VIO/pull/135

Cheng-Hsiung, 2020-11-30:
Thanks for the update. I also tried replacing the [snippet of code ](https://github.com/MIT-SPARK/Kimera-VIO/blob/f7748e1040bfa5d5391fcc6d6653b9979977cc54/scripts/docker/Dockerfile#L86-L91)in `Dockerfile` with [the other one](https://github.com/MIT-SPARK/Kimera-VIO/blob/f7748e1040bfa5d5391fcc6d6653b9979977cc54/Dockerfile_18_04#L86-L107) in `Dockerfile_18_04`, then the docker image can be built successfully. Is this modification valid?

ToniRV, 2020-11-30:
Sure, feel free to push that modification and we'll merge that.

Cheng-Hsiung, 2020-11-30:
Thanks, please check #137

PR #119 (2020-10-24): Update Dockerfile
@lucacarlone 

Fixes:
#111