MIT-SPARK/Kimera-VIO #115: Unable to build Kimera-VIO from source

yrq0208, 2020-10-04:
**Description:**
error: using typedef-name ‘gtsam::Point2’ after ‘class’, error: using typedef-name ‘gtsam::Point3’ after ‘class’

**Command:**
make -j4

**Console output:**
(base) ruiqi@ruiqi-GP62MVR-6RF:~/Kimera-VIO/build$ make -j4
Scanning dependencies of target gtest
Scanning dependencies of target kimera_vio
[  1%] Building CXX object external/googletest-build/googletest/CMakeFiles/gtest.dir/src/gtest-all.cc.o
[  2%] Building CXX object CMakeFiles/kimera_vio.dir/src/dataprovider/DataProviderInterface-definitions.cpp.o
[  3%] Building CXX object CMakeFiles/kimera_vio.dir/src/dataprovider/DataProviderModule.cpp.o
[  4%] Building CXX object CMakeFiles/kimera_vio.dir/src/dataprovider/DataProviderInterface.cpp.o
In file included from /home/ruiqi/Kimera-VIO/include/kimera-vio/frontend/CameraParams.h:27,
                 from /home/ruiqi/Kimera-VIO/include/kimera-vio/frontend/Camera.h:25,
                 from /home/ruiqi/Kimera-VIO/include/kimera-vio/dataprovider/DataProviderInterface.h:20,
                 from /home/ruiqi/Kimera-VIO/src/dataprovider/DataProviderInterface.cpp:15:
/home/ruiqi/Kimera-VIO/include/kimera-vio/utils/UtilsOpenCV.h:39:7: error: using typedef-name ‘gtsam::Point2’ after ‘class’
 class Point2;
       ^~~~~~
In file included from /usr/local/include/gtsam/geometry/Cal3_S2.h:24,
                 from /home/ruiqi/Kimera-VIO/include/kimera-vio/frontend/Camera.h:21,
                 from /home/ruiqi/Kimera-VIO/include/kimera-vio/dataprovider/DataProviderInterface.h:20,
                 from /home/ruiqi/Kimera-VIO/src/dataprovider/DataProviderInterface.cpp:15:
/usr/local/include/gtsam/geometry/Point2.h:27:17: note: ‘gtsam::Point2’ has a previous declaration here
 typedef Vector2 Point2;
                 ^~~~~~
In file included from /home/ruiqi/Kimera-VIO/include/kimera-vio/frontend/CameraParams.h:27,
                 from /home/ruiqi/Kimera-VIO/include/kimera-vio/frontend/Camera.h:25,
                 from /home/ruiqi/Kimera-VIO/include/kimera-vio/dataprovider/DataProviderInterface.h:20,
                 from /home/ruiqi/Kimera-VIO/src/dataprovider/DataProviderInterface.cpp:15:
/home/ruiqi/Kimera-VIO/include/kimera-vio/utils/UtilsOpenCV.h:40:7: error: using typedef-name ‘gtsam::Point3’ after ‘class’
 class Point3;
       ^~~~~~
In file included from /usr/local/include/gtsam/geometry/Pose3.h:23,
                 from /usr/local/include/gtsam/geometry/CalibratedCamera.h:23,
                 from /usr/local/include/gtsam/geometry/PinholePose.h:22,
                 from /usr/local/include/gtsam/geometry/PinholeCamera.h:21,
                 from /home/ruiqi/Kimera-VIO/include/kimera-vio/frontend/Camera.h:22,
                 from /home/ruiqi/Kimera-VIO/include/kimera-vio/dataprovider/DataProviderInterface.h:20,
                 from /home/ruiqi/Kimera-VIO/src/dataprovider/DataProviderInterface.cpp:15:
/usr/local/include/gtsam/geometry/Point3.h:35:17: note: ‘gtsam::Point3’ has a previous declaration here
 typedef Vector3 Point3;
                 ^~~~~~
[  4%] Building CXX object CMakeFiles/kimera_vio.dir/src/dataprovider/EurocDataProvider.cpp.o
[  5%] Linking CXX static library ../../../lib/libgtest.a
[  5%] Built target gtest
[  6%] Building CXX object CMakeFiles/kimera_vio.dir/src/dataprovider/KittiDataProvider.cpp.o
In file included from /home/ruiqi/Kimera-VIO/include/kimera-vio/frontend/CameraParams.h:27,
                 from /home/ruiqi/Kimera-VIO/include/kimera-vio/frontend/Tracker.h:26,
                 from /home/ruiqi/Kimera-VIO/include/kimera-vio/frontend/StereoVisionFrontEnd-definitions.h:19,
                 from /home/ruiqi/Kimera-VIO/include/kimera-vio/backend/VioBackEnd-definitions.h:30,
                 from /home/ruiqi/Kimera-VIO/include/kimera-vio/frontend/StereoImuSyncPacket.h:22,
                 from /home/ruiqi/Kimera-VIO/include/kimera-vio/dataprovider/DataProviderModule.h:29,
                 from /home/ruiqi/Kimera-VIO/src/dataprovider/DataProviderModule.cpp:16:
/home/ruiqi/Kimera-VIO/include/kimera-vio/utils/UtilsOpenCV.h:39:7: error: using typedef-name ‘gtsam::Point2’ after ‘class’
 class Point2;
       ^~~~~~
In file included from /usr/local/include/gtsam/geometry/Unit3.h:23,
                 from /usr/local/include/gtsam/geometry/Rot3.h:26,
                 from /usr/local/include/gtsam/geometry/Pose3.h:24,
                 from /home/ruiqi/Kimera-VIO/include/kimera-vio/frontend/StereoImuSyncPacket.h:20,
                 from /home/ruiqi/Kimera-VIO/include/kimera-vio/dataprovider/DataProviderModule.h:29,
                 from /home/ruiqi/Kimera-VIO/src/dataprovider/DataProviderModule.cpp:16:
/usr/local/include/gtsam/geometry/Point2.h:27:17: note: ‘gtsam::Point2’ has a previous declaration here
 typedef Vector2 Point2;
                 ^~~~~~
In file included from /home/ruiqi/Kimera-VIO/include/kimera-vio/frontend/CameraParams.h:27,
                 from /home/ruiqi/Kimera-VIO/include/kimera-vio/frontend/Tracker.h:26,
[... 113 lines cut ...]
CMakeFiles/kimera_vio.dir/build.make:86: recipe for target 'CMakeFiles/kimera_vio.dir/src/dataprovider/DataProviderInterface.cpp.o' failed
make[2]: *** [CMakeFiles/kimera_vio.dir/src/dataprovider/DataProviderInterface.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
CMakeFiles/kimera_vio.dir/build.make:110: recipe for target 'CMakeFiles/kimera_vio.dir/src/dataprovider/DataProviderModule.cpp.o' failed
make[2]: *** [CMakeFiles/kimera_vio.dir/src/dataprovider/DataProviderModule.cpp.o] Error 1
CMakeFiles/kimera_vio.dir/build.make:134: recipe for target 'CMakeFiles/kimera_vio.dir/src/dataprovider/EurocDataProvider.cpp.o' failed
make[2]: *** [CMakeFiles/kimera_vio.dir/src/dataprovider/EurocDataProvider.cpp.o] Error 1
/home/ruiqi/Kimera-VIO/src/dataprovider/KittiDataProvider.cpp: In member function ‘virtual bool VIO::KittiDataProvider::spin()’:
/home/ruiqi/Kimera-VIO/src/dataprovider/KittiDataProvider.cpp:54:17: warning: control reaches end of non-void function [-Wreturn-type]
   LOG(FATAL) << "Not implemented.";
                 ^~~~~~~~~~~~~~~~~~
CMakeFiles/kimera_vio.dir/build.make:158: recipe for target 'CMakeFiles/kimera_vio.dir/src/dataprovider/KittiDataProvider.cpp.o' failed
make[2]: *** [CMakeFiles/kimera_vio.dir/src/dataprovider/KittiDataProvider.cpp.o] Error 1
CMakeFiles/Makefile2:291: recipe for target 'CMakeFiles/kimera_vio.dir/all' failed
make[1]: *** [CMakeFiles/kimera_vio.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2

Please give also the following information:
* Kimera-Vio branch, tag or commit used: I simply git clone from here: https://github.com/MIT-SPARK/Kimera-VIO
* GTSAM version used: latest I suppose, git clone from here: https://github.com/borglab/gtsam
* OpenGV version used: latest I suppose, just git clone from here: https://github.com/laurentkneip/opengv
* OpenCV version used: 3.4.3
* Operating system and version (e.g. Ubuntu 16.04 or Windows 10): Ubuntu 16.04
* Did you change the source code? (yes / no): no

yrq0208, 2020-10-04:
I am following this page, https://github.com/MIT-SPARK/Kimera-VIO/blob/master/docs/kimera_vio_install.md

ToniRV, 2020-10-04:
Hi @yrq0208, what branch are you using for GTSAM? Just to check, but it should be `master`.

yrq0208, 2020-10-04:
It seems I used the "develop" branch, I will try the master one.

yrq0208, 2020-10-06:
Hi @ToniRV,

I just tried using the master branch of GTSAM, and I still got the same "error: using typedef-name ‘gtsam::Point2’ after ‘class’, error: using typedef-name ‘gtsam::Point3’ after ‘class’" error when building Kimera-VIO from source. Could you please shed some more lights on this?

Thank you in advance.

ToniRV, 2020-10-12:
I have seen this error before, but I thought I already fixed it.
Right now I can't reproduce it though: Jenkins doesn't seem to fail either (which builds gtsam using https://github.com/MIT-SPARK/Kimera-VIO/blob/master/Dockerfile_18_04)
Can you pull Kimera's latest master branch and gtsam's master as well?
Thanks for your help!

yrq0208, 2020-10-13:
Hi @ToniRV,

Thanks for your reply. 

Unfornately I think the problem still exists. I have done a git pull for gtsam and Kimera-VIO respectively, Kimera-VIO seems to be up to date, gtsam doesn't, I can see something has been added due to the output. I then build gtsam again. But when I try to build gtsam, the exact same error rise again.

yrq0208, 2020-10-13:
Hi @ToniRV,

Thanks for your reply.

Unfortunately I think the problem still exists. I have done a git pull for gtsam and Kimera-VIO respectively, Kimera-VIO seems to be up to date, gtsam doesn't, I can see something has been added due to the output. I then build gtsam again. But when I try to build gtsam, the exact same error rise again.

ToniRV, 2020-10-24:
Hi @yrq0208 , that's interesting, I can't seem to be able to reproduce this error.
Just to be sure, are you `sudo install`ing gtsam again? (rather than just building it?)

yrq0208, 2020-10-25:
Hi @ToniRV, 

Thanks for the reply.

This is what I just done, I go to the ...../gtsam/build directory and do a sudo make uninstall, then I delete the gtsam directory. I then go to https://github.com/borglab/gtsam/tree/master to git clone gtsam again and do a git pull to make sure it is up to date. Then I run "cmake -DCMAKE_INSTALL_PREFIX=/usr/local -DCMAKE_BUILD_TYPE=Release -DGTSAM_USE_SYSTEM_EIGEN=OFF -DGTSAM_POSE3_EXPMAP=ON -DGTSAM_ROT3_EXPMAP=ON ..", make -j4 check and sudo make -j4 install. After that I go to ...../Kimera-VIO/build, do a git pull first and then cmake.. and make, then I got the following errors:

(base) ruiqi@ruiqi-GP62MVR-6RF:~/Kimera-VIO/build$ make -j4
[  2%] Built target gtest
[  3%] Building CXX object CMakeFiles/kimera_vio.dir/src/dataprovider/DataProviderModule.cpp.o
[  4%] Building CXX object CMakeFiles/kimera_vio.dir/src/dataprovider/KittiDataProvider.cpp.o
[  5%] Building CXX object CMakeFiles/kimera_vio.dir/src/dataprovider/DataProviderInterface.cpp.o
[  5%] Building CXX object CMakeFiles/kimera_vio.dir/src/dataprovider/EurocDataProvider.cpp.o
In file included from /home/ruiqi/Kimera-VIO/include/kimera-vio/frontend/CameraParams.h:27,
                 from /home/ruiqi/Kimera-VIO/include/kimera-vio/frontend/Camera.h:25,
                 from /home/ruiqi/Kimera-VIO/include/kimera-vio/dataprovider/DataProviderInterface.h:20,
                 from /home/ruiqi/Kimera-VIO/include/kimera-vio/dataprovider/KittiDataProvider.h:21,
                 from /home/ruiqi/Kimera-VIO/src/dataprovider/KittiDataProvider.cpp:15:
/home/ruiqi/Kimera-VIO/include/kimera-vio/utils/UtilsOpenCV.h:39:7: error: using typedef-name ‘gtsam::Point2’ after ‘class’
 class Point2;
       ^~~~~~
In file included from /usr/local/include/gtsam/geometry/Cal3_S2.h:24,
                 from /home/ruiqi/Kimera-VIO/include/kimera-vio/frontend/Camera.h:21,
                 from /home/ruiqi/Kimera-VIO/include/kimera-vio/dataprovider/DataProviderInterface.h:20,
                 from /home/ruiqi/Kimera-VIO/include/kimera-vio/dataprovider/KittiDataProvider.h:21,
                 from /home/ruiqi/Kimera-VIO/src/dataprovider/KittiDataProvider.cpp:15:
/usr/local/include/gtsam/geometry/Point2.h:27:17: note: ‘gtsam::Point2’ has a previous declaration here
 typedef Vector2 Point2;
                 ^~~~~~
In file included from /home/ruiqi/Kimera-VIO/include/kimera-vio/frontend/CameraParams.h:27,
                 from /home/ruiqi/Kimera-VIO/include/kimera-vio/frontend/Camera.h:25,
                 from /home/ruiqi/Kimera-VIO/include/kimera-vio/dataprovider/DataProviderInterface.h:20,
                 from /home/ruiqi/Kimera-VIO/include/kimera-vio/dataprovider/KittiDataProvider.h:21,
                 from /home/ruiqi/Kimera-VIO/src/dataprovider/KittiDataProvider.cpp:15:
/home/ruiqi/Kimera-VIO/include/kimera-vio/utils/UtilsOpenCV.h:40:7: error: using typedef-name ‘gtsam::Point3’ after ‘class’
 class Point3;
       ^~~~~~
In file included from /usr/local/include/gtsam/geometry/Pose3.h:23,
                 from /usr/local/include/gtsam/geometry/CalibratedCamera.h:23,
                 from /usr/local/include/gtsam/geometry/PinholePose.h:22,
                 from /usr/local/include/gtsam/geometry/PinholeCamera.h:21,
                 from /home/ruiqi/Kimera-VIO/include/kimera-vio/frontend/Camera.h:22,
                 from /home/ruiqi/Kimera-VIO/include/kimera-vio/dataprovider/DataProviderInterface.h:20,
                 from /home/ruiqi/Kimera-VIO/include/kimera-vio/dataprovider/KittiDataProvider.h:21,
                 from /home/ruiqi/Kimera-VIO/src/dataprovider/KittiDataProvider.cpp:15:
/usr/local/include/gtsam/geometry/Point3.h:35:17: note: ‘gtsam::Point3’ has a previous declaration here
 typedef Vector3 Point3;
                 ^~~~~~
In file included from /home/ruiqi/Kimera-VIO/include/kimera-vio/frontend/CameraParams.h:27,
                 from /home/ruiqi/Kimera-VIO/include/kimera-vio/frontend/Camera.h:25,
                 from /home/ruiqi/Kimera-VIO/include/kimera-vio/dataprovider/DataProviderInterface.h:20,
                 from /home/ruiqi/Kimera-VIO/src/dataprovider/DataProviderInterface.cpp:15:
/home/ruiqi/Kimera-VIO/include/kimera-vio/utils/UtilsOpenCV.h:39:7: error: using typedef-name ‘gtsam::Point2’ after ‘class’
 class Point2;
       ^~~~~~
In file included from /usr/local/include/gtsam/geometry/Cal3_S2.h:24,
                 from /home/ruiqi/Kimera-VIO/include/kimera-vio/frontend/Camera.h:21,
                 from /home/ruiqi/Kimera-VIO/include/kimera-vio/dataprovider/DataProviderInterface.h:20,
                 from /home/ruiqi/Kimera-VIO/src/dataprovider/DataProviderInterface.cpp:15:
/usr/local/include/gtsam/geometry/Point2.h:27:17: note: ‘gtsam::Point2’ has a previous declaration here
 typedef Vector2 Point2;
                 ^~~~~~
In file included from /home/ruiqi/Kimera-VIO/include/kimera-vio/frontend/CameraParams.h:27,
                 from /home/ruiqi/Kimera-VIO/include/kimera-vio/frontend/Camera.h:25,
                 from /home/ruiqi/Kimera-VIO/include/kimera-vio/dataprovider/DataProviderInterface.h:20,
                 from /home/ruiqi/Kimera-VIO/src/dataprovider/DataProviderInterface.cpp:15:
/home/ruiqi/Kimera-VIO/include/kimera-vio/utils/UtilsOpenCV.h:40:7: error: using typedef-name ‘gtsam::Point3’ after ‘class’
 class Point3;
       ^~~~~~
In file included from /usr/local/include/gtsam/geometry/Pose3.h:23,
                 from /usr/local/include/gtsam/geometry/CalibratedCamera.h:23,
[... cut ...]
make[2]: *** [CMakeFiles/kimera_vio.dir/src/dataprovider/EurocDataProvider.cpp.o] Error 1
CMakeFiles/kimera_vio.dir/build.make:110: recipe for target 'CMakeFiles/kimera_vio.dir/src/dataprovider/DataProviderModule.cpp.o' failed
make[2]: *** [CMakeFiles/kimera_vio.dir/src/dataprovider/DataProviderModule.cpp.o] Error 1
CMakeFiles/Makefile2:291: recipe for target 'CMakeFiles/kimera_vio.dir/all' failed
make[1]: *** [CMakeFiles/kimera_vio.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2

ToniRV, 2020-10-25:
Ok! Thanks for the detailed answer! The key command missing here is that gtsam needs to be in the master branch while the develop branch is used by default when you git clone it...
Therefore:
1. go to gtsam's source code, and run: `git checkout master`
2. re-build and re-install: cd build, cmake .., make -j 8, sudo make install
3. go to kimera's source code, and re-build.

Checkout this for details on how our server builds gtsam:
https://github.com/MIT-SPARK/Kimera-VIO/blob/1a524efcd2e82ca117e243490d2d6a5f2cf60b35/Dockerfile_18_04#L46

ToniRV, 2020-10-30:
Hi @yrq0208, maybe try to re-install KimeraRPGO?
Otherwise, feel free to open a new issue in Kimera-RPGO repository and link it to this one.

yrq0208, 2020-10-31:
Hi @ToniRV,

Thanks for the reply.

I am too busy yesterday that I didn't noticed that it is actually a KimeraRPGO related issue! I rebuild KimeraRPGO and now I can run KimeraVIO.

Thanks again for your help over the last few weeks!

closed 2020-10-31 by yrq0208