MIT-SPARK/Kimera-VIO #116: How are you guys combining Kimera-RPGO and Kimera-VIO together

ujasmandavia, 2020-10-08:
Hello,

How are you guys using Kimera RPGO here or I mean to say how are you guys combining these two modules that is Kimera-VIO and Kimera-RPGO?

Is the backend folder in Kimera-VIO only doing visual-inertial bundle adjustment or is it also performing PGO? And how are you guys importing robust kernels and outlier rejection functionality from Kimera-RPGO in the backend operations in Kimera-vio?

ToniRV, 2020-10-12:
Hi @ujasmandavia,

Kimera-RPGO is a module that is used by Kimera-VIO.

In particular, we have in Kimera-VIO a so-called LoopClosureDetector which contains an instance of Kimera-RPGO (see https://github.com/MIT-SPARK/Kimera-VIO/blob/8012623e0780a9d59ab2e27bbed68b163a94d1b4/include/kimera-vio/loopclosure/LoopClosureDetector.h#L372)

Kimera-VIO acts simply as a frontend to detect loop closures and add odometry measurements, while Kimera-RPGO is used as a black box backend.

If you'd like to know how Kimera-VIO interacts with Kimera-RPGO, the relevant source code is here: https://github.com/MIT-SPARK/Kimera-VIO/blob/ec6a327650a0e3901903cd1ac5fd9dd3b9e9e87c/src/loopclosure/LoopClosureDetector.cpp

Concerning the robust kernels/outlier rejection functionality, all of it is done in Kimera-RPGO itself.
If you'd like to know more about it feel free to ask any questions in the Kimera-RPGO issue tracker.

Hope this helps!

ujasmandavia, 2020-10-12:
Hey @ToniRV 

I actually found the part of the code I was looking for and thank you very much for the explanation.

closed 2020-10-12 by ujasmandavia