MIT-SPARK/Kimera-VIO #118: Compile error on windows with visual studio

nawara72, 2020-10-14:
Hi
I am trying to compile the code on windows using visual studio 2019 and getting the following error, would appreciate any suggestions, thanks

1>C:\Program Files\Kimera-VIO-master\include\kimera-vio/frontend/feature-detector/anms/range-tree/ranget.h(172,28): error C2131: expression did not evaluate to a constant
1>C:\Program Files\Kimera-VIO-master\include\kimera-vio/frontend/feature-detector/anms/range-tree/ranget.h(172,28): message : failure was caused by a read of a variable outside its lifetime
1>C:\Program Files\Kimera-VIO-master\include\kimera-vio/frontend/feature-detector/anms/range-tree/ranget.h(172,28): message : see usage of 'this'
1>C:\Program Files\Kimera-VIO-master\include\kimera-vio/frontend/feature-detector/anms/range-tree/ranget.h(158): message : while compiling class template member function 'std::vector<data *,std::allocator<data *>> *rangetree<u16,data>::search(point,point,point,point) const'
1>        with
1>        [
1>            data=u16,
1>            point=u16
1>        ]
1>C:\Program Files\Kimera-VIO-master\src\frontend\feature-detector\anms\anms.cpp(318): message : see reference to function template instantiation 'std::vector<data *,std::allocator<data *>> *rangetree<u16,data>::search(point,point,point,point) const' being compiled
1>        with
1>        [
1>            data=u16,
1>            point=u16
1>        ]
1>C:\Program Files\Kimera-VIO-master\src\frontend\feature-detector\anms\anms.cpp(280): message : see reference to class template instantiation 'rangetree<u16,u16>' being compiled
1>C:\Program Files\Kimera-VIO-master\include\kimera-vio/frontend/feature-detector/anms/range-tree/ranget.h(185,1): warning C4365: 'argument': conversion from 'int' to 'const point', signed/unsigned mismatch
1>        with
1>        [
1>            point=u16
1>        ]
1>C:\Program Files\Kimera-VIO-master\include\kimera-vio/frontend/feature-detector/anms/range-tree/ranget.h(298,1): warning C4820: 'rangetree<u16,u16>::pty': '6' bytes padding added after data member 'rangetree<u16,u16>::pty::y'
1>C:\Program Files\Kimera-VIO-master\include\kimera-vio/frontend/feature-detector/anms/range-tree/ranget.h(191): message : see reference to class template instantiation 'rangetree<u16,u16>::pty' being compiled
1>Done building project "kimera_vio.vcxproj" -- FAILED.

ToniRV, 2020-10-14:
@nawara72 I'm not sure what this can be, but worst case, since the current implementation accepts multiple ANMS (non-max suppression) algorithms, and all erros seem to come from ranget.h (or rather RangeTree algorithm), a way would be to remove all references to that and use instead any other ANMS algorithm (such as SSC or TopN) 
See https://github.com/MIT-SPARK/Kimera-VIO/blob/d677a3dedad2da605c84a4b7f51154cd7e38a98d/include/kimera-vio/frontend/feature-detector/NonMaximumSuppression.h#L48

nawara72, 2020-10-14:
Thanks Toni for the suggestion, I followed your advice and the build seems to go through but then got the following linker error (i am linking to Gtsam 4.03), would appreciate any advice. I can't seem to figure out which line of code in RegularVioBackEnd.cpp is calling the operator<< on gtsam::Unit3

1>RegularVioBackEnd.obj : error LNK2019: unresolved external symbol "class std::basic_ostream<char,struct std::char_traits<char> > & __cdecl gtsam::operator<<(class std::basic_ostream<char,struct std::char_traits<char> > &,class gtsam::Unit3 const &)" (??6gtsam@@YAAEAV?$basic_ostream@DU?$char_traits@D@std@@@std@@AEAV12@AEBVUnit3@0@@Z) referenced in function "private: void __cdecl VIO::RegularVioBackEnd::addRegularityFactors(struct VIO::Plane const &,class std::map<long,enum VIO::RegularVioBackEnd::RegularityType,struct std::less<long>,class std::allocator<struct std::pair<long const ,enum VIO::RegularVioBackEnd::RegularityType> > > *,class std::vector<struct std::pair<unsigned __int64,long>,class std::allocator<struct std::pair<unsigned __int64,long> > > *)" (?addRegularityFactors@RegularVioBackEnd@VIO@@AEAAXAEBUPlane@2@PEAV?$map@JW4RegularityType@RegularVioBackEnd@VIO@@U?$less@J@std@@V?$allocator@U?$pair@$$CBJW4RegularityType@RegularVioBackEnd@VIO@@@std@@@5@@std@@PEAV?$vector@U?$pair@_KJ@std@@V?$allocator@U?$pair@_KJ@std@@@2@@5@@Z)

nawara72, 2020-10-18:
Hi Toni

Thanks for your advice, that worked and was able to compile kimera-vio on windows

However when I tried to build kimera-vio-ros on windows (after struggling and sorting out  various errors) I got a linker error. Can't find opencv_gapi.lib (as shown below) Which opencv version should I be using? As i understand opencv3 doesn't have gapi and opencv 4 has gapi built with opencv-world. Would appreciate any help you can give. 

Thanks again
Nawar

[ 73%] Linking CXX shared library C:\ros_workspace\Kimera-VIO-ROS\devel\bin\kimera_vio_ros.dll
LINK: command "C:\PROGRA~2\MICROS~1\2019\COMMUN~1\VC\Tools\MSVC\1427~1.291\bin\Hostx64\x64\link.exe /nologo @CMakeFiles\kimera_vio_ros.dir\objects1.rsp /out:C:\ros_workspace\Kimera-VIO-ROS\devel\bin\kimera_vio_ros.dll /implib:C:\ros_workspace\Kimera-VIO-ROS\devel\lib\kimera_vio_ros.lib /pdb:C:\ros_workspace\Kimera-VIO-ROS\devel\bin\kimera_vio_ros.pdb /dll /version:0.0 /machine:x64 /INCREMENTAL:NO /DEF:CMakeFiles\kimera_vio_ros.dir\exports.def C:\opt\ros\melodic\x64\lib\image_transport.lib C:\opt\ros\melodic\x64\lib\cv_bridge.lib C:\opt\rosdeps\x64\lib\opencv_core341.lib C:\opt\rosdeps\x64\lib\opencv_imgproc341.lib C:\opt\rosdeps\x64\lib\opencv_imgcodecs341.lib C:\ros_workspace\Kimera-VIO-ROS\devel\lib\pose_graph_tools.lib C:\opt\ros\melodic\x64\lib\interactive_markers.lib C:\opt\ros\melodic\x64\lib\pcl_ros_filter.lib C:\opt\ros\melodic\x64\lib\pcl_ros_tf.lib C:\opt\rosdeps\x64\lib\pcl_kdtree_release.lib C:\opt\rosdeps\x64\lib\pcl_search_release.lib C:\opt\rosdeps\x64\lib\pcl_features_release.lib C:\opt\rosdeps\x64\lib\pcl_sample_consensus_release.lib C:\opt\rosdeps\x64\lib\pcl_filters_release.lib C:\opt\rosdeps\x64\lib\pcl_ml_release.lib C:\opt\rosdeps\x64\lib\pcl_segmentation_release.lib C:\opt\rosdeps\x64\lib\pcl_surface_release.lib C:\opt\rosdeps\x64\lib\flann_cpp_s.lib C:\opt\ros\melodic\x64\lib\dynamic_reconfigure_config_init_mutex.lib C:\opt\ros\melodic\x64\lib\nodeletlib.lib C:\opt\ros\melodic\x64\lib\bondcpp.lib C:\Program Files (x86)\Windows Kits\10\Lib\10.0.18362.0\um\x64\RpcRT4.Lib C:\opt\ros\melodic\x64\lib\rosbag.lib C:\opt\ros\melodic\x64\lib\rosbag_storage.lib C:\opt\ros\melodic\x64\lib\class_loader.lib C:\opt\rosdeps\x64\lib\PocoFoundation.lib C:\opt\ros\melodic\x64\lib\roslib.lib C:\opt\ros\melodic\x64\lib\rospack.lib C:\opt\python27amd64\libs\python27.lib C:\opt\rosdeps\x64\lib\boost_program_options-vc141-mt-x64-1_66.lib C:\opt\rosdeps\x64\lib\tinyxml2.lib C:\opt\ros\melodic\x64\lib\roslz4.lib C:\opt\rosdeps\x64\lib\lz4.lib C:\opt\ros\melodic\x64\lib\topic_tools.lib C:\opt\ros\melodic\x64\lib\tf.lib C:\opt\ros\melodic\x64\lib\tf2_ros.lib C:\opt\ros\melodic\x64\lib\actionlib.lib C:\opt\ros\melodic\x64\lib\message_filters.lib C:\opt\ros\melodic\x64\lib\tf2.lib C:\opt\rosdeps\x64\lib\pcl_common_release.lib C:\opt\rosdeps\x64\lib\pcl_octree_release.lib C:\opt\rosdeps\x64\lib\pcl_io_release.lib C:\opt\rosdeps\x64\lib\boost_iostreams-vc141-mt-x64-1_66.lib C:\opt\rosdeps\x64\lib\boost_serialization-vc141-mt-x64-1_66.lib C:\opt\ros\melodic\x64\lib\roscpp.lib C:\opt\rosdeps\x64\lib\boost_filesystem-vc141-mt-x64-1_66.lib C:\opt\ros\melodic\x64\lib\rosconsole.lib C:\opt\ros\melodic\x64\lib\rosconsole_log4cxx.lib C:\opt\ros\melodic\x64\lib\rosconsole_backend_interface.lib C:\opt\rosdeps\x64\lib\log4cxx.lib C:\opt\rosdeps\x64\lib\boost_regex-vc141-mt-x64-1_66.lib C:\opt\ros\melodic\x64\lib\xmlrpcpp.lib C:\opt\ros\melodic\x64\lib\roscpp_serialization.lib C:\opt\ros\melodic\x64\lib\rostime.lib C:\opt\ros\melodic\x64\lib\cpp_common.lib C:\opt\rosdeps\x64\lib\boost_system-vc141-mt-x64-1_66.lib C:\opt\rosdeps\x64\lib\boost_thread-vc141-mt-x64-1_66.lib C:\opt\rosdeps\x64\lib\boost_chrono-vc141-mt-x64-1_66.lib C:\opt\rosdeps\x64\lib\boost_date_time-vc141-mt-x64-1_66.lib C:\opt\rosdeps\x64\lib\boost_atomic-vc141-mt-x64-1_66.lib C:\opt\rosdeps\x64\lib\console_bridge.lib C:\Program Files\kimera_vio\lib\kimera_vio.lib C:\opt\rosdeps\x64\lib\opencv_dnn341.lib opencv_gapi.lib C:\opt\rosdeps\x64\lib\opencv_ml341.lib C:\opt\rosdeps\x64\lib\opencv_objdetect341.lib C:\opt\rosdeps\x64\lib\opencv_stitching341.lib opencv_world.lib opencv_alphamat.lib opencv_aruco.lib opencv_bgsegm.lib opencv_bioinspired.lib opencv_ccalib.lib opencv_datasets.lib opencv_dnn_objdetect.lib opencv_dnn_superres.lib opencv_dpm.lib opencv_face.lib opencv_fuzzy.lib opencv_hfs.lib opencv_img_hash.lib opencv_intensity_transform.lib opencv_line_descriptor.lib opencv_mcc.lib opencv_optflow.lib opencv_phase_unwrapping.lib opencv_plot.lib opencv_quality.lib opencv_rapid.lib opencv_reg.lib opencv_rgbd.lib opencv_saliency.lib opencv_sfm.lib C:\opt\rosdeps\x64\lib\opencv_shape341.lib opencv_stereo.lib opencv_structured_light.lib C:\opt\rosdeps\x64\lib\opencv_superres341.lib opencv_surface_matching.lib opencv_text.lib opencv_tracking.lib C:\opt\rosdeps\x64\lib\opencv_videostab341.lib C:\opt\rosdeps\x64\lib\opencv_calib3d341.lib C:\opt\rosdeps\x64\lib\opencv_features2d341.lib C:\opt\rosdeps\x64\lib\opencv_flann341.lib C:\opt\rosdeps\x64\lib\opencv_highgui341.lib C:\opt\rosdeps\x64\lib\opencv_photo341.lib C:\opt\rosdeps\x64\lib\opencv_video341.lib C:\opt\rosdeps\x64\lib\opencv_videoio341.lib C:\opt\rosdeps\x64\lib\opencv_imgcodecs341.lib C:\opt\rosdeps\x64\lib\opencv_imgproc341.lib C:\opt\rosdeps\x64\lib\opencv_core341.lib opencv_viz.lib opencv_xfeatures2d.lib opencv_ximgproc.lib opencv_xobjdetect.lib opencv_xphoto.lib C:\Program Files\gflags\lib\gflags_static.lib C:\Program Files\glog-master\bld\win10_vs19\R [... cut ...] -1_66.lib C:\opt\rosdeps\x64\lib\boost_chrono-vc141-mt-x64-1_66.lib C:\opt\rosdeps\x64\lib\boost_system-vc141-mt-x64-1_66.lib C:\Program Files\GTSAM\lib\metis-gtsam.lib kernel32.lib user32.lib gdi32.lib winspool.lib shell32.lib ole32.lib oleaut32.lib uuid.lib comdlg32.lib advapi32.lib /MANIFEST /MANIFESTFILE:C:\ros_workspace\Kimera-VIO-ROS\devel\bin\kimera_vio_ros.dll.manifest" failed (exit code 1181) with the following output:
LINK : fatal error LNK1181: cannot open input file 'opencv_gapi.lib'

ToniRV, 2020-10-24:
Hi @nawara72,

I wish I could help with this, but I haven't really tried to build kimera-vio on Windows (only MacOS, Ubuntu16.04 and Ubuntu18.04 are well tested).
Nevertheless, I see you are trying to install kimera-vio-ros on windows? I have never tried to install ROS on Windows, but I can imagine that it could bring all sorts of problems.

My advice here would be to first try to install Kimera-VIO without ROS on Windows, and see if that works (which I believe @marcusabate managed to do at some point!), and then perhaps proceed with ROS.

BAILOOL, 2020-12-03:
@nawara72 @ToniRV sorry for causing so much trouble with the ANMS codes. The [repo](https://github.com/BAILOOL/ANMS-Codes) doesn't have the best production-ready quality code, especially c++ version. In practice, the tree-based ANMS algorithms are only recommended when you re-use the rebuilt tree later. Otherwise, SSC is the way to go in terms of speed, memory, and SLAM influence and does not have any non-STL dependencies rather than OpenCV.

closed 2023-11-03 by marcusabate