MIT-SPARK/Kimera-VIO #123: Gflags problems

mubaarik, 2020-10-21:
**Description:**
The compiler is not detecting gflags for some reason. When I include gflags explicitly, there's another issue. Both errors are added below. 

**Command:**
```
mubo@DESKTOP-IN87LBK:~/kimera/Kimera-VIO/build$ make
```

**Console output:**
```
mubo@DESKTOP-IN87LBK:~/kimera/Kimera-VIO/build$ make
/home/mubo/kimera/Kimera-VIO/src/mesh/MesherFactory.cpp:18:12: error: expected constructor, destructor, or type conversion before ‘(’ token
 DEFINE_bool(serialize_mesh_2d3d, false, "Serialize 2D/3D Mesh to file.");
            ^
/home/mubo/kimera/Kimera-VIO/src/mesh/MesherFactory.cpp: In static member function ‘static VIO::Mesher::UniquePtr VIO::MesherFactory::createMesher(const VIO::MesherType&, const VIO::MesherParams&)’:
/home/mubo/kimera/Kimera-VIO/src/mesh/MesherFactory.cpp:27:54: error: ‘FLAGS_serialize_mesh_2d3d’ was not declared in this scope
       return VIO::make_unique<Mesher>(mesher_params, FLAGS_serialize_mesh_2d3d);
                                                      ^~~~~~~~~~~~~~~~~~~~~~~~~
CMakeFiles/kimera_vio.dir/build.make:830: recipe for target 'CMakeFiles/kimera_vio.dir/src/mesh/MesherFactory.cpp.o' failed
make[2]: *** [CMakeFiles/kimera_vio.dir/src/mesh/MesherFactory.cpp.o] Error 1
CMakeFiles/Makefile2:163: recipe for target 'CMakeFiles/kimera_vio.dir/all' failed
make[1]: *** [CMakeFiles/kimera_vio.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2

mubo@DESKTOP-IN87LBK:~/kimera/Kimera-VIO/build$ make
[  1%] Linking CXX shared library libkimera_vio.so
/usr/bin/ld: /usr/local/lib/libgflags.a(gflags.cc.o): relocation R_X86_64_PC32 against symbol `stderr@@GLIBC_2.2.5' can not be used when making a shared object; recompile with -fPIC
/usr/bin/ld: final link failed: Bad value
collect2: error: ld returned 1 exit status
CMakeFiles/kimera_vio.dir/build.make:1764: recipe for target 'libkimera_vio.so' failed
make[2]: *** [libkimera_vio.so] Error 1
CMakeFiles/Makefile2:163: recipe for target 'CMakeFiles/kimera_vio.dir/all' failed
make[1]: *** [CMakeFiles/kimera_vio.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2

```

**Additional files:**
Please attach all the files needed to reproduce the error.

Please give also the following information:
* Kimera-Vio branch, tag or commit used
* GTSAM version used:
* OpenGV version used:
* OpenCV version used: type `opencv_version`
* Operating system and version (e.g. Ubuntu 16.04 or Windows 10): 
* Did you change the source code? (yes / no):

ToniRV, 2020-10-24:
How did you install gflags/glog @mubaarik ?
Looks like the version you have is not compiled with -fPIC (Position Independent Code), make sure you only have one version in your system and that it is "sudo apt" installed.
Maybe you also find useful this thread: https://github.com/ceres-solver/ceres-solver/issues/219

ToniRV, 2021-01-29:
Feel free to re-open @mubaarik

closed 2021-01-29 by ToniRV

X-NEARS, 2022-04-18:
I also encountered the same problem.
I solved this problem by reinstalling glog and gflags to the default location. 
There is no need to uninstall other versions. Just overwrite and install the new version.