MIT-SPARK/Kimera-VIO #125: Get same results as jenkins server

rocomanmelli, 2020-10-28:
Hi!
I'm trying to get the same results that are reported by jenkins server but I can't. So far I have run Kimera-VIO in MH_01_easy with these results (got them with evo_ape):
```
       max	0.514870
      mean	0.147401
    median	0.150005
       min	0.036654
```
while the reported ones are:
```
       max	0.1441272
      mean	0.07644303
    median	0.08055103
       min	0.009593532
```
I have to mention that I'm using Kimera-VIO-ROS (in a docker container) but it is configured to run offline. I have added the ground truth data to the rosbag (only poses) so autoinitialize is set to 0. All the settings and params are just as they are in the repositories. What could I be missing? I took a look at the Jenkinsfile but the Python script that is used to evaluate the performance on the EuRoC is not present.
Thank you in advance!

rocomanmelli, 2020-10-28:
I added the first linear twist to the ground truth topic in the rosbag used offline (found that it is used in the code) but did not improve the results as expected:
```
       max	0.447725
      mean	0.158973
    median	0.160194
       min	0.035205
```

rocomanmelli, 2020-10-28:
Sorry, my bad. Finally found everything [here](https://github.com/ToniRV/Kimera-VIO-Evaluation/).

rocomanmelli, 2020-10-28:
Still can't get the same results. I'm not being able to make Kimera-VIO-ROS take into account initial_k and final_k arguments from the roslaunch.

rocomanmelli, 2020-10-28:
initial_k and final_k are not being considered because the data provider class is not the `EurocDataProvider` but the `RosbagDataProvider`. So it seems that it is not possible to get same results as with jenkins server with the Kimera-VIO-ROS, right?

ToniRV, 2020-11-30:
Hi @rocomanmell, you can always start the rosbag later using `-s` flag: for example `rosbag play dataset.bag --clock -s 20`. This way you can start the dataset at the same time that in the Euroc evaluation.

closed 2020-11-30 by ToniRV