MIT-SPARK/Kimera-VIO #127: problem with bash script from example

fatrybl, 2020-11-05:
**Description:**
Hello!
I have not launched Kimera since July and seems like it changed a bit. I updated Gtsam and all other packages including Kimera itself. Im also using OpenCV 4 now.
But I can not even launch the example script now. In Summer it worked.
Any ideas, Toni ?

**Command:**
bash ./scripts/stereoVIOEuroc.bash -p /home/mark/Desktop/kimera/datasets/EUROC/V1_01_easy/

# replace this line with the command(s) you used
```

**Console output:**
```
Yamelizing dataset: /home/mark/Desktop/kimera/datasets/EUROC/V1_01_easy/mav0
Yamelizing file: /home/mark/Desktop/kimera/datasets/EUROC/V1_01_easy/mav0/body.yaml
Yamelizing file: /home/mark/Desktop/kimera/datasets/EUROC/V1_01_easy/mav0/cam1/sensor.yaml
Yamelizing file: /home/mark/Desktop/kimera/datasets/EUROC/V1_01_easy/mav0/state_groundtruth_estimate0/sensor.yaml
Yamelizing file: /home/mark/Desktop/kimera/datasets/EUROC/V1_01_easy/mav0/pointcloud0/sensor.yaml
Yamelizing file: /home/mark/Desktop/kimera/datasets/EUROC/V1_01_easy/mav0/vicon0/sensor.yaml
Yamelizing file: /home/mark/Desktop/kimera/datasets/EUROC/V1_01_easy/mav0/imu0/sensor.yaml
Yamelizing file: /home/mark/Desktop/kimera/datasets/EUROC/V1_01_easy/mav0/cam0/sensor.yaml
Yamelizing dataset: /home/mark/Desktop/kimera/datasets/EUROC/V1_01_easy/V1_01_easy.bag
(base) mark@mark:~/Desktop/kimera/Kimera-VIO$ bash ./scripts/stereoVIOEuroc.bash -p /home/mark/Desktop/kimera/datasets/EUROC/V1_01_easy/
Using dataset at path: /home/mark/Desktop/kimera/datasets/EUROC/V1_01_easy/
 Launching:

            ██╗  ██╗██╗███╗   ███╗███████╗██████╗  █████╗
            ██║ ██╔╝██║████╗ ████║██╔════╝██╔══██╗██╔══██╗
            █████╔╝ ██║██╔████╔██║█████╗  ██████╔╝███████║
            ██╔═██╗ ██║██║╚██╔╝██║██╔══╝  ██╔══██╗██╔══██║
            ██║  ██╗██║██║ ╚═╝ ██║███████╗██║  ██║██║  ██║
            ╚═╝  ╚═╝╚═╝╚═╝     ╚═╝╚══════╝╚═╝  ╚═╝╚═╝  ╚═╝

 
../build/stereoVIOEuroc: symbol lookup error: ../build/stereoVIOEuroc: undefined symbol: _ZN3VIO8Pipeline20shutdownWhenFinishedERKiRKb

# remove this line and paste your console output HERE - no screenshots please

```

**Additional files:**
Please attach all the files needed to reproduce the error.

Please give also the following information:
* Kimera-Vio branch, tag or commit used
* GTSAM version used:
* OpenGV version used:
* OpenCV version used: type `opencv_version`
* Operating system and version (e.g. Ubuntu 16.04 or Windows 10): 
* Did you change the source code? (yes / no):

RachidaSK, 2021-01-29:
I am also seeing the same error:
Using dataset at path: .../Euroc/V1_01_easy
 Launching:

            ██╗  ██╗██╗███╗   ███╗███████╗██████╗  █████╗
            ██║ ██╔╝██║████╗ ████║██╔════╝██╔══██╗██╔══██╗
            █████╔╝ ██║██╔████╔██║█████╗  ██████╔╝███████║
            ██╔═██╗ ██║██║╚██╔╝██║██╔══╝  ██╔══██╗██╔══██║
            ██║  ██╗██║██║ ╚═╝ ██║███████╗██║  ██║██║  ██║
            ╚═╝  ╚═╝╚═╝╚═╝     ╚═╝╚══════╝╚═╝  ╚═╝╚═╝  ╚═╝

 
../build/stereoVIOEuroc: symbol lookup error: .../Kimera-VIO/build/libkimera_vio.so: undefined symbol: _ZNK5gtsam13HessianFactor15hessianDiagonalEv

* Kimera-Vio branch, tag or commit used: commit f78e55978a46a7056d733e36489b8375a0807b94
* GTSAM version used: commit ca44312bc92201ad40ef7c203908a77d71a24934
* OpenGV version used: commit 91f4b19c73450833a40e463ad3648aae80b3a7f3
* OpenCV version used: 3.3.1
* Operating system and version (e.g. Ubuntu 16.04 or Windows 10):  Ubuntu 16.04
* Did you change the source code? (yes / no): no

@fatrybl any idea on how to solve this?

fatrybl, 2021-01-29:
> 
> 
> I am also seeing the same error:
> Using dataset at path: .../Euroc/V1_01_easy
> Launching:
> 
> ```
>         ██╗  ██╗██╗███╗   ███╗███████╗██████╗  █████╗
>         ██║ ██╔╝██║████╗ ████║██╔════╝██╔══██╗██╔══██╗
>         █████╔╝ ██║██╔████╔██║█████╗  ██████╔╝███████║
>         ██╔═██╗ ██║██║╚██╔╝██║██╔══╝  ██╔══██╗██╔══██║
>         ██║  ██╗██║██║ ╚═╝ ██║███████╗██║  ██║██║  ██║
>         ╚═╝  ╚═╝╚═╝╚═╝     ╚═╝╚══════╝╚═╝  ╚═╝╚═╝  ╚═╝
> ```
> 
> ../build/stereoVIOEuroc: symbol lookup error: .../Kimera-VIO/build/libkimera_vio.so: undefined symbol: _ZNK5gtsam13HessianFactor15hessianDiagonalEv
> 
>     * Kimera-Vio branch, tag or commit used: commit [f78e559](https://github.com/MIT-SPARK/Kimera-VIO/commit/f78e55978a46a7056d733e36489b8375a0807b94)
> 
>     * GTSAM version used: commit ca44312bc92201ad40ef7c203908a77d71a24934
> 
>     * OpenGV version used: commit 91f4b19c73450833a40e463ad3648aae80b3a7f3
> 
>     * OpenCV version used: 3.3.1
> 
>     * Operating system and version (e.g. Ubuntu 16.04 or Windows 10):  Ubuntu 16.04
> 
>     * Did you change the source code? (yes / no): no
> 
> 
> @fatrybl any idea on how to solve this?

Frankly speaking, i`ve not tried to solve it. Hope, Toni or some other moderators could help in it

RachidaSK, 2021-01-29:
Thank you so much @fatrybl. @ToniRV any idea?

ToniRV, 2021-01-29:
Hi everyone! Yes, I have also seen this! It is as easy as using `develop` branch in GTSAM.
We had to go yet again back to that branch because of some features we need...
SORRY for that back and forth :pray:

RachidaSK, 2021-01-30:
Thank you so much @ToniRV, that solved the issue.

closed 2021-01-30 by ToniRV