MIT-SPARK/Kimera-VIO #129: gtsam steps to install are wrong

JessMotta, 2020-11-09:
**Description:**
When I try to install the gtsam following these steps I found a problem  in c++ compiler and some parts was using almost all my memory.

**Command:**
```
make -j $(nproc) check # (optional, runs unit tests)
sudo make -j $(nproc) install

```
**Console output:**
```
![ram_issue](https://user-images.githubusercontent.com/30941796/98555096-455fda00-2280-11eb-883d-37c851fee5fc.png)
![problem](https://user-images.githubusercontent.com/30941796/98555098-46910700-2280-11eb-830a-0d4945bccdd2.png)

```

**Additional files:**
Please attach all the files needed to reproduce the error.

Please give also the following information:
* Kimera-Vio branch, tag or commit used: same as informed in readme
* GTSAM version used:same as informed in readme
* OpenGV version used:same as informed in readme
* OpenCV version used: type `opencv_version`same as informed in readme
* Operating system and version (e.g. Ubuntu 16.04 or Windows 10): Ubuntu 18.04
* Did you change the source code? (yes / no): no

I followed the steps in https://github.com/borglab/gtsam and I managed to install

ToniRV, 2020-11-09:
Hi @JessMotta,

Can you re-upload the screenshots?

If memory is a problem, then a simple fix might be to replace  `$(nproc)` for a number smaller than the total number of CPU cores, something like this should do:

```
sudo make -j 4 install
```

JessMotta, 2020-11-09:
Hi, I tried and worked but I couldn't install Kimera-VIO appeared some errors

ToniRV, 2021-01-29:
@JessMotta feel free to open another issue :+1:

closed 2021-01-29 by ToniRV

JessMotta, 2021-02-07:
> @JessMotta feel free to open another issue 👍

Thank you :D