MIT-SPARK/Kimera-VIO #138: Understanding the code

p-akanksha, 2020-12-03:
Hi, 

I was trying to understand the code and I am a little confused about one thing. What does Px-Measurement in the following line stand for? Is it the 3d world position of the feature/landmark?

https://github.com/MIT-SPARK/Kimera-VIO/blob/2483555f6c3f3dc2c94cee2f21be4001a484287f/include/kimera-vio/backend/VioBackEnd-definitions.h#L86

Thanks,
Akanksha Patel

ToniRV, 2020-12-03:
Hi @p-akanksha ,
Px-Measurement (aka StereoPoint2) stands for a "pixel" measurement.
Actually, StereoPoint2 holds a stereo measurement, meaning a left frame keypoint (pixel) and a corresponding right frame keypoint (pixel).

Hope this helps.

p-akanksha, 2020-12-03:
Thanks, Toni. A follow-up question, what does the third variable store (v)?

ToniRV, 2020-12-03:
In a stereo setup, there are three relevant variables:
- the `u, v` (cols, rows) pixel coordinates of the left-frame keypoint, and
- the `u'` coordinate of the corresponding right-frame keypoint (since it has the same `v` coordinate as the left-frame keypoint).

The left and right frames are assumed to be undistorted and **rectified** thereby making epipolar lines horizontal, or in other words, each `u, v` keypoint in the left frame has a corresponding `u', v` keypoint, with `u` different from `u'` and the `v` coordinate being the same.

p-akanksha, 2020-12-03:
That makes sense. Thanks for the clarification.

closed 2020-12-03 by p-akanksha