MIT-SPARK/Kimera-VIO #140: I need some help with calibration or using already existing calibration data

SparkyTD, 2020-12-10:
Hello. I have a device with a set of IR cameras in stereo configuration and a built-in IMU. Unfortunately this is a completely proprietary device, so there is no documentation for it whatsoever.

I can access the camera feeds form OpenCV, and the IMU data too, but I don't know how to get the calibration data required to integrate Kimera-VIO with this device. I was trying to use kalibir in the past, but I didn't find a comprehensive guide that shows me exactly what steps I need to take to calibrate my device, if I don't have any kalibir data files to start with.

I have managed to pull some calibration data from the firmware, which might be useful:
I have an [IMU calibration file](https://pastebin.com/wpBny0rD), that seems to contain the IMU transform, and some rectification data. Applying the transform, rectification and offset data to the raw IMU values is relatively easy.

But here's the more important file, the [Camera calibration file](https://pastebin.com/FLi7arRz). This one contains the sensor model (OV7251) along with a ton of calibration data like Projection and Distortion. Unfortunately this file is proprietary, so I don't have any documentation for it, but I'm hoping that the VIO experts working on this project can help me decode it, and potentially even use it instead of having to use kalibir.

There are five cameras on the device, but I only intend to use the two forward-facing ones with Kimera-VIO, at least for now.

Any bit of help is very much appreciated.

SparkyTD, 2020-12-14:
Update: With some help from Reddit, it looks like `Projection/PinholeSymmetric` in the json maps to `intrinsics` in the kalibr yaml file, and `Distortion/Fisheye62` might be an 8 parameter Kanala Brandt distortion model. I guess now the question becomes can I use this model in the kalibr file?

rpavlik, 2022-10-06:
better late than never: the Rift-S "fisheye62" is documented here now https://github.com/mapillary/OpenSfM/blob/874b9114f854d9c664b06bc187384934ce5d92a0/opensfm/src/geometry/src/camera.cc#L69 and SLAM is now available in Monado for it https://monado.freedesktop.org/ (thanks to @thaytan)

closed 2023-11-03 by marcusabate