MIT-SPARK/Kimera-VIO #143: Error running test executable.

JayShaw-git, 2021-01-22:
**Description:**
CMake Error at /usr/share/cmake-3.10/Modules/GoogleTestAddTests.cmake:39. Hope you can help me ,I feel tired of it with solving many problem three all days .Thanks very much!

**Command:**
```
catkin build
```

**Console output:**
```

Errors     << kimera_vio:make /home/jayshaw/catkin_ws/logs/kimera_vio/build.make.000.log                                      
In file included from /home/jayshaw/catkin_ws/src/Kimera-VIO/src/frontend/Camera.cpp:15:0:
/home/jayshaw/catkin_ws/src/Kimera-VIO/include/kimera-vio/frontend/Camera.h: In constructor ‘VIO::Camera::Camera(const VIO::CameraParams&)’:
/home/jayshaw/catkin_ws/src/Kimera-VIO/include/kimera-vio/frontend/Camera.h:59:40: warning: ‘VIO::Camera::camera_impl_’ will be initialized after [-Wreorder]
   gtsam::PinholeCamera<gtsam::Cal3_S2> camera_impl_;
                                        ^~~~~~~~~~~~
/home/jayshaw/catkin_ws/src/Kimera-VIO/include/kimera-vio/frontend/Camera.h:58:18: warning:   ‘gtsam::Cal3_S2 VIO::Camera::calibration_’ [-Wreorder]
   gtsam::Cal3_S2 calibration_;
                  ^~~~~~~~~~~~
/home/jayshaw/catkin_ws/src/Kimera-VIO/src/frontend/Camera.cpp:28:1: warning:   when initialized here [-Wreorder]
 Camera::Camera(const CameraParams& cam_params)
 ^~~~~~
In file included from /home/jayshaw/catkin_ws/src/Kimera-VIO/src/frontend/Camera.cpp:15:0:
/home/jayshaw/catkin_ws/src/Kimera-VIO/include/kimera-vio/frontend/Camera.h:58:18: warning: ‘VIO::Camera::calibration_’ will be initialized after [-Wreorder]
   gtsam::Cal3_S2 calibration_;
                  ^~~~~~~~~~~~
/home/jayshaw/catkin_ws/src/Kimera-VIO/include/kimera-vio/frontend/Camera.h:57:16: warning:   ‘VIO::CameraParams VIO::Camera::cam_params_’ [-Wreorder]
   CameraParams cam_params_;
                ^~~~~~~~~~~
/home/jayshaw/catkin_ws/src/Kimera-VIO/src/frontend/Camera.cpp:28:1: warning:   when initialized here [-Wreorder]
 Camera::Camera(const CameraParams& cam_params)
 ^~~~~~
In file included from /home/jayshaw/catkin_ws/src/Kimera-VIO/src/frontend/StereoCamera.cpp:15:0:
/home/jayshaw/catkin_ws/src/Kimera-VIO/include/kimera-vio/frontend/StereoCamera.h: In constructor ‘VIO::StereoCamera::StereoCamera(const VIO::CameraParams&, const VIO::CameraParams&, const VIO::StereoMatchingParams&)’:
/home/jayshaw/catkin_ws/src/Kimera-VIO/include/kimera-vio/frontend/StereoCamera.h:195:10: warning: ‘VIO::StereoCamera::baseline_’ will be initialized after [-Wreorder]
   double baseline_;
          ^~~~~~~~~
/home/jayshaw/catkin_ws/src/Kimera-VIO/include/kimera-vio/frontend/StereoCamera.h:170:16: warning:   ‘VIO::CameraParams VIO::StereoCamera::left_cam_params_’ [-Wreorder]
   CameraParams left_cam_params_;
                ^~~~~~~~~~~~~~~~
/home/jayshaw/catkin_ws/src/Kimera-VIO/src/frontend/StereoCamera.cpp:39:1: warning:   when initialized here [-Wreorder]
 StereoCamera::StereoCamera(const CameraParams& left_cam_params,
 ^~~~~~~~~~~~
In file included from /home/jayshaw/catkin_ws/src/Kimera-VIO/src/frontend/StereoCamera.cpp:15:0:
/home/jayshaw/catkin_ws/src/Kimera-VIO/include/kimera-vio/frontend/StereoCamera.h:176:35: warning: ‘VIO::StereoCamera::right_cam_undistort_rectifier_’ will be initialized after [-Wreorder]
   UndistorterRectifier::UniquePtr right_cam_undistort_rectifier_;
                                   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/jayshaw/catkin_ws/src/Kimera-VIO/include/kimera-vio/frontend/StereoCamera.h:174:24: warning:   ‘VIO::StereoMatchingParams VIO::StereoCamera::stereo_matching_params_’ [-Wreorder]
   StereoMatchingParams stereo_matching_params_;
                        ^~~~~~~~~~~~~~~~~~~~~~~
/home/jayshaw/catkin_ws/src/Kimera-VIO/src/frontend/StereoCamera.cpp:39:1: warning:   when initialized here [-Wreorder]
 StereoCamera::StereoCamera(const CameraParams& left_cam_params,
 ^~~~~~~~~~~~
/home/jayshaw/catkin_ws/src/Kimera-VIO/src/frontend/StereoFrame.cpp: In member function ‘void VIO::StereoFrame::getRightKeypointsLKunrectified()’:
/home/jayshaw/catkin_ws/src/Kimera-VIO/src/frontend/StereoFrame.cpp:835:21: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
   for (int i = 0; i < px_ref.size(); i++)  // fill in right frame
                   ~~^~~~~~~~~~~~~~~
/home/jayshaw/catkin_ws/src/Kimera-VIO/src/frontend/StereoFrame.cpp: In member function ‘VIO::StatusKeypointsCV VIO::StereoFrame::getRightKeypointsRectifiedRGBD(cv::Mat, cv::Mat, const StatusKeypointsCV&, const double&, const double&, const double&, const double&) const’:
/home/jayshaw/catkin_ws/src/Kimera-VIO/src/frontend/StereoFrame.cpp:1055:8: warning: unused variable ‘writeImageLeftRightMatching’ [-Wunused-variable]
   bool writeImageLeftRightMatching = false;
        ^~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/jayshaw/catkin_ws/src/Kimera-VIO/src/frontend/StereoFrame.cpp: In member function ‘std::pair<std::pair<VIO::KeypointStatus, cv::Point_<float> >, double> VIO::StereoFrame::findMatchingKeypointRectified(cv::Mat, const KeypointCV&, cv::Mat, int, int, int, int, double, bool) const’:
/home/jayshaw/catkin_ws/src/Kimera-VIO/src/frontend/StereoFrame.cpp:1154:7: warning: unused variable ‘result_cols’ [-Wunused-variable]
   int result_cols = stripe_cols - templ_cols + 1;
       ^~~~~~~~~~~
/home/jayshaw/catkin_ws/src/Kimera-VIO/src/frontend/StereoFrame.cpp:1155:7: warning: unused variable ‘result_rows’ [-Wunused-variable]
   int result_rows = stripe_rows - templ_rows + 1;
       ^~~~~~~~~~~
In file included from /home/jayshaw/catkin_ws/src/Kimera-VIO/src/dataprovider/EurocDataProvider.cpp:17:0:
[... 290 lines cut ...]
cd /home/jayshaw/catkin_ws/build/kimera_vio; catkin build --get-env kimera_vio | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
..............................................................................................................................
Failed     << kimera_vio:make                              [ Exited with code 2 ]                                             
Failed    <<< kimera_vio                                   [ 3 minutes and 41.9 seconds ]                                     
Abandoned <<< kimera_vio_ros                               [ Unrelated job failed ]                                           
[build] Summary: 12 of 14 packages succeeded.                                                                                 
[build]   Ignored:   None.                                                                                                    
[build]   Warnings:  7 packages succeeded with warnings.                                                                      
[build]   Abandoned: 1 packages were abandoned.                                                                               
[build]   Failed:    1 packages failed.                                                                                       
[build] Runtime: 8 minutes and 7.9 seconds total.                                                                             
[build] Note: Workspace packages have changed, please re-source setup files to use them.

```

**Additional files:**
Please attach all the files needed to reproduce the error.

Please give also the following information:
* Kimera-Vio branch, tag or commit used:v5.0
* GTSAM version used:4.0.0
* OpenGV version used:
* OpenCV version used: opencv 3.4.2
* Operating system and version  Ubuntu 18.04 
* Did you change the source code? (yes / no): no

ToniRV, 2021-01-22:
@JayShaw-git try to disable the unit test build, it’s a cmake
option(BUILD_TESTS "Build tests" ON)

Just do:

‘catkin config -a —cmake-args -DBUILD_TESTS=OFF’

JayShaw-git, 2021-01-24:
@ToniRV thank you very much! It builds successfully with your suggestions. But when I run the launch file with this command below:
               
   roslaunch kimera_vio_ros kimera_vio_ros_euroc.launch

Console output:
                  ... logging to /home/jayshaw/.ros/log/873b3fbc-5e19-11eb-9b41-a4bb6d571a89/roslaunch-jayshaw-3461.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://jayshaw:40725/

SUMMARY
========

PARAMETERS
 * /kimera_vio_ros/kimera_vio_ros_node/base_link_frame_id: base_link
 * /kimera_vio_ros/kimera_vio_ros_node/left_cam_frame_id: cam0
 * /kimera_vio_ros/kimera_vio_ros_node/map_frame_id: map
 * /kimera_vio_ros/kimera_vio_ros_node/mono_ransac_threshold: 30
 * /kimera_vio_ros/kimera_vio_ros_node/online_run: True
 * /kimera_vio_ros/kimera_vio_ros_node/params_folder_path: /home/jayshaw/cat...
 * /kimera_vio_ros/kimera_vio_ros_node/position_det_threshold: 0.3
 * /kimera_vio_ros/kimera_vio_ros_node/right_cam_frame_id: cam1
 * /kimera_vio_ros/kimera_vio_ros_node/stereo_ransac_threshold: 20
 * /kimera_vio_ros/kimera_vio_ros_node/use_lcd: False
 * /kimera_vio_ros/kimera_vio_ros_node/use_online_cam_params: False
 * /kimera_vio_ros/kimera_vio_ros_node/velocity_det_threshold: 0.1
 * /kimera_vio_ros/kimera_vio_ros_node/world_frame_id: world
 * /kimera_vio_ros/posegraph_viewer/frame_id: world
 * /rosdistro: melodic
 * /rosversion: 1.14.10
 * /use_sim_time: True

NODES
  /kimera_vio_ros/
    kimera_vio_ros_node (kimera_vio_ros/kimera_vio_ros_node)
    posegraph_viewer (pose_graph_tools/visualizer_node)

ROS_MASTER_URI=http://localhost:11311

process[kimera_vio_ros/kimera_vio_ros_node-1]: started with pid [3476]
/home/jayshaw/catkin_kvio/devel/lib/kimera_vio_ros/kimera_vio_ros_node: error while loading shared libraries: libopencv_viz.so.3.4: cannot open shared object file: No such file or directory

process[kimera_vio_ros/posegraph_viewer-2]: started with pid _[3477]
[ INFO] [1611475781.785926239]: Initializing pose graph visualizer
[kimera_vio_ros/kimera_vio_ros_node-1] process has died [pid 3476, exit code 127, cmd /home/jayshaw/catkin_kvio/devel/lib/kimera_vio_ros/kimera_vio_ros_node --use_lcd=false --vocabulary_path=/home/jayshaw/catkin_kvio/src/Kimera-VIO/vocabulary/ORBvoc.yml --flagfile=/home/jayshaw/catkin_kvio/src/Kimera-VIO/params/Euroc/flags/Mesher.flags --flagfile=/home/jayshaw/catkin_kvio/src/Kimera-VIO/params/Euroc/flags/VioBackEnd.flags --flagfile=/home/jayshaw/catkin_kvio/src/Kimera-VIO/params/Euroc/flags/RegularVioBackEnd.flags --flagfile=/home/jayshaw/catkin_kvio/src/Kimera-VIO/params/Euroc/flags/Visualizer3D.flags --logtostderr=1 --colorlogtostderr=1 --log_prefix=1 --v=0 --log_output=false --log_euroc_gt_data=false --output_path=/home/jayshaw/catkin_kvio/src/Kimera-VIO-ROS/output_logs/ --viz_type=0 --visualize=true left_cam/image_raw:=/cam0/image_raw right_cam/image_raw:=/cam1/image_raw imu:=/imu0 left_cam/camera_info:=/cam0/camera_info right_cam/camera_info:=/cam1/camera_info reinit_flag:=reinit_flag reinit_pose:=reinit_pose odometry:=odometry resiliency:=resiliency imu_bias:=imu_bias optimized_trajectory:=optimized_trajectory pose_graph:=pose_graph mesh:=mesh frontend_stats:=frontend_stats debug_mesh_img/image_raw:=debug_mesh_img/image_raw feature_tracks/image_raw:=feature_tracks/image_raw time_horizon_pointcloud:=time_horizon_pointcloud __name:=kimera_vio_ros_node __log:=/home/jayshaw/.ros/log/873b3fbc-5e19-11eb-9b41-a4bb6d571a89/kimera_vio_ros-kimera_vio_ros_node-1.log].
log file: /home/jayshaw/.ros/log/873b3fbc-5e19-11eb-9b41-a4bb6d571a89/kimera_vio_ros-kimera_vio_ros_node-1*.log

Additional files:
the libiary is installed in /usr/local/lib, so i don't konw why the node can't find this libiary.Can you help me again? thanks !

JayShaw-git, 2021-01-24:
@ToniRV when I use the command as below:
 
ldd kimera_vio_ros_node

Console output:
linux-vdso.so.1 (0x00007ffd8affe000)
	libopencv_core.so.3.2 => /usr/lib/x86_64-linux-gnu/libopencv_core.so.3.2 (0x00007f5397946000)
	libimage_transport.so => /opt/ros/melodic/lib/libimage_transport.so (0x00007f53976c8000)
	libglog.so.0 => /home/jayshaw/catkin_kvio/devel/lib/libglog.so.0 (0x00007f5397497000)
	libroscpp.so => /opt/ros/melodic/lib/libroscpp.so (0x00007f5397102000)
	libboost_system.so.1.65.1 => /usr/lib/x86_64-linux-gnu/libboost_system.so.1.65.1 (0x00007f5396efd000)
	libkimera_vio_ros.so => /home/jayshaw/catkin_kvio/devel/lib/libkimera_vio_ros.so (0x00007f5396bfe000)
	libgflags.so.2.2 => /home/jayshaw/catkin_kvio/devel/lib/libgflags.so.2.2 (0x00007f53969d9000)
	libgtsam.so.4 => /home/jayshaw/catkin_kvio/devel/lib/libgtsam.so.4 (0x00007f53963c7000)
	libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007f539603e000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007f5395e26000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f5395a35000)
	libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007f5395818000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007f5395614000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f53953f5000)
	librt.so.1 => /lib/x86_64-linux-gnu/librt.so.1 (0x00007f53951ed000)
	libtbb.so.2 => /usr/lib/x86_64-linux-gnu/libtbb.so.2 (0x00007f5394fb1000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f5394c13000)
	libmessage_filters.so => /opt/ros/melodic/lib/libmessage_filters.so (0x00007f5394a0e000)
	libclass_loader.so => /opt/ros/melodic/lib/libclass_loader.so (0x00007f53947e7000)
	librosconsole.so => /opt/ros/melodic/lib/librosconsole.so (0x00007f53945b1000)
	libroslib.so => /opt/ros/melodic/lib/libroslib.so (0x00007f539439e000)
	libboost_filesystem.so.1.65.1 => /usr/lib/x86_64-linux-gnu/libboost_filesystem.so.1.65.1 (0x00007f5394184000)
	libtinyxml2.so.6 => /usr/lib/x86_64-linux-gnu/libtinyxml2.so.6 (0x00007f5393f70000)
	libroscpp_serialization.so => /opt/ros/melodic/lib/libroscpp_serialization.so (0x00007f5393d6d000)
	librostime.so => /opt/ros/melodic/lib/librostime.so (0x00007f5393b4d000)
	libconsole_bridge.so.0.4 => /usr/lib/x86_64-linux-gnu/libconsole_bridge.so.0.4 (0x00007f5393948000)
	libxmlrpcpp.so => /opt/ros/melodic/lib/libxmlrpcpp.so (0x00007f5393729000)
	libcpp_common.so => /opt/ros/melodic/lib/libcpp_common.so (0x00007f539351e000)
	libboost_thread.so.1.65.1 => /usr/lib/x86_64-linux-gnu/libboost_thread.so.1.65.1 (0x00007f53932f9000)
	libboost_chrono.so.1.65.1 => /usr/lib/x86_64-linux-gnu/libboost_chrono.so.1.65.1 (0x00007f53930f4000)
	libcv_bridge.so => /opt/ros/melodic/lib/libcv_bridge.so (0x00007f5392eca000)
	libopencv_imgproc.so.3.2 => /usr/lib/x86_64-linux-gnu/libopencv_imgproc.so.3.2 (0x00007f5392979000)
	libopencv_viz.so.3.2 => /usr/lib/x86_64-linux-gnu/libopencv_viz.so.3.2 (0x00007f539271e000)
	librosbag_storage.so => /opt/ros/melodic/lib/librosbag_storage.so (0x00007f53924ad000)
	libtf.so => /opt/ros/melodic/lib/libtf.so (0x00007f5392284000)
	libtf2_ros.so => /opt/ros/melodic/lib/libtf2_ros.so (0x00007f5391fd8000)
	libtf2.so => /opt/ros/melodic/lib/libtf2.so (0x00007f5391da4000)
	libkimera_vio.so => /home/jayshaw/catkin_kvio/devel/lib/libkimera_vio.so (0x00007f53916db000)
	/lib64/ld-linux-x86-64.so.2 (0x00007f5397f95000)
	libboost_serialization.so.1.65.1 => /usr/lib/x86_64-linux-gnu/libboost_serialization.so.1.65.1 (0x00007f539149d000)
	libboost_timer.so.1.65.1 => /usr/lib/x86_64-linux-gnu/libboost_timer.so.1.65.1 (0x00007f5391297000)
	libtbbmalloc.so.2 => /usr/lib/x86_64-linux-gnu/libtbbmalloc.so.2 (0x00007f5391059000)
	libmetis.so => /home/jayshaw/catkin_kvio/devel/lib/libmetis.so (0x00007f5390dde000)
	libPocoFoundation.so.50 => /usr/lib/libPocoFoundation.so.50 (0x00007f5390a35000)
	librosconsole_log4cxx.so => /opt/ros/melodic/lib/librosconsole_log4cxx.so (0x00007f539081a000)
	librosconsole_backend_interface.so => /opt/ros/melodic/lib/librosconsole_backend_interface.so (0x00007f5390618000)
	liblog4cxx.so.10 => /usr/lib/x86_64-linux-gnu/liblog4cxx.so.10 (0x00007f539024f000)
	libboost_regex.so.1.65.1 => /usr/lib/x86_64-linux-gnu/libboost_regex.so.1.65.1 (0x00007f538ff47000)
	librospack.so => /opt/ros/melodic/lib/librospack.so (0x00007f538fd04000)
	libopencv_imgcodecs.so.3.2 => /usr/lib/x86_64-linux-gnu/libopencv_imgcodecs.so.3.2 (0x00007f538facb000)
	libvtkInteractionStyle-6.3.so.6.3 => /usr/lib/x86_64-linux-gnu/libvtkInteractionStyle-6.3.so.6.3 (0x00007f538f845000)
	libvtkRenderingLOD-6.3.so.6.3 => /usr/lib/x86_64-linux-gnu/libvtkRenderingLOD-6.3.so.6.3 (0x00007f538f62c000)
	libvtkIOPLY-6.3.so.6.3 => /usr/lib/x86_64-linux-gnu/libvtkIOPLY-6.3.so.6.3 (0x00007f538f412000)
	libvtkIOGeometry-6.3.so.6.3 => /usr/lib/x86_64-linux-gnu/libvtkIOGeometry-6.3.so.6.3 (0x00007f538f0fb000)
	libvtkFiltersTexture-6.3.so.6.3 => /usr/lib/x86_64-linux-gnu/libvtkFiltersTexture-6.3.so.6.3 (0x00007f538eed5000)
	libvtkIOExport-6.3.so.6.3 => /usr/lib/x86_64-linux-gnu/libvtkIOExport-6.3.so.6.3 (0x00007f538ec53000)
	libvtkRenderingContextOpenGL-6.3.so.6.3 => /usr/lib/x86_64-linux-gnu/libvtkRenderingContextOpenGL-6.3.so.6.3 (0x00007f538ea22000)
	libvtkRenderingOpenGL-6.3.so.6.3 => /usr/lib/x86_64-linux-gnu/libvtkRenderingOpenGL-6.3.so.6.3 (0x00007f538e582000)
	libvtkIOImage-6.3.so.6.3 => /usr/lib/x86_64-linux-gnu/libvtkIOImage-6.3.so.6.3 (0x00007f538e1ff000)
[... cut ...]
	libasn1.so.8 => /usr/lib/x86_64-linux-gnu/libasn1.so.8 (0x00007f537011d000)
	libhcrypto.so.4 => /usr/lib/x86_64-linux-gnu/libhcrypto.so.4 (0x00007f536fee7000)
	libroken.so.18 => /usr/lib/x86_64-linux-gnu/libroken.so.18 (0x00007f536fcd1000)
	libffi.so.6 => /usr/lib/x86_64-linux-gnu/libffi.so.6 (0x00007f536fac9000)
	libwind.so.0 => /usr/lib/x86_64-linux-gnu/libwind.so.0 (0x00007f536f8a0000)
	libheimbase.so.1 => /usr/lib/x86_64-linux-gnu/libheimbase.so.1 (0x00007f536f691000)
	libhx509.so.5 => /usr/lib/x86_64-linux-gnu/libhx509.so.5 (0x00007f536f447000)

ToniRV, 2021-01-29:
@JayShaw-git not sure if that is still a problem, can you make sure you update all the repos? Feel free to re-open if that doesn't fix the problem!

closed 2021-01-29 by ToniRV

wolfcanli, 2022-09-29:
> 

你好，请问你解决了上边报错问题吗？

wolfcanli, 2022-10-21:
> o/utils/FilesystemUtils.h:22:13: warning: ‘bool VIO::common::createDirectory(const string&, const string&, bool)’ defined but not used [-Wunused-function]
>  static bool createDirectory(const std::string& dir_path,

谢谢回复