MIT-SPARK/Kimera-VIO #148: Reduce docker image size

yhabib29, 2021-03-03:
The docker image built takes 17.9 GB which seems a lot to me.

By looking into the Dockerfile, I saw that the size might me reduced a lot by removing local binaries in:
- opencv/build/
- opengv/build/
- DBoW2/build/
- gtsam/build/
- V1_01_easy.zip
- /var/lib/apt/lists/* (apt cache)

This will reduces the image size to 7.14 GB.

ToniRV, 2021-03-03:
Amazing, thanks!
If you push that dockerfile with the `rm` commands, I'll be sure to merge it!

yhabib29, 2021-03-04:
Initiated a PR, but still have to test it.

Oh, and neither the display nor the entrypoint worked for me:
- The script was not accounting the dataset path argument. I have to manually specify the command when running the docker image: `/root/Kimera-VIO/scripts/stereoVIOEuroc.bash -p /root/euroc -r`
- For the display, I used x11docker as a workaround.

closed 2023-11-03 by marcusabate