MIT-SPARK/Kimera-VIO #150: stereoVIOEuroc.bash running but not showing any visualization window

whabilly, 2021-03-05:
**Description:**

Hello, I have been able to install Kimera-VIO and dependencies from source on both macOS 10.15.7 and Ubuntu 18.04.5LTS. 
Testing stereoVIOEuroc.bash on euroc/MH_01_easy dataset seems to work without errors, but without any visualization for both setups. I just followed the installation instructions and no parameter was changed in any configuration file. Did I miss something obvious??? Thanks...

**Command:**
```
./scripts/stereoVIOEuroc.bash -p "/home/SLAM/euroc/dataset/MH_01_easy"
```

**Console output:**
```

Using dataset at path: /home/SLAM/euroc/dataset/MH_01_easy
 Launching:

            ██╗  ██╗██╗███╗   ███╗███████╗██████╗  █████╗
            ██║ ██╔╝██║████╗ ████║██╔════╝██╔══██╗██╔══██╗
            █████╔╝ ██║██╔████╔██║█████╗  ██████╔╝███████║
            ██╔═██╗ ██║██║╚██╔╝██║██╔══╝  ██╔══██╗██╔══██║
            ██║  ██╗██║██║ ╚═╝ ██║███████╗██║  ██║██║  ██║
            ╚═╝  ╚═╝╚═╝╚═╝     ╚═╝╚══════╝╚═╝  ╚═╝╚═╝  ╚═╝

 
I0305 19:39:58.995168 136564160 EurocDataProvider.cpp:82] Parsing Euroc dataset...
I0305 19:39:59.393288 136564160 RegularVioBackEnd.cpp:114] Using Regular VIO backend.
I0305 19:39:59.423624 136564160 PipelineModule.h:449] MISO Pipeline Module: Display has no output queue registered.
I0305 19:39:59.423703 136564160 Pipeline.cpp:479] Pipeline Modules launched (parallel_run set to 1).
I0305 19:39:59.424573 222425088 Pipeline.h:199] Spinning Kimera-VIO.
I0305 19:39:59.424616 222961664 Pipeline.cpp:397] Shutting down VIO pipeline once processing has finished.
I0305 19:39:59.424621 222961664 Pipeline.cpp:410] Statistics
-----------                                  #	Log Hz	{avg     +- std    }	[min,max]
Data Provider [ms]                      	    0	
Display [ms]                            	    0	
Mesher [ms]                             	    0	
VioBackEnd [ms]                         	    0	
VioFrontEnd [ms]                        	    0	
Visualizer [ms]                         	    0	
backend_input_queue Size [#]            	    0	
data_provider_left_frame_queue Size [#] 	    0	
data_provider_right_frame_queue Size [#]	    0	
display_input_queue Size [#]            	    0	
mesher_backend Size [#]                 	    0	
mesher_frontend Size [#]                	    0	
stereo_frontend_input_queue Size [#]    	    0	
visualizer_backend_queue Size [#]       	    0	
visualizer_frontend_queue Size [#]      	    0	
visualizer_mesher_queue Size [#]        	    0	
I0305 19:39:59.435767 221888512 EurocDataProvider.cpp:115] Running dataset between frame 50 and frame 2000
I0305 19:39:59.927695 222961664 Pipeline.cpp:410] Statistics
-----------                                  #	Log Hz	{avg     +- std    }	[min,max]
Data Provider [ms]                      	    0	
Display [ms]                            	    0	
Mesher [ms]                             	    0	
VioBackEnd [ms]                         	    0	
VioFrontEnd [ms]                        	    0	
Visualizer [ms]                         	    0	
backend_input_queue Size [#]            	    0	
data_provider_left_frame_queue Size [#] 	  182	340.216	{89.5165 +- 29.0115}	[1,180]
data_provider_right_frame_queue Size [#]	  181	338.352	{89.0166 +- 29.0115}	[1,179]
display_input_queue Size [#]            	    0	
mesher_backend Size [#]                 	    0	
mesher_frontend Size [#]                	    0	
stereo_frontend_input_queue Size [#]    	    0	
visualizer_backend_queue Size [#]       	    0	
visualizer_frontend_queue Size [#]      	    0	
visualizer_mesher_queue Size [#]        	    0	

...

```

Please give also the following information:
* Kimera-Vio branch, tag or commit used: master
* GTSAM version used: master
* OpenGV version used: master
* OpenCV version used: 3.3.1 
* Operating system and version (e.g. Ubuntu 16.04 or Windows 10):  macOS 10.15.7 as well as Ubuntu 18.04.5LTS
* Did you change the source code? (yes / no): yes for macOS 10.15.7 to fix for compilation issues, but not for Ubuntu 18.04.5LTS

kkoledic1212, 2021-05-04:
Did you perhaps find a solution?

marcusabate, 2023-11-03:
Closing due to time elapsed, feel free to reopen if still interested.

closed 2023-11-03 by marcusabate