MIT-SPARK/Kimera-VIO #153: Segfault when running euroc example script

Alex-Dee, 2021-03-22:
**Description:**
Kimera-VIO crashes instantly when running the euroc example script.

**Command:**
```
bash ./scripts/stereoVIOEuroc.bash -p ~Data/EuRoC/V1_01_easy
```
I Am on Ubuntu 20.04 and build and installed everything as described without issues. The only changes I made were as described in #134. When running in debug with gdb attached i get:

**Console output:**
```
 VIO::OpenCvVisualizer3D::removeOldLines (this=0x7fff880ff630, 
    lmk_ids=std::vector of length -62625828676542522, capacity -758372867347507332 = {...})
    at /home/alexander/Projekte/Kimera/Kimera-VIO/src/visualizer/OpenCvVisualizer3D.cpp:1483
1483	             lmk_id_to_line_id_map.begin();

```
In addition I ran the tests and got following result:
**Console output:**
```
[----------] Global test environment tear-down
[==========] 187 tests from 41 test suites ran. (23812 ms total)
[  PASSED  ] 184 tests.
[  FAILED  ] 3 tests, listed below:
[  FAILED  ] LCDFixture.geometricVerificationCheck
[  FAILED  ] StereoFrameFixture.sparseStereoMatching
[  FAILED  ] TestTracker.getPoint3AndCovariance

 3 FAILED TESTS
  YOU HAVE 9 DISABLED TESTS

```
* Kimera-Vio branch, tag or commit used: master
* GTSAM version used: 4.0.3
* OpenGV version used: 
* OpenCV version used: 4.2
* Operating system and version (e.g. Ubuntu 16.04 or Windows 10): Ubuntu 20.04
* Did you change the source code? (yes / no): yes

akonneker, 2021-04-14:
I am seeing the same issue. Also, in debug mode instead of 3 failing tests I get a crash on the VioPipelineFixture.OnlineSequentialSpin test. Console output for that test is below:

```
[ RUN      ] VioPipelineFixture.OnlineSequentialSpin
I0414 10:02:04.880667 282747 testPipeline.cpp:124] Destroying pipeline.
I0414 10:02:04.880676 282747 testPipeline.cpp:40] Building pipeline.
I0414 10:02:04.880695 282747 EurocDataProvider.cpp:82] Parsing Euroc dataset...
I0414 10:02:05.273945 282747 RegularVioBackEnd.cpp:114] Using Regular VIO backend.
I0414 10:02:05.274183 282747 Pipeline.cpp:479] Pipeline Modules launched (parallel_run set to 1).
I0414 10:02:05.274195 282747 testPipeline.cpp:70] Connecting pipeline.
I0414 10:02:05.274209 282747 testPipeline.cpp:124] Destroying pipeline.
I0414 10:02:05.274215 282747 Pipeline.cpp:269] Pipeline destructor called.
I0414 10:02:05.274222 282747 Pipeline.cpp:431] Shutting down VIO pipeline.
I0414 10:02:05.274228 282747 Pipeline.cpp:437] Calling registered shutdown callbacks...
I0414 10:02:05.274235 282747 DataProviderInterface.cpp:72] Shutting down DataProviderInterface.
I0414 10:02:05.274246 282747 MesherModule.cpp:65] Shutting down queues for: Mesher
W0414 10:02:05.274343 282747 Pipeline.cpp:547] No lcd thread, not joining.
W0414 10:02:05.274361 282747 Pipeline.cpp:547] No visualizer thread, not joining.
I0414 10:02:05.274370 282747 Pipeline.cpp:451] VIO Pipeline's threads shutdown successfully.
VIO Pipeline successful shutdown.
I0414 10:02:05.274415 282747 VioBackEnd.h:88] Backend destructor called.
I0414 10:02:05.274452 282747 StereoVisionFrontEnd.h:64] StereoVisionFrontEnd destructor called.
I0414 10:02:05.274487 282747 EurocDataProvider.cpp:98] ETHDatasetParser destructor called.
I0414 10:02:05.277448 282747 DataProviderInterface.cpp:31] Data provider destructor called.
I0414 10:02:05.277458 282747 testPipeline.cpp:40] Building pipeline.
I0414 10:02:05.277477 282747 EurocDataProvider.cpp:82] Parsing Euroc dataset...
I0414 10:02:05.675397 282747 RegularVioBackEnd.cpp:114] Using Regular VIO backend.
I0414 10:02:05.675535 282747 Pipeline.cpp:482] Pipeline Modules running in sequential mode (parallel_run set to 0).
I0414 10:02:05.675549 282747 testPipeline.cpp:70] Connecting pipeline.
I0414 10:02:05.713717 282747 Pipeline.h:199] Spinning Kimera-VIO.
I0414 10:02:05.717499 282747 Pipeline.h:199] Spinning Kimera-VIO.
W0414 10:02:05.790082 282747 Mesher.cpp:1380] Missing landmark information to build 3D Mesh.
W0414 10:02:05.790103 282747 Mesher.cpp:1408] Missing landmark information for the Mesher!
W0414 10:02:05.790113 282747 Mesher.cpp:1608] List of Keypoints with associated Landmarks is empty.
W0414 10:02:05.790149 282747 Mesher.cpp:1422] 2D Mesh is empty!
W0414 10:02:05.790158 282747 Mesher.cpp:432] Missing landmark information for the Mesher: cannot generate 3D Mesh.
W0414 10:02:05.790216 282747 Mesher.cpp:523] Missing landmark information for the Mesher: cannot trim 3D mesh to time horizon.
I0414 10:02:05.795024 282747 Pipeline.h:199] Spinning Kimera-VIO.
W0414 10:02:05.805645 282747 PipelineModule.h:397] Module: VioBackEnd - Input queue: backend_input_queue didn't return an output.
I0414 10:02:05.810595 282747 Pipeline.h:199] Spinning Kimera-VIO.
W0414 10:02:05.823228 282747 PipelineModule.h:397] Module: VioBackEnd - Input queue: backend_input_queue didn't return an output.
I0414 10:02:05.827283 282747 Pipeline.h:199] Spinning Kimera-VIO.
W0414 10:02:05.835633 282747 PipelineModule.h:397] Module: VioBackEnd - Input queue: backend_input_queue didn't return an output.
I0414 10:02:05.839558 282747 Pipeline.h:199] Spinning Kimera-VIO.
F0414 10:02:05.922235 282747 VioBackEnd.cpp:688] Check failed: std::find(landmarks_kf->begin(), landmarks_kf->end(), lmk_id_in_kf_i) == landmarks_kf->end() 
*** Check failure stack trace: ***
    @     0x7f57f76551c3  google::LogMessage::Fail()
    @     0x7f57f765a25b  google::LogMessage::SendToLog()
    @     0x7f57f7654ebf  google::LogMessage::Flush()
    @     0x7f57f76556ef  google::LogMessageFatal::~LogMessageFatal()
    @     0x7f57fbac6d31  VIO::VioBackEnd::addStereoMeasurementsToFeatureTracks()
    @     0x7f57fbc2623f  VIO::RegularVioBackEnd::addVisualInertialStateAndOptimize()
    @     0x7f57fbac409d  VIO::VioBackEnd::addVisualInertialStateAndOptimize()
    @     0x7f57fbac2828  VIO::VioBackEnd::spinOnce()
    @     0x7f57fbabd285  VIO::VioBackEndModule::spinOnce()
    @     0x7f57fbabef28  VIO::PipelineModule<>::spin()
    @     0x7f57fbd171b0  VIO::Pipeline::spinSequential()
    @     0x7f57fbd16f08  VIO::Pipeline::spinOnce()
    @     0x7f57fbd2e209  std::__invoke_impl<>()
    @     0x7f57fbd2c282  std::__invoke<>()
    @     0x7f57fbd29ebe  _ZNSt5_BindIFMN3VIO8PipelineEFvSt10unique_ptrINS0_19StereoImuSyncPacketESt14default_deleteIS3_EEEPS1_St12_PlaceholderILi1EEEE6__callIvJOS6_EJLm0ELm1EEEET_OSt5tupleIJDpT0_EESt12_Index_tupleIJXspT1_EEE
    @     0x7f57fbd26886  std::_Bind<>::operator()<>()
    @     0x7f57fbd22bbc  std::_Function_handler<>::_M_invoke()
    @     0x7f57fb9c3927  std::function<>::operator()()
    @     0x7f57fb9bf5f9  VIO::DataProviderModule::getInputPacket()
    @     0x7f57fb9cbd25  VIO::PipelineModule<>::spin()
    @     0x55ee4b1e08a6  VIO::Pipeline::spin()
    @     0x55ee4b1d7579  VIO::VioPipelineFixture_OnlineSequentialSpin_Test::TestBody()
    @     0x55ee4b444f7f  testing::internal::HandleSehExceptionsInMethodIfSupported<>()
[... cut ...]
    @     0x55ee4b41a1b2  testing::Test::Run()
    @     0x55ee4b41ab9d  testing::TestInfo::Run()
    @     0x55ee4b41b2d8  testing::TestSuite::Run()
    @     0x55ee4b427394  testing::internal::UnitTestImpl::RunAllTests()
    @     0x55ee4b44630a  testing::internal::HandleSehExceptionsInMethodIfSupported<>()
    @     0x55ee4b43f58b  testing::internal::HandleExceptionsInMethodIfSupported<>()
    @     0x55ee4b425bd4  testing::UnitTest::Run()
    @     0x55ee4b1d693d  RUN_ALL_TESTS()
Aborted (core dumped)
```

akonneker, 2021-04-16:
Another datapoint: this issue goes away in the [feature/mono branch.](https://github.com/MIT-SPARK/Kimera-VIO/tree/feature/mono)

Alex-Dee, 2021-04-22:
Did you find a solution to your problem? My crash was due to a missing boolean return statement which I fixed. However the same crash as yours started occuring for me on the mono branch.

Alex-Dee, 2021-04-23:
It seems that the crash is somewhat related to the time horizon parameter. If I increase the value significantly the crash stops occuring. My guess is that there might be a bug in freezing / and or removing landmarks/smart factors from the graph. I wasn't able to find it yet though,

marcusabate, 2022-09-28:
We are patching these issues in a soon-to-be-released update to Kimera-VIO, hopefully this does the trick for you.

closed 2022-09-28 by marcusabate