MIT-SPARK/Kimera-VIO #154: Hardware requirement

amirhossein-hassankhani, 2021-04-01:
Hi,
What is the hardware requirement for running Kimera without any problems (no lag and freezing,...)?

ToniRV, 2021-04-01:
Hi @amirhossein-hassankhani 

The system requirement would be to use a CPU with at least 8 cores, and 16Gb of RAM (at least for the out-of-the-box Kimera).

Nevertheless, we've managed to run it in real-time in an Nvidia TX2 as well, by fine-tuning the number of features extracted.
We provide a lot more info in our recent paper: https://arxiv.org/abs/2101.06894

Right now, I'm running Kimera on an Intel Xeon(R) CPU E3-1505M v6 @ 3.00GHz × 8 cores and 32 Gb of RAM.

pstoebenau, 2022-02-22:
Hi @ToniRV,

I was wondering if there was anything specific you did to get Kimera to run on ARM64. I'm trying to make a Dockerfile to build an image for ARM64 but it doesn't build. I'm using the official docker image which works on x86 but not on ARM64. Hopefully you have some insight on this issue.

closed 2022-09-28 by marcusabate