MIT-SPARK/Kimera-VIO #155: gtsam::IndeterminantLinearSystemException  error on custom hardware

jonbakerfish, 2021-04-06:
**Description:**
Dear authors, I'm deploying Kimera-VIO on my own custom hardware (stereo cam + imu). I've followed the kalibr's instruction to get my system's params and converted to the Kimera format via `kalibr_params_to_kimera_vio_params.py`.  Also, the recorded data is converted to EuRoC format and followed the `examples/KimeraVIO.cpp` workflow. However, I encounter the following error which is quite confusing and I don't know how which part is wrong. Can you give me some advice?

```
I0406 18:08:05.412286 331853 RegularVioBackEnd.cpp:114] Using Regular VIO backend.
I0406 18:08:05.429567 331853 PipelineModule.h:449] MISO Pipeline Module: Display has no output queue registered.
I0406 18:08:05.429862 331853 Pipeline.cpp:479] Pipeline Modules launched (parallel_run set to 1).
I0406 18:08:05.430078 331867 Pipeline.h:199] Spinning Kimera-VIO.
I0406 18:08:05.430083 331868 Pipeline.cpp:397] Shutting down VIO pipeline once processing has finished.
I0406 18:08:05.430114 331866 MyDataProvider.cpp:105] Running dataset between frame 1 and frame 338
I0406 18:08:05.430119 331868 Pipeline.cpp:410] Statistics
-----------                                  #  Log Hz  {avg     +- std    }    [min,max]
Data Provider [ms]                                  0
Display [ms]                                        0
Mesher [ms]                                         0
VioBackEnd [ms]                                     0
VioFrontEnd [ms]                                    0
Visualizer [ms]                                     0
backend_input_queue Size [#]                        0
data_provider_left_frame_queue Size [#]             0
data_provider_right_frame_queue Size [#]            0
display_input_queue Size [#]                        0
mesher_backend Size [#]                             0
mesher_frontend Size [#]                            0
stereo_frontend_input_queue Size [#]                0
visualizer_backend_queue Size [#]                   0
visualizer_frontend_queue Size [#]                  0
visualizer_mesher_queue Size [#]                    0
terminate called after throwing an instance of 'gtsam::IndeterminantLinearSystemException'
  what():  
Indeterminant linear system detected while working near variable
8646911284551352320 (Symbol: x0).

Thrown when a linear system is ill-posed.  The most common cause for this
error is having underconstrained variables.  Mathematically, the system is
underdetermined.  See the GTSAM Doxygen documentation at
http://borg.cc.gatech.edu/ on gtsam::IndeterminantLinearSystemException for
more information.
```

Please give also the following information:
* Kimera-Vio branch, tag or commit used:  master latest
* Operating system and version (e.g. Ubuntu 16.04 or Windows 10): Ubuntu
* Did you change the source code? (yes / no):  yes

jonbakerfish, 2021-04-27:
After backtracing with gdb, the error is caused by the `gtsam::ISAM2::calculateEstimate` in `VioBackEnd::updateStates`. Any suggestion for the next step?

```
void VioBackEnd::updateStates(const FrameId& cur_id) {
  VLOG(10) << "Starting to calculate estimate.";
  state_ = smoother_->calculateEstimate();
```

marcusabate, 2022-09-28:
This usually signals an issue with calibration when it happens after Kimera has processed a few frames and the trajectory diverges. To see it right at the beginning is pretty strange. I would hazard a guess that it's the dataset and not the calibration since Kimera hasn't had enough time for a bad calibration to fail yet. 

I would recommend increasing the verbosity when logging (check the bash script near the end for `v=0` and change to 100) and possibly adding print statements at the dataprovider stage to make sure that your data is being parsed correctly.

marcusabate, 2023-08-21:
Closed for inactivity.

closed 2023-08-21 by marcusabate