MIT-SPARK/Kimera-VIO #156: Can Kimera work with the realsense T265 type of camera?

martinakos, 2021-04-19:
I have tried to setup Kimera to work with the stereo fisheye cameras available in the realsense T265.

I have tried my own setup and also using the files from [this commit](https://github.com/MIT-SPARK/Kimera-VIO/commit/09cf0d5df957c3d90ae9eb77a11367a6c391e40e) (which I see where immediately removed in the next commit) In both cases the tracking is very noisy at the beginning and eventually the camera pose keeps increasing without bound in some direction even though I'm not moving the camera. 

One thing I observe is that I have a lot of red features in the debug image in RViz. So I wonder if maybe the feature tracking doesn't work well with this camera because the undistorted images would have a lot of black space and the feature matching along raster lines may not be effective in this situation??? I know that orbslam3 avoids feature matching on undistorted fisheye images because of this reason. 

I see in this past [issue](https://github.com/MIT-SPARK/Kimera-VIO/issues/98) there was some talk about testing kimera with the T265 camera. Have you been able to get kimera to work properly with this camera?

marcusabate, 2022-09-28:
The spherical camera model we use currently (opencv's equi model) isn't great. We are experimenting with implementing a better camera model for these types of cameras, and will hopefully release that soon.

closed 2022-09-28 by marcusabate

lyfadvance, 2024-08-12:
you can use basalt developed by TUM. it runs all fine on T265. 
The reason for the poor performance on T265 is not due to the camera model, but rather because the distortion model was not optimized together during BA optimization. One solution is to build a loss projected onto shpere, similar to LF-VIO; One is similar to BASALT or ORB-SLAM3, with distortion models optimized together.